openscad/openscad #3838: Two exceptions triggered when rendering

sillyfunnypedro, 2021-07-16:
The following code previews just fine.

`$fn=30;

module line(p1,p2)
{
    minkowski()
    {
        hull()
        {
            translate(p1)sphere(d=0.001);
            translate(p2)sphere(d=0.001);
        }
        union()
        {
            cylinder(h=2, d=10);
            sphere(d=6);
        }
    }
}

points = [[-200, 453.5898384862], [-200, 0], [200, 0], [200, 453.5898384862], [200.0000000000, 453.5898384862], [199.4518139018, 474.5242209834], [197.8087581473, 495.4012237933], [195.0753362381, 516.1636245023], [191.2590402935, 536.7545148133], [186.3703305156, 557.1174565272], [180.4226065181, 577.1966362362], [173.4321705989, 596.9370183043], [165.4181830570, 616.2844957165], [156.4026096753, 635.1860383820], [146.4101615138, 653.5898384862], [135.4682271782, 671.4454524922], [123.6067977500, 688.7039394032], [110.8583845828, 705.3179949062], [97.2579301910, 721.2420810298], [82.8427124746, 736.4325509608], [67.6522425435, 750.8477686772], [51.7281564199, 764.4482230690], [35.1141009170, 777.1966362362], [17.8556140060, 789.0580656644], [0.0000000000, 800.0000000000], [-0.0000000000, 800.0000000000], [-17.8556140060, 789.0580656644], [-35.1141009170, 777.1966362362], [-51.7281564199, 764.4482230690], [-67.6522425435, 750.8477686772], [-82.8427124746, 736.4325509608], [-97.2579301910, 721.2420810298], [-110.8583845828, 705.3179949062], [-123.6067977500, 688.7039394032], [-135.4682271782, 671.4454524922], [-146.4101615138, 653.5898384862], [-156.4026096753, 635.1860383820], [-165.4181830570, 616.2844957165], [-173.4321705989, 596.9370183043], [-180.4226065181, 577.1966362362], [-186.3703305156, 557.1174565272], [-191.2590402935, 536.7545148133], [-195.0753362381, 516.1636245023], [-197.8087581473, 495.4012237933], [-199.4518139018, 474.5242209834], [-200.0000000000, 453.5898384862]];

for (i=[0:len(points)-1])
{
    p1 = points[i];
    p2 = points[(i+1)%len(points)];
    line(p1,p2);
}
`

It renders just fine 

`
Parsing design (AST generation)...
Saved backup file: /Users/juancho/Documents/OpenSCAD/backups/unsaved-backup-BMJ62320.scad
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 2276
Geometry cache size in bytes: 55783712
CGAL Polyhedrons in cache: 499
CGAL cache size in bytes: 79626968
Compiling design (CSG Products normalization)...
Normalized CSG tree has 46 elements
Compile and preview finished.
Total rendering time: 0 hours, 9 minutes, 14 seconds
`

When i try to render it triggers two exceptions

`
Parsing design (AST generation)...
Saved backup file: /Users/juancho/Documents/OpenSCAD/backups/unsaved-backup-BMJ62320.scad
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /Users/kintel/code/OpenSCAD/libraries/install/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 427
ERROR: CGAL error in CGALUtils::applyBinaryOperator union: CGAL ERROR: precondition violation! Expr: ! _is_valid (_succ2_P) || comp_f (node1_P->object, _succ2_P->object) != LARGER File: /Users/kintel/code/OpenSCAD/libraries/install/include/CGAL/Multiset.h Line: 2313
Geometries in cache: 2276
Geometry cache size in bytes: 55783712
CGAL Polyhedrons in cache: 500
CGAL cache size in bytes: 79626968
Total rendering time: 0 hours, 0 minutes, 19 seconds
Rendering finished.
`

t-paul, 2021-07-16:
This type of construction in 3D is likely causing numerical issues. in this case it's probably safer to construct via 2D. Ideally we want to support `minkowski()` also from 2D + 3D, but right now that's not supported, so the additional `linear_extrude()` is required.
```
minkowski() {
    linear_extrude(0.001) difference() {
        polygon(points);
        offset(-0.001) polygon(points);
    }
    ....
}
```

MichaelAtOz, 2021-07-16:
The problem is you have coincident faces where the end/next-start points overlap.
minkowski around a 0.001 shape is going to produce a range of points potentially really close to each other,
they will then likely get snapped to grids internally, making holy geometry.
Are you wedded to that minkowski method, or do you just want that shape?
Wouldn't it be easier to:
```
module line(p1,p2)
{
    {
        hull()
        {
            translate(p1) cylinder(h=2, d=10);
            translate(p2) cylinder(h=2, d=10);
        }
        hull()
        {
            translate(p1) sphere(d=6);
            translate(p2) sphere(d=6);
        }
    }
}
```

You also want to get rid of this overlap
```
, [0.0000000000, 800.0000000000]
//, [-0.0000000000, 800.0000000000]
, [-17.8556140060, 789.0580656644]
```

I also trimmed your numbers, possibly over caution, 
```
for (i=[0:1:(len(points)-1)/1])  // [22:1:25])
{
    pa=points[i];
    pb=points[(i+1)%len(points)];
    p1 = [floor(pa.x*100)/100, floor(pa.y*100)/100];
    p2 = [floor(pb.x*100)/100, floor(pb.y*100)/100];
    color(i%2==0? "red" : "green", 0.3)
    line(p1,p2);
    echo(p1=p1, p2=p2);
}
```

t-paul, 2022-02-13:
Closing, no feedback.

closed 2022-02-13 by t-paul