openscad/openscad #3839: Perhaps RTL text should default to halign="right"

jordanbrown0, 2021-07-19:
[ I'm not sure whether this is a good idea, or if it's worth the small incompatibility even if it *is* a good idea.  However, I thought I'd record it for comment and perhaps rejection. ]

Currently, RTL text proceeds to the right from the origin.  Rendering "abc" in RTL yields basically the same result as rendering "cba" in LTR.  You can think of it as either walking across the string backwards, or as moving the pen out to the end and then working your way leftward.

While that's undoubtedly the right result when valign="left", it's not at all clear that it's the right result as the default.

I think that a native writer of an RTL language would expect the text to be drawn to the *left* of the origin.  By way of analogy to paper, they would position their pen to the right edge of the paper, write the first glyph, then move left and write the second glyph, and so on.  Note that this is the exact mirror of what a LTR writer would do.

---

It isn't clear to me exactly what the internal ramifications of this change would be.  One is easy:  the user-visible behavior would be that the default halign value for RTL text would be "right".  But what does that mean for the internal "offset" variable, which says how far from the origin the string "starts"?  Currently, for RTL/halign=right, that number is a large negative number.  Arguably, for RTL/halign=right text it should be zero, as it is for LTR/halign=left text.

That's invisible to users, until `textmetrics()` #3684 comes along and exposes it and `advance`.

It may not be practical to make the whole stack be direction-agnostic, so that RTL is the mirror of LTR.  I believe the current right-of-origin behavior is coming out of HarfBuzz.  Still, text() and textmetrics() could paper over it and produce the "right" results, so that `offset` is the distance from the origin to the linguistically first glyph of the string (rather than to the leftmost) and "advance" is, by default, the [dx,dy] required to get to the origin for the next string in a sequence.

---

A similar argument applies to BTT text, versus TTB text, that BTT text should default to valign=bottom.  In TTB text, the linguistically first glyph is the closest to the origin.  One might expect that to be true in BTT, too, but BTT proceeds down from the origin and so the linguistically first glyph is the furthest from the origin.

jordanbrown0, 2021-07-19:
I should note that while I am *slightly* familiar with Hebrew, a RTL language, I am not by any means a speaker of the language, and I have absolutely no experience with TTB or BTT languages.

themactep, 2021-10-02:
When it comes to text aligment, I think it rather be start/finish or beginning/end, and then translated into left/right, top/bottom directions depending on the language writing system.

So that `text("oy vey");` internally becomes `text("oy vey", direction="ltr", halign="left");`,
`text("oy vey", align="end");` => `text("oy vey", direction="ltr", halign="right");`,
`text("אױ װײ", direction="rtl");` =>  `text("אױ װײ", direction="rtl", halign="right");`, 
`text("אױ װײ", direction="rtl", align="end");` => `text("אױ װײ", direction="rtl", halign="left");`