openscad/openscad #3841: Error rendering STL file

krhackbarth, 2021-07-20:
I have a simple program that imports a small STL file then unions it with a cube.  The model previews properly but produces an error when I attempt to render.  The cube is rendered but the STL does not appear.  The STL was originally produced by OpenSCAD from an SVG file.  I have tried running it through netFabb to repair any errors that may exist in the STL but it has no effect on the problem.  The OpenSCAD program is as follows:
cube([20,20,10],true);
translate([0,0,5])
import("and.stl", convexity=3);

The console window reads as follows when render "completes":
Parsing design (AST generation)...
Saved backup file: C:/Users/ken/OneDrive/Documents/OpenSCAD/backups/processSTL-backup-xQDmrQTm.scad
Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_p
[and.zip](https://github.com/openscad/openscad/files/6850464/and.zip)
olyhedron3(): CGAL ERROR: assertion violation! Expr: e_below != SHalfedge_handle() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_FM_decorator.h Line: 426 
Geometries in cache: 10
Geometry cache size in bytes: 14431184
CGAL Polyhedrons in cache: 5
CGAL cache size in bytes: 56720
Total rendering time: 0:00:03.821
   Top level object is a 3D object:
   Simple:        yes
   Vertices:        8
   Halfedges:      24
   Edges:          12
   Halffacets:     12
   Facets:          6
   Volumes:         2
Rendering finished.

I've attached the STL file used by the program.

I can replicate this behavior with many other STLs I've generated from SVGs with OpenSCAD.

MichaelAtOz, 2021-07-21:
That STL has way to many tiny triangles on the curves, did you use $fn=10000 or something to generate it?

This is zoomed-in to the limit and the triangles are still uncountable.
[image]

OpenSCAD has a snap to grid, with long really thin triangles, all it takes is one point to get bumped into the next grid and you get a hole. That will cause the CGAL errors.

If you want to share the SVG and scad code you used to make the STL I could suggest ways to avoid this.

krhackbarth, 2021-07-21:
Sure.  I’ve attached both the SVG file and the SCAD file I use to create an STL file from the SVG.  Note that I add two thin layers of negatively offset extrusions of the SVG to simulate a chamfer on top.  Without those two layers, the render of the first STL goes much quicker (it may import and union successfully as well – I haven’t tested that).  Unfortunately, my designs are for use by children and the chamfer is necessary.

 

Note that “and” uses type 1 scaling.

 

Thanks,

Ken

 

By the way, I’ve never used a $fn value greater than 40.  It’s specified as 20 in the attached program.

 

 <https://user-images.githubusercontent.com/1972961/126410980-201f3730-efd7-48fe-9ec0-b75bf107a375.JPG>

MichaelAtOz, 2021-07-21:
The attachments didn't make it.

krhackbarth, 2021-07-21:
See if you can get them via this link.  Let me know if you can’t.

https://drive.google.com/drive/folders/1Ty_SlxQT5vwXWYDMqfNM1jNB9JQkDO4y?usp=sharing

MichaelAtOz, 2021-07-21:
Got them.
I'm not a SVG or Inkscape guru, but when I look at the SVG the round ends are very round.
I can't see any faces no matter how much I zoom.
That causes the many small triangles when import()'d.
You need someone who knows how to get a SVG with segmented curves.
ATM it is

<path d="M174.5 258.5 174.5 343.5" stroke="#000000" stroke-width="14.6667" stroke-linecap="round" stroke-linejoin="round" stroke-miterlimit="10" fill="none" fill-rule="evenodd"/>

"round" end.
OpenSCAD just can't cope with such fine a fine curve.
I thought 'slices=' in linear_extrude() would help, it did when union()ing the X with a cube in the same .scad,
but not when exporting the X, and union()ing with the STL.

krhackbarth, 2021-07-21:
It’s weird… I have 240 of these images.  Every one uses round end caps (created in Microsoft PowerPoint).  Every image that includes a “plus”, fails.  Yet there are images with a rotated cross that render fine:

 

Isn’t this an OpenSCAD issue?  It’s responsible for creating the STL from the SVG after all…

krhackbarth, 2021-07-21:
Here’s the SVG for the image I just embedded in my last message.  Let me know if it gets lost.  When I use Inkscape to look at the tips, they look perfectly round.  Yet this image results in an STL that can be imported and union()ed…

krhackbarth, 2021-07-21:
I uploaded the SVG that works (called town.svg) to the same folder I shared earlier.

krhackbarth, 2021-07-21:
It’s weird… I have 240 of these images.  Every one uses round end caps (created in Microsoft PowerPoint).  Every image that includes a “plus”, fails.  Yet there are images with a rotated cross that render fine:

 

Isn’t this an OpenSCAD issue?  It is creating the

krhackbarth, 2021-07-21:
The problem seems to be related to the two thin layers at the top of the STL that simulate a chamfer.  I commented the associated lines from the STL maker program and put the resultant file in the shared folder.  I have no trouble union()ing this STL with a cube and rendering it.  

 

The problem may be related to OpenSCAD trying, and failing, to union() three SVG imports.  I’m going to try to implement this as union()ing three STLs (with different scaling) instead and see if OpenSCAD can do that cleanly.

 

Thanks,

Ken

krhackbarth, 2021-07-21:
Didn’t even get that far...  Just union()ing the and.stl and another and.stl:

                                         import(path_and_filename, convexity=3);

                                         translate([0,0,1-fudge])

                                         import(path_and_filename, convexity=3);

results in an STL that can’t be used in another program.

 

I’ve added single_and.stl (works) and double_and.stl (fails) to the shared folder.

 

This is definitely an OpenSCAD issue.

 

I’m going to try processing the combined stl through meshmixer and meshlab to simplify it and reduce the number of facets.

MichaelAtOz, 2021-07-22:
@t-paul 
The svg just has two strokes making a cross, this is one, the other is much the same.
```
<path d="M174.5 258.5 174.5 343.5"
stroke="#000000" 
stroke-width="14.6667" 
stroke-linecap="round" 
stroke-linejoin="round" 
stroke-miterlimit="10" 
fill="none" 
fill-rule="evenodd"/>
```
The linecap="round" results in `import()` of the svg having thousands of facets across a 2mm arm.
[image]

What is the import doing with the "round"?
Is there potential to have the $fx variables influence this?

t-paul, 2021-07-22:
I think the SVG import does not really support the `$f{a/s/n}` variables yet, or at least not everywhere. It certainly should.

The linecap is automatically generated by clipper though, so we may need to either add some special cases for that for probably do some post processing. Bonus on that would be auto-fix for those annoying mini problems that get fixed via tiny `offset()`.

ChrisCoxArt, 2021-12-23:
In the dev build, SVG import supports $fn and friends, when it can.

krhackbarth, 2021-12-23:
can you point me to where this change is documented?  thanks

ChrisCoxArt, 2021-12-23:
SVG import support forr $fn is still in the develop builds, not released yet, so not very well documented.

krhackbarth, 2022-03-06:
Tried using the latest snapshot (OpenSCAD 2022.03.05.ci11257 ) with the following import command:
import(file = path_and_filename,center=true,$fn=20);

Previews properly but I get the following error during render:
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: e_below != SHalfedge_handle()
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_FM_decorator.h
Line: 418
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: e_below != SHalfedge_handle()
File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_FM_decorator.h
Line: 418

I downloaded the zipped version of the snapshot and simply expanded it and put the SVG and .scad files in the same folder.  Ran the .exe and loaded the .scad file.

Happy to provide files if valuable.

Thanks,
Ken

krhackbarth, 2022-03-28:
I've tried $fn, $fa, $fs - both within the import() command and prior to the command - with different values.  I'm using snapshot 2022.03.28.ci11525.  As far as I can tell they have no effect.

MichaelAtOz, 2022-03-28:
I'm not an SVG expert, but I'll have a look if you post the file.

krhackbarth, 2022-03-29:
here you go. unzip, run the .scad file, then type "good" into the "graphic name" box

The code sets $fn=20 for the import.  I've tried other values as small as 5 and the preview looks the same.  If you can get things to work, try uncommenting the code with the other imports.

Thanks,
[Bliss Graphic STL maker.zip](https://github.com/openscad/openscad/files/8367166/Bliss.Graphic.STL.maker.zip)

Ken

krhackbarth, 2022-03-29:
Oh, and the problem really surfaces when all imports are active and you try to render.

MichaelAtOz, 2022-03-30:
I concur, it looks as if `$fn` has no effect. Using 2022.03.28.

I'll post some pictures tomorrow.

I'll see what the svg looks like too, but as I said' I'm no expert.

 

The current `import()` generates a bagillion triangles, and the triangulation scheme is not good.

This results in almost infinitely thin slivers around the edge of the curve.

These then turn into degenerate triangles.

Those, when you use more than two objects, cause the CGAL errors.

t-paul, 2022-03-30:
The `$fX` support for SVG via #3994 currently only allows to increase resolution compared to the old hard coded values. It might make sense to lift that limitation.

t-paul, 2022-03-31:
To clarify, the `$fX` variables are only meant to affect translation of curves into meshes. So if the high resolution comes from distinct line segments, there would not be any impact.

The main limitation should be easy to remove, it's in place to prevent breaking existing models. I suppose it's reasonable to review that.

If the high resolution is caused by the round line-cap as shown in the example above, it might be a bit trickier as that is handled separately.

krhackbarth, 2022-03-31:
Yes, I believe, without a great deal of evidence, that the resolution problem is being caused by the round-line end caps.  It's compounded because I'm unioning multiple of these extruded SVG images.

For context, these SVG images are coming from Blissymbolics Communication International.  I use Microsoft PowerPoint to thicken the lines for 3D printing.  I used rounded end caps and a poor-man's chamfer (i.e., multiple, decreasing offsets of the  extruded SVG) to make the end result more tactilely inviting for blind and deaf/blind children to investigate with their fingers.

3D printers are, unfortunately, too good at creating sharp 90-degree edges.

krhackbarth, 2022-03-31:
It's interesting that the fine resolution of the end caps has very little impact if I don't add the poor-man's chamfer.  Without those additional imports/offsets, the model renders very quickly.

MichaelAtOz, 2022-04-01:
I wonder if OpenSCAD needs something like a `remesh()' statement?

kintel, 2024-04-01:
Status as of 2024-03-24: The above examples all work with Manifold, but CGAL still struggles.

krhackbarth, 2024-04-01:
Thanks for keeping this issue alive!

Ken

kintel, 2024-04-02:
Since Manifold works, I'd encourage you to try it: `Preferences -> Features -> manifold`.
Once manifold becomes stable, we'll probably close this as we may deprioritize further work on the old CGAL backend.

krhackbarth, 2024-04-03:
WOW!!! It works and FAST!!!

When will this dev snapshot become official?

kintel, 2024-04-03:
No timeline yet - lots of corner cases to sort out, but we're working our way through them. The snapshot is pretty stable though, but may break from time to time.