openscad/openscad #3843: Rendering smooth PNG images from STL files

deji-nova, 2021-07-22:
When I render images using `openscad file.scad -o out.png`, the round surfaces are far from smooth. 
`file.scad` has the code ```import('file.stl')```

Example image:
[image]

Is there a way to increase the smoothness of images rendered from stl files?

Scopeuk, 2021-07-22:
Unfortunately not within open scad. the STL defines the exact geometry as a series of triangles (which you can see around the outside surface). openscad renders these without further processing.

deji-nova, 2021-07-22:
Then would increasing the number of triangles by remeshing before exporting help?

jordanbrown0, 2021-07-22:
Whatever tool you used to generate the STL needs to be set to use more triangles.

deji-nova, 2021-07-23:
Thanks! Increasing the number of triangles improved the smoothness.

closed 2021-07-23 by deji-nova