openscad/openscad #3849: Undocked window lost their window decoration

gromeck, 2021-07-31:
I'm using the nightly build under linux/debian (2021.07.25.nightly (git bb54b97)).
Since some of the latest updates, undocked window like the editor lost their window decoration.
In a mulitmonitor setup there is no way to undock the editor an make it fullscreen on the second screen.

My latest insstalled versions are build on these commits:
32c5c18e1   (installed 2021-06-27; quite certain, that this was the old behaviour)
cff035686   (installed 2021-07-07; uncertain, but assume that this was the old behaviour)
bb54b97   (installed 2021-07-26; changed behaviour as described above)

A quick look at the code changes between this commits did showed up obvious changes.
Maybe there was a change in the Qt lib, which explains the change?

Thanks in advance.

t-paul, 2021-08-01:
I don't see any change between release version and latest snapshot bb54b97 (on Debian/testing running Gnome on Wayland).

Do you have a screenshot? Possible other things that could influence the issue
- XWindows vs. Wayland
- Desktop Environment / Window manager.

gromeck, 2021-08-01:
I have attached two screenshots with the editor docked and undocked. If undocked, the decoration is missing, so no fullscreen icon appears.
System still runs X window and the Desktop environment has not changed -- except unintended changes ;-)

Are the deb packages for the listed version above still available, so that I could check these?

[image]
[image]

t-paul, 2021-08-01:
I'm pretty sure that's how it always looked with Qt undocked windows since that feature was added to OpenSCAD.

gromeck, 2021-08-02:
... and I am pretty sure that I was able to handle the editor window like a decorated window e.g. use the maximize click or push it against the top border of the desktop to maximize it.
Hmm.

gromeck, 2021-08-05:
So, in the meantime I had the chance to check with the nightly built 2021.06.19 (git 2c48fee).
@t-paul you were absolutly right, same behaviour.
I even tried different desktop/window managers: no decoration for the undocked editor window.
But I am still pretty sure, that I used the functionanlity "make full-screen" for the editor window.
Maybe somebody gives a hint how to decorate this window? Or should I request a change?

t-paul, 2021-08-05:
There might be window managers that ignore those window flags (e.g. maybe Wayland native). But there's not much that can be done on OpenSCAD side short of a full rewrite of the GUI. Maybe later Qt versions bring improvements, but with their current focus on mobile/3d, I don't have much hope.
Without having a developer who would commit taking that topic on, I'm not seeing much realistic chance for a change request.