openscad/openscad #3850: It's unintutive that "stl" export defaults to "asciistl"  from the command line but binary from the gui

jdukes, 2021-08-04:
When exporting STL via the gui (export as stl, F7), the default is binary stl. When exporting from the command line specifying "stl" as the file extension or using `--export-format stl` the default is ascii stl:
https://github.com/openscad/openscad/blob/d1c445d4c74272abd17c5e52af460dbd249b1d4a/src/export.h#L66

Seems like the default should match.

jdukes, 2021-08-04:
Either this, or there should be a big red warning in the manual about this behavior:
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_OpenSCAD_in_a_command_line_environment#Export_options

t-paul, 2021-08-04:
There's command line scripting and automation where people rely on ASCII STL for post processing, so that's not going to change for backward compatibility reasons.

Command line help already gives a long explanation:
```
  --export-format arg          overrides format of exported scad file when 
                               using option '-o', arg can be any of its 
                               supported file extensions.  For ascii stl 
                               export, specify 'asciistl', and for binary stl 
                               export, specify 'binstl'.  Ascii export is the 
                               current stl default, but binary stl is planned 
                               as the future default so asciistl should be 
                               explicitly specified in scripts when needed.
```
but adding a note to the manual seems a good idea.

closed 2021-08-04 by t-paul

JohnDMcMaster, 2022-08-23:
Why not change the GUI to be ASCII then?

jordanbrown0, 2022-08-24:
ASCII is slower (most of which is a bug) and much larger (which is intrinsic to the format).  The binary format really is better for common purposes.

jordanbrown0, 2022-08-24:
For a `sphere(10, $fn=50)` the binary STL is 122K and the ASCII STL is 491K.

jordanbrown0, 2025-03-09:
I've added some words to the [manual](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_OpenSCAD_in_a_command_line_environment#Export_options):
>Note: When exporting to STL, the GUI defaults to binary STL (smaller, faster) and the CLI defaults to ASCII STL (larger, slower). Post-processing pipelines may prefer ASCII STL; they should explicitly say `--export-format asciistl` in preparation for the default eventually changing to binary STL.

jordanbrown0, 2025-03-09:
#4316 explains why ASCII STL is slow.  (Capsule summary:  really poor buffering.)

jordanbrown0, 2025-03-09:
I misremembered the reason why ASCII STL is slow; I've deleted that comment above.