openscad/openscad #3854: Freezes if position is negitive with size being bigger than the screen.

WarlockD, 2021-08-12:
For the record I haven't opened up OpenSCAD in over a year and did a fresh install.  When I tried to just open a new file it showed up non-responding on my pc.  The debug all option just stopped half way though the SettingsReader.  So I started looking at stored settings and noticed the reg keys were set at position = @Point(-521 -264) and size = @Size(1096 1243).

I have added a new monitor and have a bigger main monitor now so I think that might of played into it.  Change size to 640 480 and point to 200 264, fixed it and it loads up fine now.  Not sure why it would just freeze up loading those values from the registry.

psteiner, 2022-02-19:
I can confirm the same issue on Dell laptop with a 15.6" 1920 x 1080 display, and two external 22" 1920 x 1080 displays. It doesn't matter which display is set as main, once I move OpenSCAD to a secondary display then exit, it saves its 'position' with negative coordinates, then hangs on reopening. 

As a workaround, I made a small batch file, that overwrites the 'position' data to @Point(0 0) then launches OpenSCAD. Github won't let me attach the batch file, but its contents is very simple:

```
@echo off
reg add HKCU\SOFTWARE\OpenSCAD\OpenSCAD\window /v position /t REG_SZ /d "@Point(0 0)" /f
start "" "C:\Program Files\OpenSCAD\openscad.exe"
```

I put the batch file in a folder that's on the PATH for my user account, then made a shortcut on the desktop pointing to the batch file, and set the shortcut's icon to the OpenSCAD icon. Now when I run the shortcut, OpenSCAD opens at the 0,0 position on the main display, at the size I last used. It's a small inconvenience to have to drag it to another window if needed, but at least it opens now!

rcolyer, 2022-04-17:
This sounds a lot like a duplicate of #3378.

MichaelPFrey, 2022-04-17:
I tried it under Linux for a few minutes. I can not reproduce it under linux.

There is Windows specific source code in the MainWindow Source code, that on account being about a second monitor, could be related:

this is in master:

https://github.com/openscad/openscad/blob/4db149f831e7b1c4382b1110f64b8ce340ed8492/src/gui/MainWindow.cc#L647-L663

this in the branch 2020.12-branch

https://github.com/openscad/openscad/blob/45a5e3bdd6c8071cb50b9c3f2b14a62554aafc4a/src/mainwin.cc#L568-L584

This code comes from https://github.com/openscad/openscad/commit/ea39c70471c1547dc52e4a348635ba6f2460cc13 and relates to #976

This sounds like a QT and Windows specific quirk. I can not find a way to force an issue under linux.

drposts, 2024-12-04:
See my comment:
https://github.com/openscad/openscad/issues/3378#issuecomment-2516539827