openscad/openscad #3856: How to give you modified source for extend openscad functions.

AGZ-france, 2021-08-14:
I do not speak well english, so I wil try to translate from french.
I have add since 2020 new function I have named spiral_extrude. The initial purpose was to be able to create screw, easy, and after, I realize that it was possible to do more other forms directly in openscad, with this method., which is very usefull for me. I take example of source for rotate extrude to create it.
I have updated sources from github, and each time, I need to modify my code in order to be compatible with modifications presents in new release.
I have not experience with github, and do not know how I can proceed to give you the source, in order to give for all users, the possibility to use this methods.
I can send result in stl of mcad files with this method, if you are interesting.

ChrisCoxArt, 2021-12-23:
It sounds like you need to create a pull request for your changes (after updating to pick up all recent changes to the master branch).

thehans, 2021-12-26:
@AGZ-france Sorry for the delayed response.  
To get familiar with contributing on github, there are some projects designed to walk you through the process, such as this:
https://github.com/firstcontributions/first-contributions

If you are also new to Git version control, then I would recommend searching for some introductory git tutorials (in addition to Github-specific ones such as the above).

There has been some recent discussion on IRC (channel `#openscad` on [libera.chat](https://libera.chat/)) about adding some type of spiral/helix extrude, and we would be interested to see your work.  Most of the main contributors to OpenSCAD are in that channel, so feel free to join us there for further discussion if you like (it can be useful for more immediate responses to small questions, compared to github comments).

To start, it would help to document the interface you have created starting with the module definition "signature": i.e. module name, parameter names, whether any are optional(and their default values when not given), what ranges of values are accepted for each, and how they affect the geometry result, etc.
If the new module is accepted we will need to document it in the manual, so writing up something based on our documentation for [linear_extrude](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/2D_to_3D_Extrusion#Linear_Extrude) or [rotate_extrude](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/2D_to_3D_Extrusion#Rotate_Extrude) as starting points would be great.

AGZ-france, 2022-02-20:
I have create a clone of openscad on github, in order to append sources files compatible with this new release.
I wil try to create some documentation, but I am not familary with english ...
I have notice that on my computer,  openscad , on which I had introduce for the first time the helicoïdal function was more fluid to display objects than release I get on Aout 2021 to update project, and the last release I get yesterday.
Perhaps some compiler options, or library used is the reason ...
I have never used git, so I try to create branch, but it does not appear.
I do not know how to send you information by another way than issue page.
Sorry.

t-paul, 2022-02-20:
Maybe for a first review just attach a diff (even an older one) and some examples so it's possible to understand what the change is trying to do.

AGZ-france, 2022-02-26:
On https://github.com/AGZ-france/openscad.git, you will find new files src/helicoidalextrude.cc src/spiralextrude.cc, src/spiraleextrude.h
The file src/geometry.cc is modified
I put examples in new directory Samples.AGZ, because I do not know where to put them.
I had never used git, but when I will be able to generate e branch with only my modifications, I will destroy my repository and build a new one.
The function spirale extrude is not really necessary, because helicoidal_extrude do every things

AGZ-france, 2022-03-02:
I have added possibility of creating mobius band with more flexibility on the form.
Since previous message, I have deleted the repository, and build new one in which I have a specific branch 'Helicoidal', with modified sources, and new sources.
Pleas, do not move content of GeometryEvaluato2.cc in the file GeometryEvaluator.cc, if you like to insert my modifications in the master branch,  because I have some others ideas to implement in the code, and I would like to use a specific source file.

pooh22, 2023-12-08:
I found this issue when I was looking if open issues already existed for my suggestion to add an optional vertical component to rotate_extrude. This would seem to address my suggestion, is there any progress on this?
I noticed there's a merge request open https://github.com/AGZ-france/openscad/pull/3, which I haven't tried yet... I've never compiled openscad before

jordanbrown0, 2023-12-08:
There's been some work done on PR#4538.

pooh22, 2023-12-09:
I don't see that number in de projects list, has it been merged already for the next release?

jordanbrown0, 2023-12-09:
No, it's still open:  https://github.com/openscad/openscad/pull/4538