openscad/openscad #3859: text spacing != 1.0 doesn't play nice with ligatures

jordanbrown0, 2021-08-16:
Text spacing is applied between Harfbuzz-generated glyphs.  In a font with ligatures, multiple letters can combine into one glyph, so the spacing does silly things.

```text("difficult", font="Palatino", spacing=2);```

yields

[image]

Perhaps when spacing is not 1.0 it could somehow disable use of ligatures.

jordanbrown0, 2021-08-16:
Also, spacing is applied as a simple multiplication of the advance value, so you get more space between `M`s than between `I`s:

```text("MMMIII", spacing=2);```

[image]

Setting the spacing to 2 does not mean "put twice as much space between glyphs"; instead, it means "advance twice as far as usual for each glyph".

Having spacing affect the inter-glyph spacing rather than the advance would probably yield more sensible results, but might be trickier since what comes out of Harfbuzz is the position of each glyph, not the space between them.  (And of course such a change would be an incompatible change.)

jordanbrown0, 2021-08-16:
(I am **not** planning to address either of these behaviors in the text cleanup work that I'm doing with `textmetrics()`.)

WillAdams, 2021-11-09:
I would argue that getting the ligature is providing what the user asked for --- if you don't want ligatures to be formed, turn them off in the text setting or insert a zero width no joiner (ZWNJ) in-between the characters to prevent the ligature.

jordanbrown0, 2021-11-09:
Forming ligatures isn't bad.  It's the interaction between ligatures and the "spacing" parameter that is bad, or at least can easily produce unexpected results.

Between these two interactions, I think "spacing" ends up being almost useless, and should probably be replaced by a different feature with entirely different rules.

Workaround:  walk through the string character by character, measuring each character's size with textmetrics() and rendering it separately.

t-paul, 2021-11-09:
FWIW, you don't want the ZWNJ (`U+200D`), but the Zero Width **Non** Joiner (`U+200C`)
```
text("dif\u200Cf\u200Cicult", font="Palatino", spacing=2);
```
Still looks not very smooth though,
[image]

But lets not forget monospaced fonts...
```
text("difficult", font="Liberation Mono:style=Regular", spacing=2);
```
[image]

WillAdams, 2024-02-02:
Have you tried using a monospace font such as Lucida Sans Typewriter?

rizwansarwar, 2024-02-02:
> Have you tried using a monospace font such as Lucida Sans Typewriter?

Yes I have, of course it is bit better with Monospace. My use case is to make any font and calculate spacing using textmetrics.

jordanbrown0, 2024-02-02:
@rizwansarwar You don't say what you mean by "consistent".  You cannot pull a string apart character by character and expect the same results that you would get from calling text() on the full string, because calling text() on the full string does kerning and ligature formation that can only be done when considering how the various characters relate to one another.

Here's a quick program that does a plausible job of evenly adding spacing between letters.  Use the customizer to experiment with its settings.
```
// String to display
s = "Hello, World!";
// Additional spacing (in OpenSCAD units)
spacing = 2;
// Font
font = "Liberation Sans";
// Font size
size = 10;

// Get the advance metrics (expected base location
// of the next glyph) for each character in the string,
// plus the desired additional spacing.
advances = [
    for (c = s)
        textmetrics(c, font=font, size=size).advance.x + spacing
];

// A function that adds successive entries in an array,
// so that runningSum([a,b,c]) yields [a, a+b, a+b+c].
function runningSum(a, i=0, sum=0) =
    i >= len(a)
    ? []
    : let(newSum = a[i]+sum)
        [ newSum, each runningSum(a, i+1, newSum) ];

// Prepend a zero, so that we have the positions of
// each character.
pos = concat(0, runningSum(advances));

// Position each character.
for (i=[0:len(s)-1])
    translate([pos[i],10,0]) text(s[i], font=font, size=size);
```

jordanbrown0, 2024-02-02:
My original example (at the very top) is flawed:  I asked for "Palatino" but that's wrong; the font is "Palatino Linotype".  I ended up getting "Times New Roman", as calling `fontmetrics(font="Palatino")` shows (reformatted for readability):
```
ECHO: {
    nominal = { ascent = 12.3766; descent = -3.0043; };
    max = { ascent = 14.4382; descent = -4.2589; };
    interline = 15.9709;
    font = { family = "Times New Roman"; style = "Regular"; };
}
```
Nonetheless, it demonstrates the behavior I was describing.

jordanbrown0, 2024-02-02:
Hmm.  As I experiment now, TNR does *not* demonstrate ligature formation.  I'm testing today on a different computer with a different Windows build; perhaps I *was* getting Palatino back in 2021.

jordanbrown0, 2024-02-02:
I'm going to edit the comments to avoid leaving that false information.

jordanbrown0, 2024-02-02:
For those trying to replicate the original experiment:  it appears that things have changed since then, probably in the transition from Windows 10 to Windows 11.  Today I do not have a font "Palatino"; the equivalent appears to be "Palatino Linotype".  Asking for "Palatino" yields "Times New Roman", which does not demonstrate the ligature.  Check with Help/Font List and/or fontmetrics(font="whatever") to be sure you're getting the font you expect.

jordanbrown0, 2024-02-02:
Setting spacing=0 and stacking the results can illustrate the spacing differences caused by kerning and ligatures.
```
translate([0,10,-1]) color("red") text(s, font=font, size=size);
```
In the examples below, the yellow is the "manually" spaced text resulting from the program above, with spacing=0.  The red is the normal `text()` spacing.

Ligatures in Palatino Linotype (combining `ffi` into one glyph):
[image]
Kerning in Times New Roman (tucking the `o` into the space under the right side of the `W`):
[image]

jordanbrown0, 2025-08-07:
Note that my "equal spacing" program above does not consider kerning, which is why the "o" in World is pushed so far to the right.