openscad/openscad #3865: Another ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_S2/SM_const_decorator.h Line: 329

rbielaws, 2021-08-21:
I'm using this library function in my project under OpenScad 2019.01
```
//////////////////////////////////////////////////////////////////////////////////////////////
// Paraboloid module for OpenScad
//
// Copyright (C) 2013  Lochner, Juergen
// http://www.thingiverse.com/Ablapo/designs
//
// This program is free software. It is
// licensed under the Attribution - Creative Commons license.
// http://creativecommons.org/licenses/by/3.0/
//////////////////////////////////////////////////////////////////////////////////////////////

// y = height of paraboloid
// f = focus distance
// fc : 1 = center paraboloid in focus point(x=0, y=f); 0 = center paraboloid on top (x=0, y=0)
// rfa = radius of the focus area : 0 = point focus
// detail = $fn of cone
module paraboloid (y=10, f=5, rfa=0, fc=1, detail=44){

  hi = (y+2*f)/sqrt(2);                             // height and radius of the cone -> alpha = 45Â° -> sin(45Â°)=1/sqrt(2)
  x =2*f*sqrt(y/f);                                 // x  = half size of parabola

  translate([0,0,-f*fc])                            // center on focus
      rotate_extrude($fn=detail )                   // extrude paraboild
      translate([rfa,0,0])                          // translate for fokus area
      difference(){
    union(){                                        // adding square for focal area
      projection(cut = true)                        // reduce from 3D cone to 2D parabola
          translate([0,0,f*2]) rotate([45,0,0])     // rotate cone 45Â° and translate for cutting
          translate([0,0,-hi/2])cylinder(h= hi, r1=hi, r2=0, center=true, $fn=detail);      // center cone on tip
      translate([-(rfa+x ),0]) square ([rfa+x , y ]);                                       // focal area square
    }
    translate([-(2*rfa+x ), -1/2]) square ([rfa+x ,y +1] );                     // cut of half at rotation center
  }
}

```
Nearly every form of following code fails to render the 2nd smaller paraboloid despite the two having no surfaces or volume even remotely in common.   While my project requires the objects to be in specific locations, I've tried moving the 2nd object to many locations and it doesn't seem to matter where it is I get the title error.  

```
paraboloid(y=18,f=15,rfa=5,fc=1,detail=120);
translate([60,0,0]) paraboloid (y=3,f=5,rfa=5,fc=0,detail=120);

```
As a side note, I did try version 2021.01 and the symptoms are the same.  Only the error text is different.  In that version I get:
ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron. 

In both versions either object will render correctly if the other object is not present.

MichaelAtOz, 2021-08-22:
Note that it is the second call causing the error.
```
*paraboloid(y=18,f=15,rfa=5,fc=1,detail=120);
translate([0,0,-40]) cube(1);  // Force CGAL Union()
translate([60,0,0]) paraboloid (y=3,f=5,rfa=5,fc=0,detail=120);
```

I'm still looking. I'm wondering if floating point may have the edge crossing an axis...

MichaelAtOz, 2021-08-22:
So I re-jigged it to use intersection() with a square at the origin and it works.
My conjecture is the `translate([-(2*rfa+x ), -1/2]) square ([rfa+x ,y +1] ); // cut of half at rotation center` is moving the square to a fraction short of the axis, due to x being floating point `ECHO: x = 7.74597`.

This is what I did
```
//////////////////////////////////////////////////////////////////////////////////////////////
// Paraboloid module for OpenScad
//
// Copyright (C) 2013  Lochner, Juergen
// http://www.thingiverse.com/Ablapo/designs
//
// This program is free software. It is
// licensed under the Attribution - Creative Commons license.
// http://creativecommons.org/licenses/by/3.0/
//////////////////////////////////////////////////////////////////////////////////////////////

// y = height of paraboloid
// f = focus distance
// fc : 1 = center paraboloid in focus point(x=0, y=f); 0 = center paraboloid on top (x=0, y=0)
// rfa = radius of the focus area : 0 = point focus
// detail = $fn of cone
module paraboloid (y=10, f=5, rfa=0, fc=1, detail=44){

  hi = (y+2*f)/sqrt(2);       // height and radius of the cone -> alpha = 45Â° -> sin(45Â°)=1/sqrt(2)
  x =2*f*sqrt(y/f);          // x  = half size of parabola
echo(x=x,hi=hi,hi/2);
  translate([0,0,-f*fc])                             // center on focus
      rotate_extrude($fn=detail )                    // extrude paraboild
      //difference(){
        intersection() {
      translate([rfa,0,0])                          // translate for fokus area
    union(){                                        // adding square for focal area
      projection(cut = true)                        // reduce from 3D cone to 2D parabola
          translate([0,0,f*2]) rotate([45,0,0])     // rotate cone 45Â° and translate for cutting
          translate([0,0,-hi/2])cylinder(h= hi, r1=hi, r2=0, center=true, $fn=detail); // center cone on tip
      translate([-(rfa+x ),0]) square ([rfa+x , y ]);              // focal area square
    }
    //translate([-(2*rfa+x ), -1/2]) 
    #square ([2*rfa+x ,y +1] );       // cut of half at rotation center
  }
}

*paraboloid(y=18,f=15,rfa=5,fc=1,detail=120);
translate([0,0,-40]) cube(1);  // Force CGAL Union()
//translate([60,0,0]) 
paraboloid (y=3,f=5,rfa=5,fc=0,detail=120);
```

MichaelAtOz, 2021-08-22:
So not an OpenSCAD bug. Closing.

MichaelAtOz, 2021-08-22:
>Update: Also, in both versions, sometimes the error noted displays and other times no error is reported despite the object not being present after the render is complete. That is, as if the error did occur but just wasn't reported. I have no idea what makes the difference. There are times when I can't cause the error to display and other times when the error occurs despite identical source file contents to times when it did not.

Note it is due to caching, when you get that error, the node is cached, so next render that part does not get re-evaluated and hence it does not get an error. When debugging GCAL errors, use Design/Flush-cache before F6.

closed 2021-08-22 by MichaelAtOz

MichaelAtOz, 2021-08-22:
Tho I do wonder if it would be worth the `rotate_extrude()` checking if something is outside the quadrant and producing a Warning?

rbielaws, 2021-08-23:
If I understand your explanation, it's not hard to see the argument that GCAL has a problem whereby it sometimes successfully renders objects containing slightly out-of-bounds areas and sometimes does not.   I don't, however, understand the logic by which the module code is entirely to blame for the existence of that out-of-bounds data.

Perhaps I don't understand what the underlying problem really is.

By my understanding, you think `difference` didn't really remove all of the 'out of bounds' area in the 2D object due to rounding errors and that this problem is easily fixed by switching to `intersection`.  If true, that too makes some sense although in other circumstances the reverse could be true so it's not a rule-of-thumb solution.

In any case, what doesn't quite make sense is that rotate_extrude typically reports such problems with "ERROR: all points for rotate_extrude() must have the same X coordinate sign..." and yet NOT recognizing the condition in this case is somehow "not a bug".

This seems to say to users that hidden loopholes in the built-in error checking might exist ANYWHERE and that any checking done by openscad should not be considered reliable.  Any parameterized user code that wants to be considered reliable must somehow explicitly check for every possible error itself.

I'm not exactly sure if this is even possible given the code one must use to do the checking is, necessarily, the same code that might be the origin for whatever problem one is hoping to catch.  This seems all the more difficult because I don't think functionality exists to reliably test numeric values except in comparison to literals, and even then comparison is risky unless the value is representable with absolute precision in binary.

nophead, 2021-08-23:
"all points for rotate_extrude() must have the same X coordinate sign" Is not sufficient to get a manifold object. You must also not have single points on the axis but you can have lines. I.e. a semi circle is fine to generate a circle but a triangle would generate a sort of donut with a singularity in the centre.

rbielaws, 2021-08-23:
Since I'm not sure I completely understand how the rectangle left anything (other than understanding that rounding problems in binary floating point are common and why) I can't really tell if you are saying something, *perhaps useful for me to know but otherwise,* not really relevant or if you think my problem involved such a circumstance.  

It seems to me that removing an unrotated rectangle could only leave a line on or parallel to the X axis.  It would not leave a single point unless the rectangle wasn't quite big enough and an edge point on the original object remained unremoved.  

Since the module writer used y+1 translated by 1/2 [presumably to insure that would never happen] I'm left suspecting your comment was only informational rather than relevant to this particular case.

It does leave me wondering if perhaps the original author used `difference` rather than `intersection` because a similar rounding error acting in the opposite direction for some set of parameters resulted in a tiny hole in the center that using 'difference' avoided.  

The thought of this makes it look like there is no way to develop anything stable with this tool (that involves certain math) until a way of dealing with rounding issues is made part of the package.  Even fixing 'echo' would be useful.  Currently echo(2.0/10.0) yields 0 despite documentation stating there is no integer math in the system.  It can't be used to verify single precision calculations much less double precision.

nophead, 2021-08-23:
echo(2.0/10.0); gives ECHO: 0.2 on my system.

rbielaws, 2021-12-20:
After an absolutely horrible 10 hours of trying to find a way to display the full precision value of a variable I'm now of the opinion that a reasonably supportable excuse can't exist for echo truncating any value whatsoever.  If, as alluded to above, echo(0.1) actually displayed something like 0.09999999999999999 or 0.1000000000000001 it would not only enlighten users about binary representation issues which might impact their calculations but it would allow easy debugging, validation and testing of code that currently can't be dealt with in any reasonable manner.

Even the stupidly massive amount of code below that is the result of countless other failed attempts, cannot display an arbitrary floating-point value reliably.  That is, to a point that one can tell if a rounding issue exists in a given case.  My attempts at doing so, show that use of the subtraction method for showing un-displayed digits can introduce rounding problems itself.  Moreover, str() also does rounding which complicates things more.

The code below seems to be successful at avoiding many problems that introduce new rounding errors but there are still a large number of cases where it can't.  So, it's always unclear if results can be trusted.  Thankfully the code DID help me solve the problem that prompted its development ***BUT PLEASE IF YOU KNOW OF*** something more accurate or reliable I'd be thankful for a pointer towards it.

USE:

    echo(NumberToLongString(n));  // where 'n' is an arbitrary numeric value
    
    
````

function NumberToLongString(n) = _NumberToLongString(n,DigitSet=_NumberToLongStringL1(n));

function _NumberToLongString(n,DigitSet,Result="",Depth=-1,ELoc=[],Exp="",Sign="") =
    Depth<0      // Proper default parameters can't be coded in the declaration so this step exists to compute them
    ?_NumberToLongString(abs(n),DigitSet=DigitSet,Result=Result,Depth=0,ELoc=search("e",Result),Exp=Exp,Sign=n<0?"-":"")
    :(Depth==0                          // First term may define 'e' OR may imply '.' so custom checks are needed on first term only
        ?(ELoc==[]                      // If no 'e' expression then '.' might be implied
            ?_NumberToLongString(n-DigitSet,DigitSet=_NumberToLongStringL1(n-DigitSet)// if '.' implied, insert one else don't
                ,Result=str(DigitSet,search(".",str(DigitSet))==[]?".":""),Depth=Depth+1,ELoc=ELoc,Exp=Exp,Sign=Sign)
            // When 'e' exists, value must be saved for tacking to the end of the full result
            :_NumberToLongString(n-DigitSet,DigitSet=_NumberToLongStringL1(n-DigitSet),Result=substring(str(DigitSet),0,ELoc[0])
                ,Depth=Depth+1,ELoc=ELoc,Exp=substring(str(DigitSet),ELoc[0],len(Result)),Sign=Sign)
          )
        :(Depth<2   // Float has 53 bits or 16 decimal digits so 3 instances of 6 digits should yield more than the valid values
            ?_NumberToLongString(n-DigitSet,DigitSet=_NumberToLongStringL1(n-DigitSet)
                ,Result=str(Result,_DigitsOnly(str(DigitSet))),Depth=Depth+1,ELoc=ELoc,Exp=Exp,Sign=Sign)
            // When recursion is done compute final result here
            :str(Sign,Result,_DigitsOnly(str(n)),Exp)
          )
      );
function _NumberToLongStringL1(n)      = _NumberToLongStringL2(n,str(n));  // Returns digit set from 'n' in a form ready for proper subtraction
function _NumberToLongStringL2(n,s)    = _NumberToLongStringL3(n,StringToNumber(s)); // compute test subtraction value
function _NumberToLongStringL3(n,n2)   = _NumberToLongStringL4(n,n2,n-n2);           // perform the subtraction
function _NumberToLongStringL4(n,n2,r) = r<0?n2-pow(10,floor(log(abs(r)))+1):n2;     // adjust test value if needed or return it if OK

function _DigitsOnly(s, pos=0) =
    pos<len(s)
    ?str(_IsDigit(ord(s[pos]))?s[pos]:"",s[pos]=="e"?"":_DigitsOnly(s, pos + 1))
    :"";

function _IsDigit(n) = n>=48&&n<=58;

function substring(string, start, end) =
    start < end
    ?str(string[start], substring(string, start + 1, end))
    : "";

/************************************************************************
    Converts any(?) string value matching str() generated format to the equivalent numeric value

    s:          String matching syntax generated by str() when rendering numeric values.  
*/
function StringToNumber(s) =_BuildNumFromParts([for(x=[0:1:len(s)-1]) ord(s[x])-48]);

function _BuildNumFromParts(Array,Position=0,Sign=1,LeftSide=0,RightSide=0,E=0, ES=1, Op=0) =       //
    Position >= len(Array)                                                                          // All done.  Return result
    ? (Op>=3                                                                                        // Was E explicit
        ? Sign*(ES==1?LeftSide*pow(10,E):LeftSide/pow(10,E))                                        // Yes, exponent belongs to left side
        : Sign*(LeftSide+(ES==1?RightSide*pow(10,E):RightSide/pow(10,E)))                           // No, exponent belongs to right side
       )
    :(Op==0                                                                                         // Looking for SIGN of result
        ?(Array[0]==-3                                                                              // Neg sign denoted by -3
            ?_BuildNumFromParts(Array,Position+1,-1,LeftSide,RightSide,E,ES,1)                      // negative found
[... cut ...]
                    ?(Array[Position]==-3                                                           // is it '-' (-3)
                        ? _BuildNumFromParts(Array,Position+1,Sign,LeftSide,RightSide,0,-1,4)       // YES, set exponent sign negative
                        : _BuildNumFromParts(Array,Position  ,Sign,LeftSide,RightSide,0, 1,4)       // NO, Set sign positive
                      )
                    :_BuildNumFromParts(Array,Position+1,Sign,LeftSide,RightSide,Array[Position]+10*E,ES,Op) // Consume exponent digit(s)
                  )
              )
          )
      );
````