openscad/openscad #3871: ~ContextMemoryManager(): Assertion `heapSizeAccounting.size() == 0' failed.

thehans, 2021-09-03:
OpenSCAD Nightly 2021.09.01.nightly (git 1a5d98e)  
Linux Mint 20.2

F5 preview this code:
```
module x() {}
x(123);
```

Normally this will give a warning of:
```
WARNING: Too many unnamed arguments supplied in file , line 2
```

But if "Stop on first warning" preference is enabled, then the application just crashes instantly on that failed assertion.
```
openscad-nightly: /usr/src/packages/BUILD/src/context-mm.cc:354: ContextMemoryManager::~ContextMemoryManager(): Assertion `heapSizeAccounting.size() == 0' failed.
```

thehans, 2021-09-05:
I just remembered there was another case where I ran into this same assertion crash.  Mentioned in the comment here: https://github.com/openscad/openscad/issues/2413#issuecomment-858364026

ChrisCoxArt, 2022-01-17:
last assert fails in 
```
ContextMemoryManager::~ContextMemoryManager()
{
	collectGarbage(managedContexts);
	assert(managedContexts.empty());
	assert(heapSizeAccounting.size() == 0);
}
```

ChrisCoxArt, 2022-01-17:
LOG / PRINT_NOCACHE throws inside UserModuleContext::UserModuleContext / apply_variables / Parameters::parse

ChrisCoxArt, 2022-01-17:
OK, the memory management for contexts is convoluted as heck, partly because they parse code during construction, partly because they're sometimes managed by other classes, and partly because it has evolved in a less-than-organized fashion.
The most straightforward way to solve this is to track whether we finished construction and added the context to the tracker or not, and only subtract if we managed to add.  It isn't clean, it isn't pretty, but it does avoid the mess and the assert/crash.

PR #4055 (2022-02-07): add status to context accounting, fixes #3871
Only subtract from accounting if we added to accounting in the first place, avoiding crash/assert due to subtracting too many counts from the heapSizeAccounting.
Fixes #3871
commit fa5b25fd (2022-02-07): add status so we know if we added this context to accounting, fixes #3871

Only subtract from accounting if we added to accounting in the first place.
closed 2022-02-07 by t-paul