openscad/openscad #3872: Problems with scale markers when using the right, left, front and back views

sangorrin, 2021-09-05:
Hi,

# Background

I am not a very experienced user, but I think the following is a bug. 
I have tested it using OpenSCAD version 2021.01 (stable) and the latest version right now (2021.09.01.ci8511). I guess this is related to #1206 but I feel it is still a bit different.

# The bug

When I render my model in orthogonal perspective with scale markers enabled everything seems to be fine. 

[image]

However, when I click on the right or left view the scale markers don't show numbers anymore (this problem can be reproduced both in the stable and the latest version)

[image]

If I use the front or back view only one axis shows the numbers on the 2021.01 version (it seems that this is fixed in the latest version)

[image]

Thanks,
Daniel

P.S.: The bottom and top views are displayed correctly.

sangorrin, 2021-09-05:
P.S.: I prefer the view button icons in the stable version (the ones in the latest version are too convoluted)

jordanbrown0, 2021-09-05:
The scale markers are drawn in three dimensions.  The markers on the X axis are at Z=0 and extend into +Y, the markers on the Y axis are at Z=0 and extend into +X, and the markers on the Z axis are at Y=0 and extend into +X.

They are infinitely thin.

In the view from the right or the left, you are looking at all of them edge on.  In the front and back views, you're looking at the X and Y axis markers edge-on, and at the Z-axis markers properly, but note that in the back view you're looking at the back of the Z-axis markers and so they are reversed.

Perhaps they should float around the axes so that they are always up and down on the display, but they don't.

jordanbrown0, 2021-09-05:
To make this clearer, select the right-side view and then rotate the display and watch the numbers appear.

sangorrin, 2021-09-08:
Thanks a lot @jordanbrown0, I understood the situation. Now that I think of it, it was pretty obvious.
As you mentioned, it would be nice if the markers floated around the axes, but I guess I can instead rotate the object as a workaround. I will leave the ticket open for the maintainer to comment whether he wants that feature or not.
Thanks

closed 2022-12-22 by sangorrin