openscad/openscad #3878: Weird render glitch

CAPSBOY, 2021-09-11:
I Have a VESA mount SCAD file that the screw holes on one side render fie but the other side is a glitchy mess

MichaelAtOz, 2021-09-11:
@CAPSBOY perhaps something like [this](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview?)?

CAPSBOY, 2021-09-11:
thats the one ... thank you so much ill have a look and see if its fixed

CAPSBOY, 2021-09-11:
{
        cube([plate_square_side,plate_square_side,plate_thickness], 
            center=true);   
        holes();
}

CAPSBOY, 2021-09-12:
$fn=20;

ball_dia = 25;
cyl_dia = 11;
cyl_height = 12;
correction=2;
plate_thickness = 6;
plate_countersink_dia = 10;
plate_countersink_depth = 3;
plate_square_side = 85;
screw_dia=4;

// Given that we're rotating 45 degrees and that also helpfully
// is the "deepest" part of our rotation, we can calculate the 
// ratio of the hypotenuse to the distance to a side of the square
// to find where the distance to our holes pattern.
ratio = (
    sqrt(
         (pow((plate_square_side/2),2))
         +(pow((plate_square_side/2),2))
         )
         /(plate_square_side/2));

mount_spacing = (75/2)*ratio;

start_tube = (ball_dia/2)-correction;

sphere(d=ball_dia);

translate([0,0,start_tube]){
    cylinder(d=cyl_dia,h=cyl_height+correction);
}

translate([0,0,start_tube
                    +cyl_height
                    +correction
                    +plate_thickness
                    -(plate_thickness/2)]
                    ) {
                        plate();
                    }
                   
module holes()
{
 for(a = [45:90:405])
 {
  rotate(a) {
     translate([mount_spacing,0,-plate_thickness/2]) {
        cylinder(d=screw_dia, h=plate_thickness*2);
         cylinder(d=plate_countersink_dia,h=plate_countersink_depth);
     }
 }
 }   
}

                    
module plate()
{
    difference()
    {
        cube([plate_square_side,plate_square_side,plate_thickness], 
            center=true);   
        holes();
}
}

Here is the code in full
funny your right two of the same parts are glitched, ive tried to add that code and it renders weird.
cheers

MichaelAtOz, 2021-09-12:
When posting code  use the `<>` icon above, or use three backticks (\```) on a line before & after the code.

Here I moved the two cylinders down by 1, and made them longer by 1.
So they break through the surface of the plate.

```
module holes()
{
  for(a = [45:90:405])
  {
    rotate(a) {
      translate([mount_spacing,0,-plate_thickness/2-1]) {
        cylinder(d=screw_dia, h=plate_thickness*2+1);
        cylinder(d=plate_countersink_dia,h=plate_countersink_depth+1);
      }
    }
  }
}
```

MichaelAtOz, 2021-09-12:
> When posting code use the `<>` icon above, 

As in 'above' in the GitHub edit box.
If you are emailing, indent the code and use courier/mono spaced font.

MichaelAtOz, 2022-10-05:
I think this is done. @t-paul should I remove the feedback label on closing?

closed 2022-10-05 by MichaelAtOz

t-paul, 2022-10-05:
I removed it now, but usually it does not matter much as in most searches the closed are excluded anyway.