openscad/openscad #3879: Latest snapshot exports relative to the source file instead of the current directory.

nophead, 2021-09-11:
Trying the latest snapshot on Win7 64 sometimes fails to export files because it tries in the wrong directory. E.g.

```
openscad -o tmp/foot.png tmp/png.scad --colorscheme=Nature --projection=p --imgsize=4096,4096 --camera=0,0,0,70,0,315,500 --preview --autocenter --viewall --hardwarnings
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
Geometries in cache: 1
Geometry cache size in bytes: 32696
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.183
Can't open file "tmp/foot.png" for export
```

If I create a directory tmp\tmp then it succeeds and places the file there. 

tmp\png.scad containls the following one liner:
```
color("darkorange") import("../stls/foot.stl");
```

However this seems to work and is very similar:
openscad -o stls/foot.stl tmp/stl.scad -D$bom=1 -d stls/deps/potbox.deps --hardwarnings

nophead, 2021-09-12:
The thing that seems to fix is is the -d option. 

This works

```
>openscad -o stls/foot.stl tmp/stl.scad -D$bom=1 -d stls/deps/potbox.deps --hardwarnings
ECHO: "~foot.stl(colour='[0, 0.572549, 0]')"
Geometries in cache: 12
Geometry cache size in bytes: 26880
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 389904
Total rendering time: 0:00:00.604
   Top level object is a 3D object:
   Simple:        yes
   Vertices:      210
   Halfedges:     894
   Edges:         447
   Halffacets:    464
   Facets:        232
   Volumes:         2
```

This doesn't:
```
>openscad -o stls/foot.stl tmp/stl.scad -D$bom=1  --hardwarnings
ECHO: "~foot.stl(colour='[0, 0.572549, 0]')"
Geometries in cache: 12
Geometry cache size in bytes: 26880
CGAL Polyhedrons in cache: 1
CGAL cache size in bytes: 389904
Total rendering time: 0:00:00.552
   Top level object is a 3D object:
   Simple:        yes
   Vertices:      210
   Halfedges:     894
   Edges:         447
   Halffacets:    464
   Facets:        232
   Volumes:         2
Can't open file "stls/foot.stl" for export
```

nophead, 2021-09-12:
The problem seems to be this line: https://github.com/openscad/openscad/blob/a277198ccb5dbf511b81ff582eceff0bee0b4058/src/openscad.cc#L469

It seems to change directory to the parent directory of the output path. In this case it will be `tmp` and then it will try to open relative to that `tmp/foot.stl` i.e. `tmp/tmp/foot.stl`

If the dependencies are output first this line restores the current directory, so it then works:  https://github.com/openscad/openscad/blob/a277198ccb5dbf511b81ff582eceff0bee0b4058/src/openscad.cc#L499

t-paul, 2021-09-13:
From what I can find, it's messing with the path for module instantiation including some extra exceptions due to #128 for CSG (and also active for AST). 
I'll move the path restore so it's active in general but keep the exception cases for CSG and AST. I suppose the real fix would be to not mess with CWD at all, but that's going to be quite a bit of slightly risky work.

nophead, 2021-09-13:
If it is necessary to change CWD then perhaps the output file path should be made absolute first, so it isn't affected.

t-paul, 2021-09-14:
I suspect that's only part of the issue. I think the core of the problem is that CWD is misued to handle parsing code containing relative path (e.g. `import()`). Changing that is not really a complicated topic, but it's difficult to make sure catching all the places.

closed 2021-09-14 by t-paul

nophead, 2021-09-14:
So this bug was related to this change ?:
```
DEPRECATED: Imported file (../stls/box_top.stl) found in document root instead of relative to the importing module. This behavior is deprecated
```

The importing module is in my library, so of course I don't want it relative to that. I want it relative to the file that included the library module, like it was before.

t-paul, 2021-09-14:
Imports are relative against the file doing the import, not relative to the main file (if found relative to the main file this warning happens) This behavior exists for 8 years.

I don't know what could have changed, but the only thing that comes to mind would be with `include` now picking up the actual include instead of the file doing the include.

nophead, 2021-09-14:
It used to be the case that if the import was in a used module it was relative to the file it was in but if the module was included it was relative to the file that included it.

So now I can't have a library module that imports relative to the client of the library and not relative to the library itself without getting a deprecated warning.

t-paul, 2021-09-14:
Can we please for once stop going into complain and accusation mode all the time. This really again drops my motivation to look into issues, continue that and I might just drop my OpenSCAD work completely.

I don't even know the details yet. And again that code was introduced 8 years ago. If include handling may have changed (e.g. to fix error reporting and other issues lots of people have complained about and which are important to be fixed) it might be possible and even sensible to go to the previous behavior. But that should be a discussion with an example, not some random throw-in comments on a mostly unrelated ticket.
There are test cases for include/use and none have picked up failures as far as I can tell, so this might be a case nobody anticipated in the past. The only sensible path is to first look at examples and understand if and why it would behave differently.

nophead, 2021-09-15:
The warning may have been there for 8 years but it wasn't triggered by an import in an included file that found a file relative to the file that included it. Since the last release it now gives the warning.

The only work around I can think of is each project setting $cwd to its path and using that to make an absolute path in the library import but that means it has to be configured if the project is moved, or downloaded from Github. Perhaps OpenSCAD could have $cwd built in variable.

nophead, 2021-09-15:
>Can we please for once stop going into complain and accusation mode all the time. 
Aren't all bug reports complaints that something isn't working? And if they used to work an implied accusation that somebody broke it? Github even has a Blame button to point the finger.

It is very demoralising for me realising that 10 years work is down the drain. For all that time up to and including the last release OpenSCAD has been relatively stable and the only changes I have needed to make to projects and libraries were trivial, like checking for zero size objects, replacing assign with let, checking things were a list before calling len, replacing == undef with is_undef(), etc. After the last release it has become incompatible with my library and 10 years worth of projects.

t-paul, 2021-09-15:
This is still an ongoing discussion and nothing goes down anywhere. A number of people have also spent a lot of time to keep OpenSCAD even alive not just you. If that would not have happened, the current release would be 2015.03. That would save the discussion but also mean OpenSCAD is history. I'm not sure that's the better option.

I have some ideas how to reduce the risk going forward. I would assume you have some interest in doing that, but it will need  your help.

But the way you push and demand is not compatible with me spending free time on OpenSCAD. You have put in a lot of work into your library I would like to help bringing things together. But not at the cost of my sanity.

There is already some ideas how to provide some ways for $variables to support some additional backward compatibility, but it needs a clear definition of how it needs to behave. The old randomly broken behavior should not come back, but that does not mean there could be a solution that provides enough compatibility to make things work and maybe allow slow conversion to the documented behavior.

Also yet again this is an endless discussion on an unrelated ticket. Specifically for the include file issue, it seems to be a change that was not meant to happen but probably happened when other bugs were fixed - including some you advocated for. So it still needs to be clarified what exactly happened and I see a reasonable chance to keep the bugfixes and change back to the include behavior. Same as this actual bug here was fixed.

Instead of discussing ways to fix things and trying to find common ground, all the messages lately just sounded like demands and accusations that somone is trying to hurt your library. I don't think that is the case but if we continue in that spirit, I'm not ready to keep going. That's not something I want to do in my spare time. For me that would be a sad day, seeing what amazing things people create with OpenSCAD despite it's limitations.

t-paul, 2021-09-15:
@nophead please suggest designs in #3884 to be used for integration testing covering the use cases now discussed in various places and what details to check as output.

commit 1c2aac04 (2021-09-14): Merge pull request #3880 from openscad/fix-export-cwd

Restore CWD after module instantiation (fixes #3879).
commit e4b91de3 (2021-09-14): Restore CWD after module instantiation (fixes #3879).

CSG and AST get extra treatment due to #128, trying to make sure relative
path work inside the files.