openscad/openscad #3881: Scope & changes to $special variables in "use<>" context

MichaelAtOz, 2021-09-15:
Sorry, this took longer than I anticipated. It has threads from many places.
I've raised a new issue, as #3781 was getting large, and I want to try keeping the focus concentrated.

---
Firstly I wanted to confirm my understanding of how OpenSCAD **WAS** meant to operate prior to recent changes.
Everything has a historical context.

---
### History.

We all know that there is no formal definition of the OpenSCAD language.
> "I think the main issue is that the language was never designed, but rather evolved based on immediate need." @kintel , (Nov 2012).

I had been using OpenSCAD (from 2011.something) for a while and started to have questions of how it was meant to work. Whether things I observed were bugs or undocumented features etc.

At that time the wiki was fairly basic.
So I joined GitHub (c 2012) to report bugs, which often generated debate on OpenSCAD behaviour.
When clarity arose I tried to improve the wiki with the new details.

I note that as a reminder that the wiki is NOT strictly the definition of the language.

'Variables' & scope were not documented anywhere, except maybe in @kintel's @donbright's heads at the time. Read [this](https://github.com/openscad/openscad/issues/50) if you want to know how it was back then.

---
I hadn't joined the Forum at that time, but was an avid lurker.
Things like [Variable-bug](https://forum.openscad.org/Variable-bug-td3523.html), where the above @kintel quote came from, had something familiar to recent discussions:
```
C = 250;   // Comment this line to hide bug
B = 100;
C = B + 25; // Error: unknown var 'B'

echo("B: ",B);
echo("C Size: ",C);
```

That was at a time where you could not assign in higher scopes, you required the `assign()` module,
which basically created a child module scope, there were root variables and `module()`, `function()` & `assign()` scopes.

---
The first meaty one is [Can't override include variables](https://github.com/openscad/openscad/issues/289)  (March 2013)
> Reopened. Something is wrong with includes/use I think.

> To clarify your question about the difference between $ variables and normal variables:
$ variables are meant as internal, special variables. It's not recommended to define your own, but if you do, this is how it works (today):
> * As you pointed out, warnings are suppressed when searching for unknown $ variables
> * Also, variables are searched for in all parent scopes, meaning that they are inherited downwards into modules. Normal variables are only visible in the scope where they are defined (typically a file) and if you want to pass variables down, you send them as parameters to modules.

& (at that time)
> Since $ variables are meant as internal variables, we don't support "fancy" behavior as calculating the value of a $ variable based on values of variables in some child scope. This is the reason you get the "unknown variable 'inch'" message. Instead of trying to fix this, I would rather look into **redefining variable scoping properly in a future version** of OpenSCAD, and for now, avoid using custom defined $ variables.

& (at that time)
> Regarding overriding variables: Variables in a USE'd file are hidden and you cannot override them. When writing a library with configurable parts, pass these as parameters to modules.

---
Then came [Scope refactoring](https://github.com/openscad/openscad/pull/338) & [Variable assignments in local blocks](https://github.com/openscad/openscad/issues/347), (May 2013) this introduced File-Scope and Local-Scope, allowing assignments at local scope. Note `use<>` predates this, they have the hidden File-Scope including $variables. 

---
The above Variable-bug lead to the next issue, brining `-D var=val` into the equation, [Regression: Variable override issue](https://github.com/openscad/openscad/issues/399), (Jun 2013) a range of topics.
 
> Does openscad have lexical scope or dynamic scope?

@kintel 
>I struggle a bit with that question - every module has a local, lexical scope. If a variable isn't found in that scope, it's searched in the scope *where the module was defined*, etc. until the global scope. The global scope in this context, is not the main file, but a private scope above that one, where builtins are defined.

Note the 'etc.' includes the hidden `use<>` File-Scope.

@kintel 
> Thanks for mentioning the $variables - these have a peculiar feature in that they will be searched for in all parent execution scopes, while other variables are *only* lexically scoped. I guess this means that $variables are dynamic scoped and all others are lexically scoped.
Yes, it's possible to create your own $variables.

---
I'll skip forward for brevity to a comments in [Feature request: Enums / Enumerations
](https://github.com/openscad/openscad/issues/1032) (comments Apr 2016)

>I think you will that find variables are not imported via use by design not
a bug. Neither are instantiated objects, just modules and functions.

@kintel
> @doug-moen It's not a bug - it's a workaround for not having namespaces. This makes *all top-level values* local the the use'd file.

[Note: primarily to not pollute the global scope]

---
So basically the `use<>` special hidden File-Scope has always included top level $special variables.
They have been actively utilised there as defaults (particularly the $fX's) and been able to be dynamically redefined by the parent file of the `use<>`.

This has been my understanding too.
I haven't specifically gone looking, but I have seen plenty of cases in the wild.

This all comes down to having to do workarounds to get needed functionality for complex projects.

---
The new behaviour introduced recently is a breaking change. It is not fixing a historical bug.
Yes, from a purist view it can be considered fixing a flaw, but OpenSCAD has always been a bit impure when compared to formal languages.

Nophead obviously has a lot invested in this, his framework and libraries are a phenomenal achievement working without formal mechanisms usually available.

BUT this is not just nophead, as I said it is in the wild, there will be plenty of discreet libraries that get broken if it went forward as it.

Where too?  NOTING THIS IS IN CONTEXT OF `use<>` - a particularly OpenSCAD-esq context.

For a start, I would propose taking some heat out of this situation.
Initially by saying the dynamic scope change does not need to be introduced right now.
I know there are other drivers for fixing other issues, but there is time to consider that.

I have yet to go over the last few days of counter-discusions, and impact on the performance problems. I'll get to that (later). Particularly the re-evaluation aspects. And performance is important, but perfection takes time.

Perhaps one or both of the below may be an interim step.
a. Make the change experimental, so noting it is not an accepted final design solution,
b. Nophead suggested at one stage, see below*, a backwards compatible direction.

> \* If the file scope of a used module was searched after the call stack but before the file scope of the main file then I think it would be largely backwards compatible but it fix the bug where it took precedence of the call stack unless you used if_undef() guards.

Thus a. could allow testing impact, without breaking frameworks, while letting other development changes accessible.

The real solution is to progress to the various namespace-esq library versioning and packaging solutions (discussed off & on for a while), then depricate `use<>`, that will mean staging 'fixes' to scope.

So that is the ravings of this particular idiot. I'll go and read all the latest conjecture...
I hope this hasn't upset @tp too much. `:)`

t-paul, 2021-09-15:
Thanks for digging up the past discussions. Some of those predate my involvement so I'll try to read up on those.

Using the "experimental" feature is not an option as this is a) global to the application b) defined by the user not the library author and c) not available in releases.

I've outlined the options I'm seeing already, with basically those 2 allowing progress in development:

1) Get some collective effort to update the library and document the details
This mostly means using already existing features in a way they are meant to use making it compatible with both older and newer versions!

2) Add some sort of pragma feature on file level
This can be tricky but hopefully doable, it needs a clear definition about how that extra lookup is added. It's not possible to "just go back to the broken version". This option would be a new feature trying to add backward compatibility for the given use cases. It's unlikely that it can be 100% bug-to-bug compatible.

I'm open to other suggestions, except "just revert the bugfix".

(note: you pinged the wrong user, sorry for using different nick names everywhere :smile:)

rcolyer, 2021-09-15:
> For a start, I would propose taking some heat out of this situation.

I approve.  :)  Scope discussions have the capacity to be enjoyable.

> They have been actively utilised there as defaults (particularly the $fX's) and been able to be dynamically redefined by the parent file of the use<>.

That's in fact one of the core things that was broken.  They could NOT be dynamically redefined by the parent file.  The call chain dynamic scope is ignored in the old behavior when it is overwritten by top level values in the lexical scope of the used file.  Under the old behavior it was possible to do things like the is_undef trick at the top level to dynamically decide in the lexical scope of the used file whether or not to keep the old value based on the dynamic scope value being passed along to the function or module, but the actual process of this is that the lexical scope assignment dominates over the dynamic value from the call chain.

> a. Make the change experimental, so noting it is not an accepted final design solution,

In other cases I'd be right beside you with that proposal.  But because of the nature of the code involved with this change, it's not a very viable approach for this one.  It was actually changed during a major clean-up of the context management system, across huge sections of the code, designed to make that code easier to work with for future development.  It would be tragic to lose that progress that was in the way of other things (like enabling memory stable function literals).  It's also not very viable to maintain two versions of such a large sweeping area of the codebase (where in this case the non-experimental version would have active memory leaks).

> This makes all top-level values local the the use'd file.

There's no problem with them being local to the used file.  I think that part isn't in any disagreement.  The problem is the surprise overrides at certain call boundaries.  Let me demonstrate the core logic issue with this example:

main.scad:
```
use <used.scad>
  
module Bar1() {
  $test = "from Bar1";
  Foo();
}

Bar1();
Bar2();
```

used.scad:
```
$test = "from used.scad top level";

module Foo() {
  echo($test);
}

module Bar2() {
  $test = "from Bar2";
  Foo();
}
```

Bar1 and Bar2 have identical code, except for the $test having a 1 or a 2 in it for clarity.  They call exactly the same module.  Bar1 and Bar2 are called from exactly the same place in the main file.  So you would expect the same code called from the same place and calling the same thing to do exactly the same thing.  And it does in master.  But, in 2021.01 this is the output:
```
ECHO: "from used.scad top level"
ECHO: "from Bar2"
```

I think that's some very perplexing behavior.  If someone posted this exact example as a bug report in an issue, I'd promptly say yes, fix it.

To me the core question is how can we support the old behaviors identically?  Well we can already implement them all the same way with functions and/or function literals, in a way fully supported by both 2021.01 and master.

Imagine for a moment the hypothetical where the current behavior in master is how it always was.  If someone right now proposed we supplement dynamic scope $special variables with top-level !magic variables which do not follow dynamic scope, but instead get processed every time a used file is entered and then are lexically available to all the functions and modules in that file, I'd have to think really hard about this proposal.  I'm not sure if I could justify !magic variables where EVERY !magic variable is reprocessed at every external call of a function or module, when it's so easy to get the same functionality with functions in a way that's clear, explicit, in-place, and selective to only the ones needed in a particular called function or module.

These are my thoughts about the general language design questions involved.

MichaelAtOz, 2021-09-15:
Bear with me, I tend to need to pick things up and examine them from different perspectives.
I'm just developing some instrumentation to follow what's happening.
I'm not expecting critique yet, I'm now going back to other examples shown in the last few days.

OK, I took that and turned it into a more common case with `$fn`.
`use<>` (had) behaves differently to `include<>` intentionally.
As `use` is different, you will see 'strange' effects as shown above.

[main#3881&used#3881.zip](https://github.com/openscad/openscad/files/7167126/main.3881.used.3881.zip) (disregard $fX - something I'm still playing with)

## With 2021.01.

###  Included:

```
ECHO: "Method: included - main-level"
ECHO: "OpenSCAD Version = [2021, 1, 0]"
ECHO: "*Report: included:", $fn = 10, "$fX = 10"
ECHO: " Method: included: from:["Foo"]: ", $fn = 10
ECHO: " Method: included: from:["Bar1", "Foo"]: ", $fn = 11
ECHO: " Method: included: from:["Bar2", "Foo"]: ", $fn = 22
ECHO: " Method: included: from:["Bar3", "Foo"]: ", $fn = 13
ECHO: "*Called from Bar4: ", $fn = 14
ECHO: " Method: included: from:["Bar4", "Foo"]: ", $fn = 14
```

### Used:

```
ECHO: "Method:     used - main-level"
ECHO: "OpenSCAD Version = [2021, 1, 0]"
ECHO: "*Report:     used:", $fn = 20, "$fX = 10"
ECHO: " Method:     used: from:["Foo"]: ", $fn = 20
ECHO: " Method:     used: from:["Bar1", "Foo"]: ", $fn = 20
ECHO: " Method:     used: from:["Bar2", "Foo"]: ", $fn = 22
ECHO: " Method:     used: from:["Bar3", "Foo"]: ", $fn = 13
ECHO: "*Called from Bar4: ", $fn = 14
ECHO: " Method:     used: from:["Bar4", "Foo"]: ", $fn = 20
```

The default `$fn` with `use` applies to modules within the `use`, it is the file-scope default.
If you specifically want a different `$fn` you pass it to the module as a parameter, ie `Bar3`. 

This is the beginner mode of library development, create and test your library, from the days prior to tabs in the editor. Then `use` it somewhere and you don't get the testing parts and pollution of the main scope.
(Note the `is_undef (main)` bit for testing is something I just did for this, `is_undef` is new and people didn't typically play with undef, typically there would just be a bunch of test calls in the library. I'm not saying that is best practice, just what a lot of people have done for years)

---
## With 2021.7.29.

###  Included:

```
ECHO: "Method: included - main-level"
ECHO: "OpenSCAD Version = [2021, 7, 29]"
ECHO: "*Report: included:", $fn = 10, "$fX = 10"
ECHO: " Method: included: from:["Foo"]: ", $fn = 10
ECHO: " Method: included: from:["Bar1", "Foo"]: ", $fn = 11
ECHO: " Method: included: from:["Bar2", "Foo"]: ", $fn = 22
ECHO: " Method: included: from:["Bar3", "Foo"]: ", $fn = 13
ECHO: "*Called from Bar4: ", $fn = 14
ECHO: " Method: included: from:["Bar4", "Foo"]: ", $fn = 14
```

### Used:

```
ECHO: "Method:     used - main-level"
ECHO: "OpenSCAD Version = [2021, 7, 29]"
ECHO: "*Report:     used:", $fn = 10, "$fX = 10"
ECHO: " Method:     used: from:["Foo"]: ", $fn = 10
ECHO: " Method:     used: from:["Bar1", "Foo"]: ", $fn = 11
ECHO: " Method:     used: from:["Bar2", "Foo"]: ", $fn = 22
ECHO: " Method:     used: from:["Bar3", "Foo"]: ", $fn = 13
ECHO: "*Called from Bar4: ", $fn = 14
ECHO: " Method:     used: from:["Bar4", "Foo"]: ", $fn = 14
```

So now they both have the same effect, all modules in the `use` file have changed behaviour.
And from one perspective that's good, but [later].
Whereas if you want it to behave like `include` just use `include`.

Anyway, onwards...ignore me for now.

nophead, 2021-09-15:
> They could NOT be dynamically redefined by the parent file.

They could with the `is_undef` paradigm. It solved both problems. You could override them statically or dynamically. And the dynamic override works because all variables are re-evaluated.

So this change doesn't really fix a bug or provide a means to get rid of the re-evaluation hit. It just breaks a common paradigm.

I think the correct fix is to put them back in the search order in the same place where their neighbour file scope variables are searched. It make no sense that `$a` is not found but if `b = $a` at the same scope then `b` is found.

The only bug that needed fixing was file scope $variables beat dynamic scope. As long as dynamic scope beats file scope just as local variables in functions beat normal file scope variables then it all makes sense.

rcolyer, 2021-09-15:
> It make no sense that $a is not found but if b = $a at the same scope then b is found.

I updated the testcase to evaluate this.

main.scad (identical to above):
```
use <used.scad>
  
module Bar1() {
  $test = "from Bar1";
  Foo();
}

Bar1();
Bar2();
```

used.scad:
```
$test = "from used.scad top level";
test = $test;
  
module Foo() {
  echo("dynamic:", $test);
  echo("lexical:", test);
}

module Bar2() {
  $test = "from Bar2";
  Foo();
}
```

2021.01 outputs:
```
ECHO: "dynamic:", "from used.scad top level"
ECHO: "lexical:", "from used.scad top level"
ECHO: "dynamic:", "from Bar2"
ECHO: "lexical:", "from used.scad top level"
```
The master branch outputs:
```
ECHO: "dynamic:", "from Bar1"
ECHO: "lexical:", "from used.scad top level"
ECHO: "dynamic:", "from Bar2"
ECHO: "lexical:", "from used.scad top level"
```

This seems right in master, as "$test = "from used.scad top level"; test=$test;" behaves properly for something which is evaluated once declaratively in a single-valued manner for the top-level scope, rather than dynamically at each call.  It is a different $test value in the top-level evaluation because it is at a very different point (the top level) in the call chain process, so the variables pick up what is set right next to them.

nophead, 2021-09-15:
>This seems right in master, as "$test = "from used.scad top level"; test=$test;" behaves properly for something which is evaluated once declaratively in a single-valued manner for the top-level scope, rather than dynamically at each call. 

That is one interpretation. The problem is OpenSCAD has interpreted it differently up until now. Any lexical expression that depends on a $variable is dynamically re-evaluated when the $variable gets a new value.

If I change main to:
```
use <used.scad>

$test = "from main";
  
module Bar1() {
  $test = "from Bar1";
  Foo();
}

module Main() {
    Bar1();
    Bar2();
}

Main();
```

And used to : 
```
$test = is_undef($test) ? "from used.scad top level" : $test;
test = $test;
  
module Foo() {
  echo("dynamic:", $test);
  echo("lexical:", test);
}

module Bar2() {
  $test = "from Bar2";
  Foo();
}
```

Then I get this with recent master and the last release:
```
ECHO: "dynamic:", "from Bar1"
ECHO: "lexical:", "from Bar1"
ECHO: "dynamic:", "from Bar2"
ECHO: "lexical:", "from main"
```

It shows that lexical value of $test is re-evaluated on each call, as does adding echo's in the static assignments. I know is is what causes a massive performance hit but I don't think it was an accidental bug. It allows dynamic variables to modify file scope constants.

A simple example is here: https://github.com/nophead/NopSCADlib/blob/52e9c1d7fd97555a03ea9c029887902d28aefae5/printed/cable_grommets.scad#L28

```
slot_height = round_to_layer(1.27) + layer_height;
```

`slot_height` depends on `layer_height`, which is defined in my `global_defs.scad` as:

```
layer_height    = is_undef($layer_height)    ? 0.25   : $layer_height;    // layer height when printing
```

So clients of the library can set `$layer_height` at the top of their main file and expect the library to deliver objects with printable dimensions. That still works but if I call if from a stub it doesn't.

```
use <main.scad>

Main();
``

With the last release I get:
```
ECHO: "dynamic:", "from Bar1"
ECHO: "lexical:", "from Bar1"
ECHO: "dynamic:", "from Bar2"
ECHO: "lexical:", "from main"
```

But from master I get:
```
ECHO: "dynamic:", "from Bar1"
ECHO: "lexical:", "from Bar1"
ECHO: "dynamic:", "from Bar2"
ECHO: "lexical:", "from used.scad top level"
```

This creates subtle bugs with my library. Everything uses the default values and ignores any file scope overrides, when used from another file.

martinbudden, 2021-09-16:
@rcolyer , you say:

> To me the core question is how can we support the old behaviors identically? Well we can already implement them all the same way with functions and/or function literals, in a way fully supported by both 2021.01 and master.

> Imagine for a moment the hypothetical where the current behavior in master is how it always was. If someone right now proposed we supplement dynamic scope $special variables with top-level !magic [...]

My interpretation of what you are saying is that the old behaviour is incorrect, and if we had a chance to start over again, we would never have done it that way.

So let's work on the assumption that the old behaviour was wrong and the new behaviour is correct. Even if that is the case, it doesn't mean that it is correct to change to the new behaviour. There is a body of code that relies on the old behaviour, and it is a non-trivial and error-prone exercise changing that code, see https://github.com/openscad/openscad/issues/3781#issuecomment-920799080 . A piece of software like OpenSCAD needs to ensure that new releases don't break old code, even if that involves maintaining "bug compatibility" (ie not fixing bugs because the bug is relied upon).

Maintaining bug compatibility is a well established practice in software development, and although it sometimes leaves a bad taste in the mouth, it's just one of those aspects of software development that you have to accept.

One of the classic examples is the Lotus 123 leap year bug - the Lotus developers incorrectly assumed 1900 was a leap year, which meant that the serial date number was out by one after March 1900, and many date functions relied on this value. So when Microsoft Excel came out they reproduced this bug, even though it could easily have been fixed. See https://docs.microsoft.com/en-us/office/troubleshoot/excel/wrongly-assumes-1900-is-leap-year

> If this behavior were to be corrected, many problems would arise, including:
> * Almost all dates in current Microsoft Excel worksheets and other documents would be decreased by one day. Correcting this shift would take considerable time and effort, especially in formulas that use dates.
> * Some functions, such as the WEEKDAY function, would return different values; this might cause formulas in worksheets to work incorrectly.
> * Correcting this behavior would break serial date compatibility between Microsoft Excel and other programs that use dates.

Or look at the steps Python took when moving from Python 2 to Python 3 so that existing code would not be broken.

OpenSCAD is a piece of software with a large body of users. It is important that when those users migrate to the next version their designs are not broken.

jordanbrown0, 2021-09-16:
Your comments on compatibility are in general correct.  However, taking the opposing side for a moment...  Changing consumers to cope with an incompatible change is a one-time cost, while leaving a bug in place (or failing to implement an improvement) is a long-term cost (or the lack of a long-term gain).  The cost associated with the incompatible change will only grow with time; however many cases are affected today, the number will only be larger tomorrow.  One of the reasons why new systems occasionally replace old is that the old are hobbled by their legacy, while the new are not.

The challenge is to find a way to improve while maintaining compatibility or, failing that, to improve while minimizing the cost of fixing the consumers to get past the incompatibility.

I haven't tried to deeply understand how Nophead's library uses this behavior, but perhaps there are alternate behaviors that are similar *enough* while still allowing for improvement.

Perhaps, for instance, the file-scope variables (both lexical and dynamic) might be calculated once, at "use" time, and those values preserved.  Later, at call time, those dynamic values might be inserted in the call stack between the dynamic values from the "using" file and the call-scope values from the first call.  Effectively, the call stack would be a stack of file-scope sets of variables, and then a stack of call-scope variables.  The file-scope dynamic variables would still be visible to their file, which might be enough.  The incompatibility would be that they would not override call-scope variables passed by the caller, but maybe that's close enough to the current behavior.

Alternatively, perhaps "use" could be deprecated in favor of "include", with new features allowing one to achieve the same goals.  There could be a special variable that says whether the current file is the top-level file; one could use that to control whether test/demo objects are created.  Getting a private namespace for file-scope lexical variables is harder, but object literals would help, maybe, by allowing one to gather all file-scope variables under one name.  (But:  it would be hard for them to depend on one another.)  Or perhaps something more ambitious, some kind of explicit namespace marking that would allow for a layer of lexical scoping with a mechanism for explicitly exporting names (modules, functions, or variables) as desired.

Or perhaps "use" might be deprecated in favor of still another inclusion mechanism with better semantics.

nophead, 2021-09-16:
Having file scope dynamic variables that are set once means they are not really dynamic. If you capture their value in a static file scope variable it would be fixed at one value. Yes you could override the file scope dynamic one but that would be confusing unless you made accessing file scope dynamic variables illegal form file scope.

> Effectively, the call stack would be a stack of file-scope sets of variables, and then a stack of call-scope variables. The file-scope dynamic variables would still be visible to their file, which [...]

I think the only bug in the old system was file scope dynamic beats stack based dynamic. If that were fixed I don't think I would have any problems. Instead they are now just missing from the lookup. I think they should be looked up just the same as static file scope variables as the same level but only if there isn't a dynamic version on the call stack. Just like local variables in functions. They are found in preference to file scope ones with the same name.

The separate issue is re-evaluating all the static variables at call time. I think that should be tackled by marking which ones depend on dynamic variables and only recalculating those if the dynamic variables they depend on change. In the vast majority of cases file scope variables don't depend on dynamic ones and even ones that do, the variable doesn't often change from one call to the next.

So for every file scope variable and function maintain the set of dynamic values they depend on and the last value of those variables it was calculated with and only re-evaluate it of one has changed by being overriden.

martinbudden, 2021-09-16:
@jordanbrown0 , I don't disagree with anything you say.

> The challenge is to find a way to improve while maintaining compatibility or, failing that, to improve while minimizing the cost of fixing the consumers to get past the incompatibility.

Yes, sometimes compatibility needs to be broken, which is why I mentioned Python 2 vs Python 3. However the decision to break compatibility should be a considered one, looking at the benefits, but also who is affected and by how much, and how easy it is to work around the incompatibility. And as you say, you need to minimize the cost of fixing the consumers to get past the compatibility.

Part of the problem here is that the incompatibility seems to be the result of a significant code refactoring and so is difficult to isolate.

It does seem there are a number of suggestions on how to move forward, perhaps someone (maybe you @MichaelAtOz ) could enumerate these so they could be discussed.

There are two things to be discussed:

1. What is the desired solution? Ie the solution that everyone can agree solves the problems
2. How to implement the solution

I think it is important to discuss the desired solution without concern about how difficult it is to implement. That is nothing should be excluded because it is too difficult to implement. Once a solution is agreed, then is the time to look at what compromises, if any, are required to actually implement it.

nophead, 2021-09-16:
IMHO 1. is what I posted above and 2. is also outlined above.

I think we can all agree that file scope $variables trumping stack based ones is a long standing bug. I worked around it with my `is_undef` paradigm and that also allows me to choose if my file scope $variable inherits is value from the callers file scope, or if it replaces it.  The caller can then enforce its will and override the callee by passing the value on the stack. So it was all very flexible, except with builtins like $fa, which were broken because you can used the `if_undef` trick as they are always defined.

A backwards compatible way to fix the bug is to just tweak the search order. The current master just ignores file scope $variables in used files and that breaks everything. Simply looking them up in the lexical scope if they are not found in the dynamic scope fixes everything I think.

Fixing the massive speed hit from evaluating everything on every call is a separate issue because the current master still does that. It is an optimisation problem, not a reason to completely change the semantics.

jordanbrown0, 2021-09-17:
I'm having a problem wrapping my head around the notion of (a) file-level dynamic-scope variables being at the bottom of the call stack, and at the same time (b) file-scope assignments (of both lexical and dynamic variables) being evaluated on each call.

I come up with two more-or-less simple mental models of how these variables interact.  In both of them, the file-scope variables are all evaluated at the same time.  In both of them, for both types of variables, the "most recent" assignment wins.  In one of them, that time was "now, at the time we called into this file", and the other is "a long time ago".  I think those both produce relatively simple and understandable results.  (NB, however, that "simple and understandable" does not always mean "good".)

The model that says "for dynamic variables, favor call stack over file-scope, and evaluate at call time" says that the evaluation is done "now", but the values of dynamic variables are as if they were created a long time ago.

The idea that
```
$a = 5;
b = $a;
module foo() {
    echo(b=b);
}
```
might produce any number other than 5 makes my head hurt.  I understand, technically, how at call time we could assign 5 to $a, and then when we look up $a to assign to b we find a "newer" $a, but I think I would find that very confusing and unintuitive.

For extra fun, consider
```
$a = 5;
b = $a;
$c = b;
d = $c;
module foo() {
    echo(b=b, d=d);
}
```
and observe that b and d might not be equal, if the caller overrides $c.

---
I think that file-level variables should effectively form a level of the call stack, both lexically and dynamically.  We should be able to identify one point in the call stack where they are evaluated - and they should then behave exactly as if they are evaluated at that point in the call stack, as if there was a module definition/invocation that included them.  That point might be at "use" time, or it might be at "call" time.

Although I think that abstractly both models are equally consistent, I favor the "at use time" model.  I don't think of the file level as a true level of the call stack.  I think of it as part of the "background", stuff that in a conventional language happens in the compiler or, at the latest, at file-load time, before you start executing anything - and certainly not every time you call a function in the file.

And of course the "at use time" model is always going to perform better, without needing any complex optimization to avoid pathological behavior.

jordanbrown0, 2021-09-17:
Performance tidbit:  I measured the performance of using a constant, a lexical variable, and a function to retrieve a number.

My file looks like
```
function f() = 5;
x = 5;
$y = 5;

for (i = [0:9999], j=[0:99]) {
    z = SOMETHING;
}
```
where SOMETHING is either a `5`, `x`, `$y`, or `f()`.  I also measured the loop alone, with no assignment.

Execution times were:
```
-    1.3s
5    1.7s
x    1.8s
$y   1.9s
f()  3.2s
```
So the function costs more than the variables, but not orders of magnitude more.

jordanbrown0, 2021-09-17:
In the following, what should the "stack" of values be, where the top-most (last) value in the stack is the one printed?

1.scad
```
use <2.scad>
$x = 1;
union () {
    $x = 2;
    foo();
}
```
2.scad
```
$x = 3;
module foo() {
    $x = 4;
    echo($x);
}
```

Of course 4 comes last, but what about the other three?  1,2,3,4?  1,3,2,4?  3,1,2,4?
---

In thinking about that, I prefer 1,3,2,4 or 3,1,2,4, but I'm not sure which.  (1,2,3,4 requires evaluating used-file file-level assignments on every call.)

My first thought was 1,3,2,4, that "used" files would layer in atop the files that "used" them, but in thinking about it more I'm not so sure.  1,3,2,4 would mean that a main file that says
```
$fn = 10;
foo();
```
would be very different from one that says
```
union() {
    $fn = 10;
    foo();
}
```
... and that's a very bothersome thought.

3,1,2,4 also has the advantage that 2.scad has lots of control:  it can supply a default (3), accept a default from the main file (1,2), and override that default (4).

I guess that 1,3,2,4 can get the same control by having the "used" file do
```
$x = is_undef($x) ? 3 : $x;
```
but... bleah.

nophead, 2021-09-17:
```
$a = 5;
b = $a;
module foo() {
    echo(b=b);
}
```

>might produce any number other than 5 makes my head hurt. I understand, technically, how at call time we could assign 5 to $a, and then when we look up $a to assign to b we find a "newer" $a, but I think I would find that very confusing and unintuitive.

But that is exactly how it behaves now and has always done. The only difference with master is you can't see $a in foo() but you can see b and b depends on the dynamic value of $a. Previously $a would be visible in foo() and would hide the dynamic value.

The way I got around the long standing bug was to not set $a unless I wanted it to override everything. I always copy my $ variables to ordinary ones with the same name. Like this:
```
a = is_undef($a) ? 5 : $a;
```
And I include the file with this definition in every file in my library the project files that use it.

So if `$a `is never set anywhere then `a` always has the default value 5, but if it is set, either on the stack, or anywhere on the stack of used modules then `a` gets the new value. If I set `$a` to 6 at file in my main file and then use `b.scad,` that uses `c.scad` that uses `d.scad` then `a` will be 6 in all of them. But with master now that is only true if I open `main.scad` in the GUI. If I use main.scad to make STLs then `$a = 6` is ignored, `is_undef()` is true and `a = 5` everywhere.

Also with the last release and all versions I can remember before it going back to Mendel90 in 2011, I can set `$a = 7` in say `c.scad` and then `a` will have the value 7 in `c.scad` and `d.scad`, when `d.scad` is used by `c.scad` but if `d.scad` is also used by `main.scad` it `a `would have the value 6. This is all broken now.

If I wanted to just change 'a' on a single call I would do `let($a = 10)' module_in_b();` and that would affect `a` in `module_in_b()` and any module, or function called by or child of 'module_in_b()'.

So it was an amazingly flexible mechanism to have global defaults in a library and be able to override them very selectively.

nophead, 2021-09-17:
>So the function costs more than the variables, but not orders of magnitude more.

Currently any function call into a module evaluates all the file scope variables of that module. My library has very few file scope variables internally so calls into it are fairly quick but as soon as I have a multi-file project calls between files explode in time because each file in the project has lots of list constants from the library, possible hundreds and fasteners, etc. Any call then re-evaluates all of them despite the fact they are constant and it gets to hundreds of thousands of evaluations and can take 2 minutes.

nophead, 2021-09-17:
To be clear: I don't think any amount of re-writing my library can get this flexible configuration scheme back. even if replace all my file scope constants the depend on dynamic variables with functions, the fact that file scope dynamic variables are no longer seen by functions means that I can't override things at file scope. I would have to do it on every call.

martinbudden, 2021-09-17:
There is also the issue that even if NopSCADlib could be re-written, then (virtually) all projects using that library would also have to be re-written.

jordanbrown0, 2021-09-17:
[ Jordan wrote: ]
> > The idea that
> > ```
> > $a = 5;
> > b = $a;
> > 
> > module foo() {
> >     echo(b=b);
> > }
> > ```
> > might produce any number other than 5 makes my head hurt. [...]

[ Nophead wrote: ]
> But that is exactly how it behaves now and has always done.

That that echo could produce something other than 5?

I can't get any build of OpenSCAD to produce anything other than 5, because either:
* If this is the top-level file:  the assignments of $a and b are done before any module execution begins.
* If this is a "used" file, the file-level setting of $a overrides everything else in the call stack when it is assigned to b.

jordanbrown0, 2021-09-17:
> Currently any function call into a module evaluates all the file scope variables of that module.

Yes, and that needs to be fixed.  (And the need to fix it is one of the biggest reasons for playing with the evaluation and scope rules.)

But it's also not entirely relevant.  A function call into a "used" file evaluates all file-scope variables in that file.  But a function call from one point in that file to another does not; file-internal use of functions is cheap.

nophead, 2021-09-17:
The latter is true I think. When the file is used it will see `$a` from the file using it but if it blindly sets it to 5 then it will be 5, so I stand corrected.

But if you change it to this:
foo.scad:
```
$a = is_undef($a) ? 5 : $a;
b = $a;

module foo() {
    echo(b=b);
}
```

bar.scad:
```use <foo.scad>

let($a = 6) foo();

```
Then 
ECHO: b = 6

The fact that file internal  function use is cheap is not relevant because if I  make all my global constants into functions they won't be internal.

>Yes, and that needs to be fixed.

It needs to be fixed by not re-evaluating variables that can never change, or haven't changed since the last call. That is nearly all of them. If it is fixed by changing the semantics then my library can not work, so I will have to stick on the last release or fork OpenSCAD and maintain a version with the current semantics. That seems like a lot of work though, so unless OpenSCAD came out with a  killer new feature I would just stick with the last release for the command line to get correct STLs, etc and wait for the first snapshot that fixes the black window problem and use that for the GUI.

nophead, 2021-09-17:
For a real world example consider the following scenario: I want to design a big machine that has lots of tiny toolheads, so I decide to print the chunky parts of the machine with a 0.6mm nozzle and the tool heads with a 0.3mm nozzle. With the last release I can do this:

In `main.scad` I set `$layer_height` to 0.4mm and `$extrusion_width` to 0.8mm and in my `toolhead.scad` I override them to 0.2mm and 0.4mm at file scope.

The toolhead is made from lots of little parts and they call into the library to make teardrop holes that are corrected for the staircase effect. When teardrops.scad is envolked from toolhead.scad it sees `layer_height` set to 0.2mm bu when it is called from `main.scad` it makes teardrops for 0.4mm layers.

So the static file level variable in `teardrops.scad` will have two values and any other file level variables that depend on it will have two values. When most of the parts are made `layer_height` will be 0.4 but occasionally 0.2. So the file scope variables don't need to be evaluated on every call they just have to be aware they depend on $layer_height and be re-evaluated when that changes. In most of my projects it will never change. Most of my constants have the same value throughout.

jordanbrown0, 2021-09-17:
You bring up an interesting point.  In the "evaluated once" model of file-level variables, simple A>B>C stacks will evaluate the file-level variables once, and A, B, and C will each have their own particular sets of values.  But that's not really right, because inclusion is a DAG, not a stack.  You can have A>B, A>C, and B>C, with C included by both A and B.  In such a case, you need a set of values for each path, not for each file per se.

I think much (but perhaps not all) of your desire would be met by evaluating the file-level variables once per "use" (and thus twice for C, for the case above), and then considering the call stack to be [ using, used, ..., calling, called, ... ].

Thus, if we have the "use" graph above, and a module Alpha in A calls a module Charlie in C, Charlie would see the last value set by:  [ file-A, file-C, module-Alpha, module-Charlie ].  If Alpha in A calls Bravo in B, and that calls Charlie in C, Charlie would see the last value set by [ file-A, file-B, file-C, module-Alpha, module-Bravo, module-Charlie ].

File A could set a value that would be used across the project, unless overridden by B, C, or one of the modules.
File B could set a value that would be used by B and by C-when-called-by-B, unless overridden by C or one of the modules.
File C could set a value that would be used by C, unless overridden by one of the modules.

Note that a particular file could use the "is_undef" trick to supply a value only if its "parent" file didn't supply one.

Note that module settings always override file settings, so when Alpha calls Bravo calls Charlie, if Alpha sets a $ variable its value would override any file-level values anywhere in the picture.

---

What that **doesn't** allow for is file-level variables (whether lexical or dynamic) that are set based on **module**-supplied values.

rcolyer, 2021-09-18:
A core goal needs to be having a language which is clear and teachable, following consistent rules.  I'm pretty sure that the majority of the program's users come in with somewhere from modest through no programming background.  Therefore we should sustain simple rules to give them for where to look for where a value came from.  Extra rules for dynamic values slipping in from side-locations potentially very far away (e.g., in an included file of an included file inside of a used file?!) can make it very difficult to debug any code, because you literally have to check an entire tree of included files just to be sure there are no side-sections of code that conditionally change a dynamic value.  Finding where a variable's value was set should not resemble the C++ rules for argument dependent lookup.  :)  The lexical and dynamic rules in standard form are pretty straightforward to explain, and pretty straightforward to track when debugging, which is why they persisted across so many languages.

Absolutely everything needing a dynamic default can be done with a function.  I stipulate there is a very minor performance overhead to a function call over a variable setting at the top level, but there are three very important things to consider:  One, this overhead is so small that you have to do quite a lot of repetitions just to measure it, whereas the typical usage does not have this property at all.  Two, the overhead is less than a factor of two, which means as soon as we fix the bug from repeated evaluations, then in most code that will make up for the function call performance overhead several times over (this is a fundamental gain, from doing less repetitive unnecessary work).  Three, the current function call performance head is not intrinsic to function calls.  The old assignments and the function calls are doing the same fundamental work internally, so there should certainly be an opportunity to optimize the function call to get the same speed, which means we should not design semantics around a small arbitrary performance difference in the current implementation.

A benefit of the function calls is that it is clear what they are doing.  There is no magic appearance of a dynamic change.  You call a function specifically to evaluate some value dynamically, and thus the result you get from it has exactly the behavior of the semantics on the label, and there are no surprises.  This makes them easy to understand, and straightforward to teach and explain.

nophead, 2021-09-18:
Well I would have to rewrite everything I have written in OpenSCAD for the last 10 years or so and can no longer set $layer_height at file scope to change an entire file full of parts. I would have to add it to all the functions and modules in that file, so when it is used and called from somewhere else then it uses the correct value. Hardly any point to dynamic variables then, I might as well pass more parameters with a default.

And any file scope capture of a dynamic variable would need to be an error because its value would be fixed, totally different from the current behaviour. So to avoid odd bugs and force everybody to rewrite their code it would need to be at least a warning.

In @jordanbrown0's example:
```
$a = 5;
b = $a;

module foo() {
    echo(b=b);
}
```

If you pass a new value of `$a` on the call stack it will be seen inside `foo()` but `b` will no longer be the same value. At least the old behaviour was consistent. $a and b would be 5 regardless of what was passed but if you use the `if_undef` trick both `$` and `b` will get the dynamic value.

So basically you are making the language less powerful to make it easy to teach?

martinbudden, 2021-09-18:
@rcolyer , you haven't answered my previous comment addressed to you https://github.com/openscad/openscad/issues/3881#issuecomment-920848099

In particular

> So let's work on the assumption that the old behaviour was wrong and the new behaviour is correct. Even if that is the case, it doesn't mean that it is correct to change to the new behaviour. There is a body of code that relies on the old behaviour, and it is a non-trivial and error-prone exercise changing that code, see #3781 (comment) . A piece of software like OpenSCAD needs to ensure that new releases don't break old code, even if that involves maintaining "bug compatibility" (ie not fixing bugs because the bug is relied upon).
> ...
> OpenSCAD is a piece of software with a large body of users. It is important that when those users migrate to the next version their designs are not broken.

What is you view on addressing the fact the this change has broken one of the OpenSCAD libraries, and all projects that depend on that library?

My view is that a breaking change can be made, but it has to be made in a considered way, weighing the costs and benefits, and also done in a way that allows a managed migration to the change (cf the change from Python 2 to Python 3).

rcolyer, 2021-09-18:
> Well I would have to rewrite everything I have written in OpenSCAD for the last 10 years or so and can no longer set `$layer_height` at file scope to change an entire file full of parts.

If you want to change dynamically reactive default behavior, you don't edit $layer_height in a used module, you edit layer_height().  This does not break any ability, it just requires adjusting the way you were thinking about how to do this.

I don't think so much hyperbole is needed about this.  I edited the two variables which you indicated were the big issue, updated 115 lines, and it took about 30 minutes for the changes since it was very formulaic.  The amount of effort put into the issue reports about this has easily been more than 10x the effort to update the library to use standard dynamic scope.

> I would have to add it to all the functions and modules in that file, so when it is used and called from somewhere else then it uses the correct value. Hardly any point to dynamic variables then, I might as well pass more parameters with a default.

There is a ton of utility to dynamic variables, and they don't need to be thrown out.  Just, their utility is dynamic, not mixed dynamic/lexical.  Their utility actually increases when they work cleanly, just not for the specific (and apparently uncommon) pattern you were using.  That pattern is a functional pattern, and should be done with functions, getting the same result.

> And any file scope capture of a dynamic variable would need to be an error because its value would be fixed, totally different from the current behaviour. So to avoid odd bugs and force everybody to rewrite their code it would need to be at least a warning.

Using a dynamic variable in the top-level of a file is not an error, and should absolutely not have a warning.  This is a well-defined process, and is extremely important to the behavior of a main file.  The problem your library experienced was only a specific issue with attempting to get active dynamic behavior from the top-level of a used file.  Additionally, using a file should definitely not introduce new warnings based on parts of the code not being used for a call, when the manual and other learning material we give people have already described for a long time the current behavior of the master branch.

> In @jordanbrown0's example:
> 
> ```
> $a = 5;
> b = $a;
> 
> module foo() {
>     echo(b=b);
> }
> ```
> 
> If you pass a new value of `$a` on the call stack it will be seen inside `foo()` but `b` will no longer be the same value. At least the old behaviour was consistent. $a and b would be 5 regardless of what was passed but if you use the `if_undef` trick both `$a` and `b` will get the dynamic value.

b is a lexical scope declarative variable that should have one constant value.  $a is a dynamically scoped variable that should update based on the call chain.  The example quoted right there is actually an example of a hazardous bug from the old behavior, because the caller sets $a and the value is completely overwritten, without caller control, wiped out like it's a non-dynamic constant by something outside of the call chain.  If $a were used directly in foo() this would be very bad.  b, however, should stay constant, because it's not dynamic.  The person editing foo should not have to inspect how the constant lexical value of b was obtained in the outer scope to determine that it stays constant.

> So basically you are making the language less powerful to make it easy to teach?

It's not one drop less powerful.  I've demonstrated how to do every dynamic behavior with ease.  It's more clear, more debuggable, and more teachable, because lexical constants stay constant (and don't go surprise-dynamic), dynamics stay dynamic (and don't go surprise-constant), and active generation of dynamic values in a standardized manner is done by active function calls instead of by implicit function-like behavior that is not labeled as a function.  Everything can still be done, just everything actually does what it says on the label.  This is a very good thing.

> What is your view on addressing the fact the this change has broken one of the OpenSCAD libraries, and all projects that depend on that library?

My view is that it required minor changes to one library, and I showed that it is easy to fix that library, even providing a PR for the part that was highlighted as the biggest issue.  Notably, the change I demonstrated in the PR should not require changing projects that depend on that library via use<>.  The dynamic variables from the caller projects are still picked up in the same way during the function calls, so it's only an internal change in how behavior is made dynamic.

> My view is that a breaking change can be made, but it has to be made in a considered way, weighing the costs and benefits, and also done in a way that allows a managed migration to the change (cf the change from Python 2 to Python 3).

Yes.  And I agreed with the changes from Python 2 to Python 3 as well, even though this required much more massive code updates (some of which I had to do).  In weighing them, there are huge costs to the old use<> behavior in an accidental potentially-massive performance overhead and in the ability to read, reason about, and debug code, as behavior can sneak in as dynamic value changes from far away places, meaning even a tiny function or tiny module might actually have globally impacted dynamic behavior unless you check all over th [... cut ...] widespread issue, and I haven't seen that rely-on-reprocessing usage pattern elsewhere.  I think most users followed the guidance in the manual and other instructions for designing their code.  I tried to help with this particular library, by both explaining and submitting a library PR with a chunk of the work to do this update.  It should really not be very difficult for someone more familiar with the library to identify the remaining variables that need identical changes and slipping those in.

martinbudden, 2021-09-18:
@rcolyer , you state:

> My view is that it required minor changes to one library, and I showed that it is easy to fix that library, even providing a PR for the part that was highlighted as the biggest issue.

@nophead , is a smart guy, and I'm sure he has considered your proposal. If he says it doesn't work, then I am inclined to believe him. 

> Notably, the change I demonstrated in the PR should not require changing projects that depend on that library via use<>. 

The problem is is that is not how projects depend on the NopSCADlib library - projecst both `use` and `include` files, and in particular it is standard practice to `include` a project defined `global_defs.scad`. Certainly the current change requires all my NopSCAD projects to be modified.

You are a smart guy too, but you are not familiar with the NopSCADlib library, and have not used it for a large project. The fact of the matter is the library has been broken, as have the projects that depend on it.

As I said, we need a managed way forward. What that means is something that can be done on a timescale dictated by the users, not a forced immediate change when a user upgrades to a new release.

One example of this would be @jordanbrown0 's suggestion that `use` is deprecated and a new keyword is introduced. So anything that currently uses `use` would continue to work without any changes required, and the new behaviour would be associated with the new keyword.

nophead, 2021-09-18:
@rcolyer ,
  In OpenSCAD if a file scope variable depends on a dynamic variable it also becomes dynamic. By changing it to be a one time static snapshot it breaks everything. When you changed `layer_height` to `layer_height()` you failed to appreciate that now all file scope expressions that use it are broken if they are only calculated once. So they all have to become functions also. And so on all the way out to the project. The only way to trap this bug is to give a warning if a file scope variable captures a dynamic one because to get the same behaviour as before it has to be a function.

And I can no longer set $ variables in a used file to affect it all the files it uses. I would have to set it in all the functions and modules of that file that get called externally. So when used and called the correct values are enforced.

E.g. in bob_main.scad I set `$extrusion_width` and `$layer_height` to non-default values but they get ignored if the file is used. So I would have to set them in all the modules that get called.

So now I would have to write a lot more code to get the same functionality as before, so in my mind the language is less powerful.

rcolyer, 2021-09-19:
> you failed to appreciate that now all file scope expressions that use it are broken if they are only calculated once. So they all have to become functions also.

I did appreciate this need, and yes, that is the solution I recommend.  What I've been emphasizing repeatedly is that all centralized dynamic calculations should be functions.  In my own code I have functions calling functions calling functions.  It's a functional language.  The ability to do this sort of dynamic behavior got even more flexible since we got function literals.

> And I can no longer set $ variables in a used file to affect it and all the files it uses. I would have to set it in all the functions and modules of that file that get called externally. So when used and called the correct values are enforced.

Yes, this is the correct solution for this behavior.  It must be somewhere in the call chain, as these are call chain following values.

> So now I would have to write a lot more code to get the same functionality as before, so in my mind the language is less powerful.

It's actually not a lot more code.  In most cases it will be very little code, like taking away $ and adding () plus the word "function", or adding an assignment statement.  Also, the only rule is that it needs to be in the call chain.  It doesn't necessarily need to be repetitive.  A random example of a solution option (there are many other ways to do this sort of thing):

main.scad:
```
use <used.scad>
  
$a = "main_a";
$b = "main_b";

Foo();
Bar();
```

used.scad:
```
module UsedDefaults() {
  $a = is_undef($a) ? "used_a" : $a;
  $b = is_undef($b) ? "used_b" : $b;
  $c = is_undef($c) ? "used_c" : $c;
  $d = is_undef($d) ? "used_d" : $d;
  children();
}

module Foo() { UsedDefaults() _Foo(); }
module _Foo() {
  echo($a=$a, $b=$b, $c=$c, $d=$d);
}

module Bar() { UsedDefaults() _Bar(); }
module _Bar() {
  echo($a=$a, $b=$b, $c=$c, $d=$d);
}
```

That's pretty simple and clean already.  But I think we can make things even cleaner with structured binding, which is not implemented yet, but that we've talked about with some decent interest.  That idea would look something like [$a, $b, $c, $d] = DefaultValues();  One could still have a module like the above that then applies this automatically, giving two ways to explicitly grab the defaults of dynamic variables.

The advantage of these approaches is that they are robust, consistent, and well controlled in a manner that has nothing to do with specific files, include order, use order, file reprocessing, or distant code that is not specified in the call chain.  The price is a few extra characters, and typing a few extra key strokes takes language users a lot less time than hunting down bugs due to distant surprise effects.  (The problem with the use file reprocessing is that those variable side-insertions into the scope potentially affect every user of the language whether they're trying to use it or not.  It's not about one library which used it.  It's about many thousands of users who can't figure out why their code is doing wacky things.)  This is why it is extremely common in modern language design to favor syntax with a few extra key strokes for clarity regarding what behaviors are invoked over the most text-compact possible way to write statements.  A good language should set a goal of specific behavioral features for the language at a high level (e.g., centralized ways to dynamically calculate values), and then look for a way to provide those which is consistent, explicit, and locally apparent what the resulting behavior will be.

nophead, 2021-09-19:
So I have to wrap all my functions and module, bonkers and ugly. And while I can hide the _functions using function literals I can't hide the _modules.

Looks I backed the wrong horse with OpenSCAD all those years ago. I didn't imagine the semantics of how variables would work would suddenly change. I will use the last release and look around for something else that is better designed and won't suddenly  change in a non-backwards compatible way.

Python3 was a nightmare and I only updated when 2 went end of life. But the changes to my code were not ugly like this. The new code is just as clean as the cold code and possibly simpler.

There have been other breaking changes to OpenScad but it has been fairly trivial to fix this library and in most or perhaps all cases the library changed were internal. There have also been breaking changes to this library but these have been simple renames and changes of default parameters that are trivial to update in projects.

I won't be releasing a version where all projects have to replace a lot of their variables with functions with no way to ensure it is done all the way out.

rcolyer, 2021-09-19:
I acknowledge the intensity of the responses has gotten a bit high, but I'm not here for that, and I don't think it will be productive.  I'm here in this discussion to provide a perspective on design rationale and to advise on implementation solutions.

At the top level, while we would generally all prefer never having a breaking behavioral change impacting our code, in the broader sense none of us would want to work with a language which promised to never alter implementation behavior.  If you truly need that for a time, you can version lock.  Implementation mistakes with behavioral impacts are made all the time, and they need to be fixed, or the implementation decays into chaos.  Many of the more mature languages deal with this partially with a formal specification.  We don't have the labor to have produced a formal one, but instead we work with an informal set of specified features arising from written documentation, labels for things, and collective discussions.  But even with a formal specification, mistakes are made in the specification resulting in some breaking changes being rolled out to address it.  OpenSCAD is written in C++, and the C++ specification has had quite a number of breaking changes roll out with each new version.  I maintain C++ libraries and code, and I've had to make changes to my C++ code to deal with these issues.  So even the people trying really hard to minimize this cannot avoid it, while also achieving the goals of progress and cleanliness.  I don't think it's a reasonable ask of any language under development.  The more reasonable issues in front of us are what, when, and how to deal with them.  An example of a productive solution in this area was @t-paul 's proposal to setup a library testing suite system so that we can flag the fixes that result in behavioral changes early.  Some will still be needed, because that's fairly fundamental.  But knowing about it lets us plan and address them ahead of time (potentially with concurrent changes) rather than retroactively.

> And while I can hide the _functions using function literals I can't hide the _modules.

That was a solution option, not a requirement.  We currently have no notion of private modules, like Python which has no proper notion of private, so for the example I borrowed from Python the leading underscore convention to indicate a private implementation detail.  If you don't like it, there are many other options, e.g.:

```
module Foo() {
  UsedDefaults() {
    echo($a=$a, $b=$b, $c=$c, $d=$d);
  }
}
```

Note that this was a suggested option in lieu of using functions to dynamically evaluate values.  If what you desire is evaluating dynamic variables in a centralized manner upon entering a set of modules, you can also address the need with this approach, which specifies to do exactly that, rather than the function one if you prefer.  I can't tell you what will exactly meet your whole set of desired behaviors, because you never listed off the whole set.  But there are many flexible ways to do these things which I'm trying to provide as options.

> I won't be releasing a version where all projects have to replace a lot of their variables with functions with no way to ensure it is done all the way out.

I'm not sure why you've identified this as required "all the way out", but I believe all of the examples I showed were designed to work with no changes required to external files that use<> the library.  If your users have been instructed to wedge their usage somewhere more in the middle then maybe there is a web of dependencies that are more challenging.  It might still be solvable with no external changes, but that would be more difficult to reason through.  If that's the case, and after reflecting on it you do decide to make future updates, it might help to define an API boundary for external usage so that you can free up more opportunities to change internals of the library without affecting the external usage.

nophead, 2021-09-19:
You changed `extrusion_width to extrusion_width()` so that affects all uses in a project. Not only does it clash with an existing name, it means all expressions that define a wall width using `extrusion_width` now need to be functions. And any derived expressions also need to be made into functions. That is what I mean by "all the way out". It is quite likely the outside dimensions of a project might change slightly with the `extrusion_ width` and `layer_height`. But it might vary between components in the project. This was all handled elegantly by OpenSCAD. 

And now instead of being able to simply set `$extrusion_width` at the top of a used file and have it affect everything in that file I now have to inject into the call stack of all entries into the file. I find this ugly and I much prefer the way variables have worked for the the last 10 years. I have always assumed the language was designed to work like that, and I like it.

martinbudden, 2021-09-19:
Summary of suggestions so far.

Looking through the discussions, I think there are 4 suggestions for moving forward:

1. Leave the codebase as is currently is and require dependent libraries (NopSCADlib, but possibly others as well) and dependent projects to migrate. One suggested migration mechanism is to use functions instead of special variables (@rcolyer )
2. Add some sort of pragma feature on file level This can be tricky but hopefully doable, it needs a clear definition about how that extra lookup is added. It's not possible to "just go back to the broken version". This option would be a new feature trying to add backward compatibility for the given use cases. It's unlikely that it can be 100% bug-to-bug compatible. (@t-paul )
3. perhaps "use" might be deprecated in favor of still another inclusion mechanism with better semantics. ( @jordanbrown0 )
4. add a command line flag to control the behaviour ( @martinbudden )

Possibilities that have been rejected by @t-paul:

5. Using the "experimental" feature. This is not an option as this is a) global to the application b) defined by the user not the library author and c) not available in releases.
6. "just revert the bugfix".

Of course some of these suggestions may be impossible or impractical to implement.

@t-paul , from the submission history, l presume you are the shot-caller on the project. Do you have any thoughts about / objections to any of the suggestions?

t-paul, 2021-09-20:
@martinbudden no objections, but I'm not sure affecting each and every existing script containing `use<>` is a good idea. That said I guess technically it's a valid option for discussion.

nophead, 2021-09-20:
Why change anything, just restore what was broken by the refactoring?

I don't remember anybody saying special variable shouldn't have lexical scope. They accidentally lost it and then it was argued that it was more correct and should always have been like it is now. But that is just somebody's opinion. 

It has always been the case that you could set $variables at the top of a file to affect everything in that file. That now doesn't work if the file is used and instead you have to set the new value in every module and function that can be called externally. That is ugly and far less convenient.  Why can't we have it working as before?

MichaelAtOz, 2021-09-20:
Perhaps it is a good time for me to jump in again.
It is 08:30am here, so I just need to do a few things, then I'll add my thoughts for what they're worth.

>  Why can't we have it working as before?

So perhaps we can leave that question hanging for a few hours.

jordanbrown0, 2021-09-20:
My instinctive reaction is that $ variables do not, and have never, had lexical scope, per se.

They have only dynamic scope.  The reason that (in 2021.01) they work, sort of, in "used" files is that the file level of the "used" file appears to be considered to be a layer of the call stack, as if a call into a module (or function) in the file was first a call to the top level of the file, and then a call to the actual target module.

That is, their behavior is intimately tied to the "evaluate the file level on every call into the file" behavior.

---

Here's a thought experiment.  What output does this produce?  What output should it produce?
```
// 1.scad
use <2.scad>
foo();
bar();
```
```
// 2.scad
$a = rands(0,1,1)[0];
module foo() {
    bar();
    bar();
}
module bar() {
    echo($a=$a);
}
```

For extra fun, merge the two files and compare the results.

MichaelAtOz, 2021-09-21:
@jordanbrown0 thanks for that side track.

I like instrumenting.

```
// jb#3881-2.scad the file to be use<>'d

// not usual to use random $fn, used just to show re-evaluation
$fn = let(x = 10+floor(rands(0,1,1)[0]*50) ) 
      echo(" used-$fn-assign:\t",x) x; 
n   = let(i=$fn) echo(" used-n-assign:\t",i) i;

calls(" used-file", $fn,n);
foo();

module foo() {
    calls(" used-foo", $fn,n);
    bar();
    bar();
}
module bar() {
    calls(" used-bar", $fn,n);
    echo($fn=$fn,n=n);
}

module calls(location=undef,f,n) {
  calls = [ for(p = [$parent_modules-1 : -1 : 0]) parent_module(p) ];
  echo(str(  " Calls: <", location = location, ">")
           , f=f,n=n, " from:", calls );
}
```

```
// jb#3881-1.scad
use <jb#3881-2.scad>
calls("main",$fn,n);  // unknown n obviously
echo("#1");
foo();
echo("#2");
bar();
echo("========");
echo("#3");
foo($fn=100);
echo("#4");
bar($fn=200);
```
Using `rand()` could be a bit ingenuous as it is a side-effect. However it does allow introspection.

With `jb#3881-2.scad` by itself is internally consistent in both versions.
```
ECHO: " used-$fn-assign:	", 17
ECHO: " used-n-assign:	", 17
ECHO: " Calls: < used-file>", f = 17, n = 17, " from:", ["calls"]
ECHO: " Calls: < used-foo>", f = 17, n = 17, " from:", ["foo", "calls"]
ECHO: " Calls: < used-bar>", f = 17, n = 17, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 17, n = 17
ECHO: " Calls: < used-bar>", f = 17, n = 17, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 17, n = 17
```

With `jb#3881-1.scad` with `2021.01` it behaves in a fashion I have expected, where the `use<>' provides defaults.
```
ECHO: " used-$fn-assign:	", 56
ECHO: " used-n-assign:	", 56
WARNING: Ignoring unknown variable 'n' in file jb#3881-1.scad, line 3 
ECHO: " Calls: <main>", f = 56, n = undef, " from:", ["calls"]
ECHO: "#1"
ECHO: " used-$fn-assign:	", 58
ECHO: " used-n-assign:	", 58
ECHO: " Calls: < used-foo>", f = 58, n = 58, " from:", ["foo", "calls"]
ECHO: " Calls: < used-bar>", f = 58, n = 58, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 58, n = 58
ECHO: " Calls: < used-bar>", f = 58, n = 58, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 58, n = 58
ECHO: "#2"
ECHO: " used-$fn-assign:	", 56
ECHO: " used-n-assign:	", 56
ECHO: " Calls: < used-bar>", f = 56, n = 56, " from:", ["bar", "calls"]
ECHO: $fn = 56, n = 56
ECHO: "========"
ECHO: "#3"
ECHO: " used-$fn-assign:	", 42
ECHO: " used-n-assign:	", 42
ECHO: " Calls: < used-foo>", f = 100, n = 42, " from:", ["foo", "calls"]
ECHO: " Calls: < used-bar>", f = 100, n = 42, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 100, n = 42
ECHO: " Calls: < used-bar>", f = 100, n = 42, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 100, n = 42
ECHO: "#4"
ECHO: " used-$fn-assign:	", 37
ECHO: " used-n-assign:	", 37
ECHO: " Calls: < used-bar>", f = 200, n = 37, " from:", ["bar", "calls"]
ECHO: $fn = 200, n = 37
```
and the default can be overridden, $fn=100 & 200 above, but not the five re-evaluations of the assigns.
As `rand()` is involved, I would expect differences.
With a carefully chosen `rand()` expression, it will produce a variety of results as intended.

In `2021.07.29`

```
ECHO: " used-$fn-assign:	", 41
ECHO: " used-n-assign:	", 41
WARNING: Ignoring unknown variable 'n' in file jb#3881-1.scad, line 3
ECHO: " Calls: <main>", f = 0, n = undef, " from:", ["calls"]
ECHO: "#1"
ECHO: " used-$fn-assign:	", 52
ECHO: " used-n-assign:	", 52
ECHO: " Calls: < used-foo>", f = 0, n = 52, " from:", ["foo", "calls"]
ECHO: " Calls: < used-bar>", f = 0, n = 52, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 0, n = 52
ECHO: " Calls: < used-bar>", f = 0, n = 52, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 0, n = 52
ECHO: "#2"
ECHO: " used-$fn-assign:	", 53
ECHO: " used-n-assign:	", 53
ECHO: " Calls: < used-bar>", f = 0, n = 53, " from:", ["bar", "calls"]
ECHO: $fn = 0, n = 53
ECHO: "========"
ECHO: "#3"
ECHO: " used-$fn-assign:	", 44
ECHO: " used-n-assign:	", 44
ECHO: " Calls: < used-foo>", f = 100, n = 44, " from:", ["foo", "calls"]
ECHO: " Calls: < used-bar>", f = 100, n = 44, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 100, n = 44
ECHO: " Calls: < used-bar>", f = 100, n = 44, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 100, n = 44
ECHO: "#4"
ECHO: " used-$fn-assign:	", 32
ECHO: " used-n-assign:	", 32
ECHO: " Calls: < used-bar>", f = 200, n = 32, " from:", ["bar", "calls"]
ECHO: $fn = 200, n = 32
```
Note the re-evaluations and the lose of defaults.

That just consumed a couple of hours.

The above is not the thoughts I mentioned earlier, I'll work on that this afternoon.

rcolyer, 2021-09-21:
> where the `use<>' provides defaults

@MichaelAtOz  But it does not, it clobbers/overwrites.  I've added one more line at the very end of your jb#3881-1.scad.  My challenge to you is to reason out why the two bottom lines are different from each other in 2021.01 (but fine in the master branch).  Hint:  Order of operations in the old process.

```
// jb#3881-1.scad
use <jb#3881-2.scad>
calls("main",$fn,n);  // unknown n obviously
echo("#1");
foo();
echo("#2");
bar();
echo("========");
echo("#3");
foo($fn=100);
echo("#4");
bar($fn=200);
{$fn=200; bar();}
```

> My instinctive reaction is that $ variables do not, and have never, had lexical scope, per se.
>
> They have only dynamic scope. The reason that (in 2021.01) they work, sort of, in "used" files is that [...]

@jordanbrown0 Yes, your description there nicely and accurately summarizes everything, both in the way to think and talk about it, and in describing what was actually happening in the code.

martinbudden, 2021-09-21:
> Why change anything, just restore what was broken by the refactoring?

My apologies, that should also have been stated as an option. I'd over-interpreted @t-paul 's statement that he did not want to "just revert the bugfix" (ie revert PR https://github.com/openscad/openscad/pull/3745) to mean that he did not want to go back to the old behaviour.

@t-paul , can you confirm that it would be acceptable to re-instate the old behaviour as long this doesn't mean reverting PR https://github.com/openscad/openscad/pull/3745? If so I'll update my summary list.

nophead, 2021-09-21:
Yes the scope has always been dynamic but the call stack included the used file's file scope variables. That is what has changed. And the file scope variables have to be in the search order because `b=42; function a() = b;` must find `b` when it is called and it used to find `$b` defined alongside but now it doesn't.

It isn't anything to do with the file scope variables needing to be re-evaluated. They still are and need to be because any expression using a file scope variable becomes dynamic. If you do` a = $a` then `a`'s value is dynamic and visible to a function or module in the same file but `$a` is invisible. The dynamic expression should be marked and only those re-evaluated when they are used to prevent the massive speed hit.

They do have to be dynamic and the @rcolyer suggestion is to make them all functions. But then any expression that uses them has to be a function and so on so you get exactly the same dynamic expressions being re-evaluated on every call.  Just done in user land instead of in OpenSCAD. The difference is you don't get all the non dynamic constant expressions evaluated and you don't get the expressions that aren't used on that call evaluated. But that optimisation could be done inside OpenSCAD and then it would be backwards compatible.

MichaelAtOz, 2021-09-21:
> where the `use<>' provides defaults

Defaults to the modules/functions of `// jb#3881-2.scad the file to be use<>'d`.

> it clobbers/overwrites. I've added one more line

Instead I used:
```
{
  echo("#5");
  $fn= let(j=300) echo("@main-$fn-assign:\t", j) j;  //300 so you can distinguish from the other 200 call
  bar();
}
```
I presume you know that `{}` does not create a new scope in that instance.

As per my history in the OP.

comments in [REOPENED Can't override include variables](https://github.com/openscad/openscad/issues/289) (Mar 2013)
@kintel 
>Regarding overriding variables: Variables in a USE'd file are hidden and you cannot override them. When writing a library with configurable parts, pass these as parameters to modules.

(as per my calls in sections \#3 & \#4 of the calling file)

@kintel 
>
>If you want to change something, you can fall back to include. 

comments in [Feature request: Enums / Enumerations](https://github.com/openscad/openscad/issues/1032) (comments Apr 2016)
@kintel
>    @doug-moen It's not a bug - it's a workaround for not having namespaces. This makes all top-level values local the the use'd file.

That IS WHAT IT WAS DESIGNED TO DO. To have a self contained package.
If you want to do it the other way code `include<>` instead of `use<>`.

Why would you code a `use<>` otherwise.

MichaelAtOz, 2021-09-21:
For the record with
```
// jb#3881-1.scad
echo(version());
use <jb#3881-2.scad>
calls("main",$fn,n);  // unknown n obviously
echo("#1");
foo();
echo("#2");
bar();
echo("========");
echo("#3");
foo($fn=100);
echo("#4");
bar($fn=200);
{
  echo("#5");
  $fn= let(j=300) echo("@main-$fn-assign:\t", j) j;
  bar();
}
```

`2021.01`
```
ECHO: "@main-$fn-assign:	", 300
ECHO: [2021, 1, 0]
ECHO: " used-$fn-assign:	", 14
ECHO: " used-n-assign:	", 14
WARNING: Ignoring unknown variable 'n' in file jb#3881-1.scad, line 4 
ECHO: " Calls: <main>", f = 14, n = undef, " from:", ["calls"]
ECHO: "#1"
ECHO: " used-$fn-assign:	", 55
ECHO: " used-n-assign:	", 55
ECHO: " Calls: < used-foo>", f = 55, n = 55, " from:", ["foo", "calls"]
ECHO: " Calls: < used-bar>", f = 55, n = 55, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 55, n = 55
ECHO: " Calls: < used-bar>", f = 55, n = 55, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 55, n = 55
ECHO: "#2"
ECHO: " used-$fn-assign:	", 55
ECHO: " used-n-assign:	", 55
ECHO: " Calls: < used-bar>", f = 55, n = 55, " from:", ["bar", "calls"]
ECHO: $fn = 55, n = 55
ECHO: "========"
ECHO: "#3"
ECHO: " used-$fn-assign:	", 24
ECHO: " used-n-assign:	", 24
ECHO: " Calls: < used-foo>", f = 100, n = 24, " from:", ["foo", "calls"]
ECHO: " Calls: < used-bar>", f = 100, n = 24, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 100, n = 24
ECHO: " Calls: < used-bar>", f = 100, n = 24, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 100, n = 24
ECHO: "#4"
ECHO: " used-$fn-assign:	", 39
ECHO: " used-n-assign:	", 39
ECHO: " Calls: < used-bar>", f = 200, n = 39, " from:", ["bar", "calls"]
ECHO: $fn = 200, n = 39
ECHO: "#5"
ECHO: " used-$fn-assign:	", 38
ECHO: " used-n-assign:	", 38
ECHO: " Calls: < used-bar>", f = 38, n = 38, " from:", ["bar", "calls"]
ECHO: $fn = 38, n = 38
```

With `2021.07.29`
```
ECHO: "@main-$fn-assign:	", 300
ECHO: [2021, 7, 29]
ECHO: " used-$fn-assign:	", 59
ECHO: " used-n-assign:	", 59
WARNING: Ignoring unknown variable 'n' in file jb#3881-1.scad, line 4
ECHO: " Calls: <main>", f = 300, n = undef, " from:", ["calls"]
ECHO: "#1"
ECHO: " used-$fn-assign:	", 52
ECHO: " used-n-assign:	", 52
ECHO: " Calls: < used-foo>", f = 300, n = 52, " from:", ["foo", "calls"]
ECHO: " Calls: < used-bar>", f = 300, n = 52, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 300, n = 52
ECHO: " Calls: < used-bar>", f = 300, n = 52, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 300, n = 52
ECHO: "#2"
ECHO: " used-$fn-assign:	", 57
ECHO: " used-n-assign:	", 57
ECHO: " Calls: < used-bar>", f = 300, n = 57, " from:", ["bar", "calls"]
ECHO: $fn = 300, n = 57
ECHO: "========"
ECHO: "#3"
ECHO: " used-$fn-assign:	", 44
ECHO: " used-n-assign:	", 44
ECHO: " Calls: < used-foo>", f = 100, n = 44, " from:", ["foo", "calls"]
ECHO: " Calls: < used-bar>", f = 100, n = 44, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 100, n = 44
ECHO: " Calls: < used-bar>", f = 100, n = 44, " from:", ["foo", "bar", "calls"]
ECHO: $fn = 100, n = 44
ECHO: "#4"
ECHO: " used-$fn-assign:	", 40
ECHO: " used-n-assign:	", 40
ECHO: " Calls: < used-bar>", f = 200, n = 40, " from:", ["bar", "calls"]
ECHO: $fn = 200, n = 40
ECHO: "#5"
ECHO: " used-$fn-assign:	", 23
ECHO: " used-n-assign:	", 23
ECHO: " Calls: < used-bar>", f = 300, n = 23, " from:", ["bar", "calls"]
ECHO: $fn = 300, n = 23
```

nophead, 2021-09-21:
>Why would you code a use<> otherwise.

I use `use` to prevent all the expressions and objects defined in the file being public. and because used file are cached, whereas included files are reparsed. My framework can't include a file because it would instantiate its assembly view.

I do customise library files by passing parameters for most aspects but not certain fundamental ones like layer height and not ones that say whether I am making a BOM or an exploded view.

MichaelAtOz, 2021-09-21:
> prevent all the expressions and objects defined in the file being public

Sorry, yes that was presumed as part of `self contained package`.

nophead, 2021-09-21:
Also it allows global options that are backwards compatible. For example I just added a `$plugs` variable that makes all PCBs show plugs on their headers, so I can see if there is room for them. It would be a complete pain for users if they became a new parameter to `pcb()` as they would have to pass it all the way up the call chain.

[image]

MichaelAtOz, 2021-09-21:
...and that still wasn't my thoughts that I mentioned earlier...not sure that they were now...looks like that's a job for tomorrow. 
Getting dark here.

MichaelAtOz, 2021-09-21:
For posterity.
My posts today were in regard to the side-track mentioned above, but I intentionally framed it as $fn (could also have been $fa/$fs too) as they can be important defaults in context of a self contained package. They get ignored with the new behaviour.

I was not covering the bigger picture that @nophead just added a couple of examples.

jordanbrown0, 2021-09-21:
In case it wasn't clear, the point of my thought experiment was to demonstrate the slightly odd effects that can come from the fact that the file level is evaluated when you call into the file, not when modules in the file are called, so that it's evaluated once for the call to foo(), and not for the two calls to bar() from foo(), and then evaluated again for the call to bar().  And when you put them all in one file, the file level is evaluated only once.

rcolyer, 2021-09-21:
> But then any expression that uses them has to be a function and so on so you get exactly the same dynamic expressions being re-evaluated on every call.

This is almost never the case.  The only scenario in which this is true is if every single function and module which is externally called in a used file requires access to every single value which is obtained by either a dynamically set variable or a dynamically determined function.  I'm pretty sure that in the vast majority of files designed for use<> out there, individual functions and modules make use of only a small fraction of set values.

To test my hypothesis, I decided to not rely on my own code, and pick the biggest library I could think of, so I googled BOSL and started rummaging through the source code.  I saw nearly every single use<>d file having 83+ assignments which were getting re-evaluated on EVERY function and module call, even though the vast majority of the functions and modules used zero or only a handful of these.  This imposes a ridiculous overhead, especially when you consider that many BOSL functions are used in tight loops as part of comprehensions.

At that point I realized I was looking at BOSL and I remembered there was a BOSL2, so I went to the BOSL2 repository.  To my surprise, every use was replaced with an include, littering the namespace with internal variable names.  So I searched for why.  I found revarbat's explanation, "If you look, BOSL2 dropped all use of `use` due to OpenSCAD bugs."  So essentially, revarbat found himself forced to litter the namespace with variable names that he didn't want there, because he found OpenSCAD's use<> unusably buggy with its terrible performance for libraries making use of functions and large numbers of variables.  This is precisely what we're trying to fix.

This is a functional language where we are trying to broaden the ability to do creative and computationally intensive designs with increased functional features, and we ran right up against an issue where things run potentially hundreds or thousands of times slower than they need to because of this use<> behavior.  The issue requesting this to be fixed was opened in 2014, but it didn't matter to what people were doing for a long time.  But now people have more powerful computers and are trying to do more intense things, and this reprocessing performance issue matters a lot.

I find myself sticking to three principles:
1) Variables should follow the scope on the label.
2) Calling a function or module should go to the function or module, like it says on the label.
3) Performance should be good, and not have orders of magnitude overhead penalties.

nophead, 2021-09-21:
>This is almost never the case. 

You misunderstand my point entirely. What I am saying is the uses of dynamic variables need to be reevaluated. Either OpenSCAD does it for me or I have to do do it myself with functions calling functions calling functions all the way up to the outer dimensions of my objects. 

But on entry to a module only the dynamic expressions need re-evaluating and only if they are used and only if the dynamic variables have changed since the last call into the module. So if that is fixed the overhead would be tiny.  Files full of maths functions probably don't use any dynamic variables at all, so they would never need to be recalculated.

And in the vast majority of projects dynamic might only have one value and only a small number of expressions use them. But those that do must be recalculated, when they change and are referenced.

jordanbrown0, 2021-09-22:
I don't think that there's any way to bring these two viewpoints together.

That leaves a couple of possibilities:
(A) Ignore nophead's concerns and charge on.
(B) Give up on getting rid of the performance hit from "use" re-evaluating file-level values on calls into the file.
(C) Use a pragma (on the called-file side) or a new keyword (on the calling-file side) to switch behaviors.
(D) Massively overhaul expression processing so that a file-level assignment is evaluated only when the variable involved is accessed.  Or perhaps only if it (transitively) depends on a dynamic-scope variable.

(A) and (B) are unappealing.

(D) requires a developer who is intimately familiar with the expression evaluator infrastructure, and carries substantial risk.  Is there any developer interested in spending that effort?

I think that leaves (C), which is still non-trivial development but probably doesn't involve a massive overhaul, just setting up options as to when the assignments are evaluated and how the call stack is built.  It would also require figuring out what the semantics of the new style should be.

MichaelAtOz, 2021-09-22:
It is still re-evaluating.

```
// testA#3881-main.scad
echo(str("=== ",version()," =============================") ); // help in console view
echo();
main=true;  // flag for testA#3881-used 

use <testA#3881-used.scad>

calls("main",$fn,undef);  

normal = trk("main - normal",1);
used_simple("main");
used_simple2("main"); 
used_fn("main");
used_simple_dollar("main"); 
```

```
// testA#3881-used.scad - file to be use<>'d

// not usual to use random $fn, used just to show re-evaluation
$fn = trk(" used - $fn", 10+floor(rands(0,1,1)[0]*50) ); 
n   = trk(" used - n",$fn);     // assignment with $var
i   = trk(" used - i",1);       // simple var
j   = trk(" used - j",i + 1);   // simple evaluation
$v  = trk(" used - $v","dollarv");

calls(" used-file", $fn,n);
used_simple("used");

function trk(t,v) // track - expose assignments. t=text,v=value
         = echo(str(" trk:   <",t,"> \tassigned:"),v) v;

module calls(location=undef,f,n) {
  calls = [ for(p = [$parent_modules-1 : -1 : 0]) parent_module(p) ];
  echo(str(  " Calls: <", location = location, ">")
           , f=f,n=n, " from:", calls );
}

module used_simple(t) {
    echo(t, parent_module(0), i );
    avar = i;
}

module used_simple2(t) {
    echo(t, parent_module(0), j );
    avar = j;
}

module used_fn(t) {
    echo(t, parent_module(0), $fn );
    avar = $fn;
}

module used_simple_dollar(t) {
    echo(t, parent_module(0), $v );
    avar = $v;
}
```

Results.
```
Compiling design (CSG Tree generation)...
ECHO: " trk:   < used - $fn> 	assigned:", 42
ECHO: " trk:   < used - n> 	assigned:", 42
ECHO: " trk:   < used - i> 	assigned:", 1
ECHO: " trk:   < used - j> 	assigned:", 2
ECHO: " trk:   < used - $v> 	assigned:", "dollarv"
ECHO: " trk:   <main - normal> 	assigned:", 1
ECHO: "=== [2021, 7, 29] ============================="
ECHO: 
ECHO: " trk:   < used - $fn> 	assigned:", 55
ECHO: " trk:   < used - n> 	assigned:", 55
ECHO: " trk:   < used - i> 	assigned:", 1
ECHO: " trk:   < used - j> 	assigned:", 2
ECHO: " trk:   < used - $v> 	assigned:", "dollarv"
ECHO: " Calls: <main>", f = 0, n = undef, " from:", ["calls"]
ECHO: " trk:   < used - $fn> 	assigned:", 20
ECHO: " trk:   < used - n> 	assigned:", 20
ECHO: " trk:   < used - i> 	assigned:", 1
ECHO: " trk:   < used - j> 	assigned:", 2
ECHO: " trk:   < used - $v> 	assigned:", "dollarv"
ECHO: "main", "used_simple", 1
ECHO: " trk:   < used - $fn> 	assigned:", 24
ECHO: " trk:   < used - n> 	assigned:", 24
ECHO: " trk:   < used - i> 	assigned:", 1
ECHO: " trk:   < used - j> 	assigned:", 2
ECHO: " trk:   < used - $v> 	assigned:", "dollarv"
ECHO: "main", "used_simple2", 2
ECHO: " trk:   < used - $fn> 	assigned:", 42
ECHO: " trk:   < used - n> 	assigned:", 42
ECHO: " trk:   < used - i> 	assigned:", 1
ECHO: " trk:   < used - j> 	assigned:", 2
ECHO: " trk:   < used - $v> 	assigned:", "dollarv"
ECHO: "main", "used_fn", 0
ECHO: " trk:   < used - $fn> 	assigned:", 34
ECHO: " trk:   < used - n> 	assigned:", 34
ECHO: " trk:   < used - i> 	assigned:", 1
ECHO: " trk:   < used - j> 	assigned:", 2
ECHO: " trk:   < used - $v> 	assigned:", "dollarv"
WARNING: Ignoring unknown variable '$v' in file testA#3881-used.scad, line 40
WARNING: Ignoring unknown variable '$v' in file testA#3881-used.scad, line 39
ECHO: "main", "used_simple_dollar", undef
Compiling design (CSG Products generation)...
```

MichaelAtOz, 2021-09-22:
As you can see, `n` is getting the new value of `$fn`.
Interestingly `$fn` in `used_fn()` doesn't get the warning. Presumably it is $special rather than $dymanic.

jordanbrown0, 2021-09-22:
Yes, I know that it's still doing that every-call-into-the-file evaluation.  I'm actually puzzled by the current-snapshot behavior, from a black-box perspective, because it changed the semantics *without* addressing that issue.  But that doesn't really matter, because that issue, the "evaluating file-level assignments on calls into a file" issue, is one that several people are interested in and should be assumed to be on the horizon - and (absent the major overhaul in (D) above) it will require changes much like this one, that will break nophead's use cases.

If we abandon that issue, if we decide that we're not going to fix the performance hit, then (without looking at the actual implementation) I expect that restoring the old semantics would be practical.

But if we address that issue (without the overhaul in (D)), nophead's library will break.

rcolyer, 2021-09-22:
> Yes, I know that it's still doing that every-call-into-the-file evaluation. I'm actually puzzled by the current-snapshot behavior, from a black-box perspective, because it changed the semantics without addressing that issue.

Yeah, one step at a time.  I think fixing that performance issue is one of the important steps before the next release.  I was actually thinking about starting to look at that issue closer tonight, but this issue keeps chiming away with notifications.  :)

> (C) Use a pragma (on the called-file side) or a new keyword (on the calling-file side) to switch behaviors.
> I think that leaves (C), which is still non-trivial development but probably doesn't involve a massive overhaul, just setting up options as to when the assignments are evaluated and how the call stack is built. It would also require figuring out what the semantics of the new style should be.

Well, I think a keyword is the more OpenSCAD type of solution to it.  First, let me propose a keyword possibility to declare a context that will be fully evaluated before entering called modules:

```
use_context {
  $foo = 5;
  $bar = is_undef($bar) ? 6 : $bar;
}

module Foo() {
  echo($foo=$foo, $bar=$bar);
}
```

The behavior of the above example would be that "use_context" is evaluated before entering the modules, so all the dynamically scoped variables are set as desired.  In this tiny example, it looks great, and solves the first problem.

There is one critical issue with it but that we can solve, which is that in large projects it leaves potential serious debugging confusion, because if ever you are confused about how a dynamically scoped value is changing, you have to suddenly search every included file in the used file to see if there is a use_context statement hiding in it.  (For example, it can come 3 include layers deep from someone ELSE's library that you happen to have included instead of used.)  That's a confusion risk that we should avoid, and that we can avoid easily, by simply making sure the keyword is applied near the modules that will use that specific context.  So basically, we want the use_context keyword right next to the module, so that it's clear they will use that context.  Here's an example of how to do that:

```
module use_context() {
  $foo = 5;
  $bar = is_undef($bar) ? 6 : $bar;
  children();
}

module Foo() { use_context() {
  echo($foo=$foo, $bar=$bar);
}}
```

This already works in every version of OpenSCAD, so please feel free to use it and get the desired behavior.

nophead, 2021-09-22:
I really don't understand why the language has to change to fix the re-evaluation issue.

If you do `a = 4` that only needs to be evaluated once. If you do `b = $b` then that has to be reevaluated if `$b` changes and b is referenced, so mark it as such.

If you do `c = a` that only needs to be evaluated once because a is static. If you do `d = b`  then d inherits the fact it depends on `$b`. 

It isn't rocket science but you do need to mark functions that depend on dynamic variables as well.

The suggestion above is ugly because instead of doing `$fn = 500` at the top of your file you have to inject it into the call stack of all the module and functions that get called externally and it has an overhead.

If dynamic expressions are only evaluated when they need to be in C++ that is more efficient than simulating them in OpenSCAD userland.

MichaelAtOz, 2021-09-22:
> I don't think that there's any way to bring these two viewpoints together.

Looks that way.

`use<>` was intentionally developed to provide something in the absence of namespace/class/public/private/packaging etc. mechanisms. It was pretty much a sledge hammer solution.

It hides complexity by looking after the mechanics under the hood. That's why we automate things.
I does not require what I'm calling syntactic sourness; things causing the writer to implement (comparatively) complex mechanisms which could otherwise done automatically and need to kept track of.

A key aspect of OpenSCAD that I try to maintain is its simplicity.
Compare what [other languages require](https://rosettacode.org/wiki/Draw_a_sphere#Openscad) to draw a sphere.
Yet it still allows advanced users to do more sophisticated projects using higher level techniques.
Having everything in a single file package (which usually needs some of the $fX's) to `use<>` is simplicity.
That is one of the first steps new users go to with more complex projects.

I turns out that its current form creates performance issues with more expansive implementations.
That is not necessarily the reason to dispense with `use<>` ease of use. 
Particularly in the absence of a proper class/packaging mechanism.

An additional option to add to Jordan's options list:

(E) [possibly a variant of (C) depending on how you read it]
Leave `use<>` to do the dynamic mechanics under the hood*.
Introduce a *pure* version ([pick a name](https://rosettacode.org/wiki/Include_a_file) perhaps `load`, `requires`) for those wanting to do it manually, old-school.
Plan for a next evolution into class like mechanisms.

\*Later if resources are available, consider something like what nophead just proposed or build a dependency tree to only re-evaluate variables with a dynamic dependency.

If there is recognition that something is needed for backwards compatibility, e.g. a pragma, then I don't see debate in circles is needed any further. It seem it is ideological.

nophead, 2021-09-22:
The disadvantage of creating a new way of loading modules or adding pragmas is that anybody wanting improver performance would need to use it and then their code needs the latest OpenSCAD. Keeping the language as it has always been and optimising the reevaluation means no old code will break, it will just get much faster and still be compatible with old versions of OpenSCAD and all the examples and books, etc.

[32 more replies]