openscad/openscad #3882: Pattern bug or something

Hateyaall, 2021-09-15:
I have this issue that I just can't resolve. I am basically making a pattern of a honeycomb foundation cell. But the mirrored pattern for some reason doesn't show the penultimate row. Is this some sort of bug or I have messed up my code. Any help will be greatly appreciated.

[image]

`$fn= 6;
a = 2.658;      // offset for the second row of cells to make the slope
y = 5.107;      // offset for the second row in Y direction
d1 = 5.897;     // spacing for the columns  
d2 = 10.213;    // vertical distance between cells
row = 28;       // rolls of cells
column = 37;    // column of cells 
cell_D = 6;     // cell diameter
cw = 5.897;     // cell width
cww = 1;        // cell wall width
bt = 2;         // base thickness
bbwt = 1;       // base bottom wall thickness
hct = 2;        // honeycomb thickness
hcw = 230;      // honeycomb width
hch = 316;      // honeycomb height
xw = 0.4;       // extrude width
m = 5-2.37-7;     // modifier for offsetting the cells so they fit with the honeycomb

fillet = 2;     // R2 fillet for the bottom of the foundation

fn_value = 100; //the value for the number of facets on holes 

m3_hole = 3.8;

union()
{
difference()
{
difference()    //cutting out hexagons from the foundation to be replaced with hexagons with the weird shape cut out
{
    frame();
        
//top side
        cut_out_cells();
    
    
    
      
    translate( [ 0, -286, -5 ] )
    rotate( [ 180 , 0 , 0 ] )
    cut_out_cells();
    
}

}

rectangular_array( column, row, d1,-d2 ) 

difference()        //cutting strange hexagons row 1
{
translate( [ 0, 0, 0 ] )
translate( [ 0, m, 2 ] )
rotate( [ 0, 0, 90 ] )
cylinder(  h = 5, d = cell_D, center = true );
        
translate( [ 0, m, 3.51] )
rotate( [0,0,30] )
cylinder(h=bt-bbwt+xw/.4, d1=0, d2=cw/cos(65)-cww/2-xw, $fn=3, center = true);

}

rectangular_array( column-1, row+1, d1,-d2 ) 

difference()        //cutting strange hexagons row 2
{
translate( [ -d1, 0, 2 ] )
translate( [ d1 + ( d1 / 2 ) , y+m, 0 ] )
rotate( [ 0, 0, 90 ] )
cylinder(  h = 5, d = cell_D, center = true ); //7.9 for ~6.9 from side to side cell
        
       
translate( [ -d1, 0, 0 ] )
translate( [ d1 + ( d1 / 2 ) , y+m, 0] )
translate( [ 0, 0, 3.51] )
rotate( [0,0,30] )
cylinder(h=bt-bbwt+xw/.4, d1=0, d2=cw/cos(65)-cww/2-xw, $fn=3, center = true);
}

rectangular_array( column, row, d1,-d2 ) 
translate( [ 0, 0, -2.3] )
rotate( [ 180, 0, 0 ] )
difference()        //cutting strange hexagons row 1
{
translate( [ 0, 0, 0 ] )
translate( [ 0, m, 2 ] )
rotate( [ 0, 0, 90 ] )
cylinder(  h = 5, d = cell_D, center = true );
        
translate( [ 0, m, 3.51] )
rotate( [0,0,30] )
cylinder(h=bt-bbwt+xw/.4, d1=0, d2=cw/cos(65)-cww/2-xw, $fn=3, center = true);

}

rectangular_array( column-1, row+1, d1,-d2 ) 
translate( [ 0, 0, -2.3] )
rotate( [ 180, 0, 0 ] )
difference()        //cutting strange hexagons row 2
{
translate( [ -d1, 0, 2 ] )
translate( [ d1 + ( d1 / 2 ) , y+m, 0 ] )
rotate( [ 0, 0, 90 ] )
cylinder(  h = 5, d = cell_D, center = true ); //7.9 for ~6.9 from side to side cell
        
       
translate( [ -d1, 0, 0 ] )
translate( [ d1 + ( d1 / 2 ) , y+m, 0] )
translate( [ 0, 0, 3.51] )
rotate( [0,0,30] )
cylinder(h=bt-bbwt+xw/.4, d1=0, d2=cw/cos(65)-cww/2-xw, $fn=3, center = true);
}

}

module cut_out_cells()
{
    rectangular_array( column, row, d1,-d2 ) 

translate( [ 0, 0, 0 ] )
translate( [ 0, m, 2 ] )
rotate( [ 0, 0, 90 ] )
cylinder(  h = 5, d = cell_D, center = true );
        
        
    rectangular_array( column-1, row+1, d1,-d2 ) 

translate( [ -d1, 0, 2 ] )
translate( [ d1 + ( d1 / 2 ) , y+m, 0.001 ] )
rotate( [ 0, 0, 90 ] )
cylinder(  h = 5, d = cell_D, center = true ); //7.9 for ~6.9 from side to side cell
}

module frame()
{
        minkowski()
        {
color( [ 255/255, 0/255, 0/255 ] )
translate( [ 106 , -138, -1.15] )
cube( [ 226, 315, 11 ], center = true ); //compensated dimentions for the R2 fillet on the next line
           cylinder( h = 0.0001, d = 4, $fn = fn_value, center = true);
        }
    }

module rectangular_array( rows, colum, distance1, distance2 )  
                //pattern row 
{
        for( k = [ 0: 1: rows - 1 ] )
        {
          for( j = [ 0: 1: colum - 1 ]) // start value,step value, end value
                {
                 translate( [ distance1 * k, distance2 * j ] )
                        children();
                }
        }
        
}  

`

MichaelAtOz, 2021-09-15:
Well that just drove my two system into quagmire.

Please cut it down in size to a more manageable model still showing the problem.

 

Also attach the .scad as a file, as that will preserve formatting.

 

 

  _____  

 

 <https://user-images.githubusercontent.com/67191565/133387514-3fc65f41-8272-4b3a-b1ff-7e34f59662dc.jpg> InkedScreenshot 2021-09-15 095853_LI

 

This email has been checked for viruses by AVG.
https://www.avg.com

rcolyer, 2021-09-15:
@Hateyaall I cannot reproduce the reported bug.  I copied and pasted this code, and it both previewed correctly and rendered correctly, with no artifact like shown in the included image.  If the provided code is identical to what you used to make the image, perhaps you have some edge-effect preview artifact going on, based on a subtlety of how your display driver is working?  I don't have a solid explanation for the report, I just see it didn't reproduce.  Does preview look funny but render turn out okay for you in this code?

t-paul, 2021-09-15:
If it's preview only, it could be the "Turn of rendering at X elements" preferences being too low.

thehans, 2021-09-15:
Its a script logic error and not a bug in OpenSCAD.  line 84 requires `row+1` to fix it.  Also its only visible on the underside of the object.

closed 2021-09-15 by thehans