openscad/openscad #3885: Blank screen error.

44knight44, 2021-09-16:
I am using OpenSCAD version 2021.09.09.ci8596 (git 227325893) . When I right click the view screen around the middle the window turns black. I found this error by accident. Lucky the menu still works so I can save my work. Just to let you know what I found.

nophead, 2021-09-16:
Yes I get the same problem on Win7 64. I have to quit and start again each time I accidentally right click when dragging.

Right click shows the call stack that drew the pixel you clicked on but it has broken after the last release. If I resize the black window I get a broken view drawn in it when it is repainted.

[image]

nophead, 2021-09-21:
Interestingly if I do copy viewport image the result is correct.  It seems that the windows are drawn correctly but displayed with a black mask over the top with transparent holes in it. In the holes the window contents are shown correctly and menus are drawn correctly over the top.

MichaelAtOz, 2021-09-21:
It may be unrelated, but I have been unable to run snapshots for a long while, #3719 with windows display issues, WITH S/W RENDERING. But I suspect it points to some underlying problem.

PRouzeau, 2021-09-30:
See also  #3725

t-paul, 2021-09-30:
Thanks, this is indeed a duplicate of  #3725.

closed 2021-09-30 by t-paul