openscad/openscad #3889: Disable or even better modify this effect I don't right now know the name of

etkaar, 2021-09-18:
Unfortuantely I don't know how to call that, but I am pretty sure you know why I mean: There is an effect if you difference() for instance a 5 mm thick cube with a 5 mm thick cylinder (difference of 0.0):

[image]

This effect makes sense, it is very helpful. But sometimes it is very aggressive - is there any way to modify this, lower the opacity or something like that? Or even disable it?

t-paul, 2021-09-18:
See [this FAQ entry](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview?).

closed 2021-09-18 by t-paul

etkaar, 2021-09-18:
I know this workaround – is/was there no plan to integrate an option for that?

t-paul, 2021-09-18:
At this point, there's no known general solution that could be implemented in the application core. There might be more specialized options with user space libraries.

etkaar, 2021-09-18:
So, this is a bug (which is helpful though), but it is still a bug and therefore one would need to put into much effort to enhance that, am I correct?

Scopeuk, 2021-09-18:
This is a common edge case which people run into. Where a user defines a shape that shares a surface with another shape and performs a difference they generate a zero thickness surface. The rendering engine is unable to determine if this space is filled or not as there exists a shell there but it has a thickness of zero. This is mathematically completely sound and what was defined but does not translate well to the "real world".
This is a mathematically reality which causes one part of the model to be both positive and negative space. This is almost certainly not what they user wishes to describe where in the surface would be expected to have a hole showing through it. This is why the FAQ and examples recommend subtracted objects protrude beyond the end surface.

rcolyer, 2021-09-18:
As the FAQ explains, it's not a bug, it's a fundamental property of math.  While certain cases could in principle be automatically handled (giving the false sense that the right thing to do should be obvious for the program), in general this is an unsolvable problem because the correct result is not clearly specified mathematically.  Designs which lack clear overlap for differences in the general case have undefined mathematical representation and might completely fail to render at the render step.  The solution path is to fix any scad code that has these invalid coincident faces, which is explained in the FAQ and again in the manual under "difference" and "union".

etkaar, 2021-09-18:
Thanks for the explanations! It can be helpful anyway.