openscad/openscad #3890: difference() with no effect

etkaar, 2021-09-18:
The hexagons and cylinders (hole) are not cut out from the cube in that example. While it works on the long side, it does not on the short side. Am I just doing it wrong?

Please see line 103–154 when putting my code into OpenSCAD.

**Preview ([F5])**

[image]

**Rendering ([F6])**

[image]

**Full Code**

```
BASE_THICKNESS = 3;
HOLDER_THICKNESS = 3.75;

BOX_LENGTH = 100;
BOX_WIDTH = 70;
BOX_HEIGHT = 51;

BOLTS_DMTR = 3 + 0.4;
BOLTS_HEAD_HEIGHT = 1.7;

THREADED_NUTS_HEIGHT = 2.4 + 0.2;
THREADED_NUTS_DMTR = 5.5 + 0.8;

HOLDER_HEIGHT = THREADED_NUTS_DMTR * 1.5;

/**
*	Hole diameters
*/
MUTE_BUTTON_HOLE_DMTR = 22 + 0.5;

XLR_IN_HOLE_DMTR = 21.6 + 0.5;
XLR_IN_RECESS_SIZE = 26.5;

XLR_OUT_HOLE_DMTR = 21.6 + 0.5;
XLR_OUT_RECESS_SIZE = 26.5;

DC5V_HOLE_DMTR = 7.8 + 0.5;
DC5V_RECESS_SIZE = 12.5;

/**
*	Enable recesses for XLR and DC5V jacks
*/
DRAW_JACK_RECESSES = true;

/**
*	Top plate
*/
difference() {
	cube([BOX_LENGTH, BOX_WIDTH, BASE_THICKNESS]);

	// Hole for the big red button
	translate([(BOX_LENGTH / 2 + 10), (BOX_WIDTH / 2), 0])
	cylinder(h=BASE_THICKNESS, d=MUTE_BUTTON_HOLE_DMTR, center=false);
}

/**
*	Long holders with holes (fitted with threaded nuts) to hold the side plates
*/
for(s = [0, 1]) {
	difference() {
		_x = (BASE_THICKNESS + HOLDER_THICKNESS);
		_y = (s == 0)
			? BASE_THICKNESS
			: (BOX_WIDTH - BASE_THICKNESS - HOLDER_THICKNESS);
		_z = BASE_THICKNESS;
		
		_translation = [_x, _y, _z];
		_length = (BOX_LENGTH - 2 * BASE_THICKNESS - 2 * HOLDER_THICKNESS);
		
		translate(_translation)
		cube([_length, HOLDER_THICKNESS, HOLDER_HEIGHT]);

		// Recesses for the holders
		for(n = [0, 1]) {
			_x = (THREADED_NUTS_DMTR + BASE_THICKNESS + HOLDER_THICKNESS);
			_y = BASE_THICKNESS + HOLDER_THICKNESS;
			_z = THREADED_NUTS_DMTR / 2 + BASE_THICKNESS + ((HOLDER_HEIGHT - THREADED_NUTS_DMTR) / 2);
			
			_translation = [
				(n == 0)
					? _x
					: (BOX_LENGTH - _x),
				(s == 0)
					? _y
					: (_y + (BOX_WIDTH - (2 * BASE_THICKNESS + 2 * HOLDER_THICKNESS))),
				_z
			];
			
			_rotation = [
				(s == 0)
					? 90
					: 270,
				0,
				0
			];
			
			translate(_translation) {
				rotate(_rotation) {
					// Hexagon for fitting the threaded nut
					cylinder($fn=6, h=THREADED_NUTS_HEIGHT, d=THREADED_NUTS_DMTR, center=false);
					
					// Hole for the bolt
					cylinder($fn=20, h=HOLDER_THICKNESS, d=BOLTS_DMTR, center=false);
				}
			}
		}
	}
}

/**
*	Short holders
*/
for(s = [0, 1]) {
	difference() {
		_x = (s == 0)
			? (BASE_THICKNESS + HOLDER_THICKNESS)
			: (BOX_LENGTH - BASE_THICKNESS);
		_y = BASE_THICKNESS;
		_z = BASE_THICKNESS;
		
		_translation = [_x, _y, _z];
		_rotation = [0, 0, 90];
		_length = (BOX_WIDTH - 2 * BASE_THICKNESS);
		
		translate(_translation)
		rotate(_rotation)
		#cube([_length, HOLDER_THICKNESS, HOLDER_HEIGHT]);

		// Recesses for the holders
		for(n = [0, 1]) {
			_x = (s == 0)
				? (BOX_LENGTH - BASE_THICKNESS - HOLDER_THICKNESS)
				: (BASE_THICKNESS + HOLDER_THICKNESS);
			_y = BASE_THICKNESS + HOLDER_THICKNESS + THREADED_NUTS_DMTR;
			_z = THREADED_NUTS_DMTR / 2 + BASE_THICKNESS + ((HOLDER_HEIGHT - THREADED_NUTS_DMTR) / 2);
			
			_translation = [
				_x,
				(n == 0)
					? (_y)
					: (BOX_WIDTH - _y),
				_z
			];
			
			_rotation = [
				0,
				(s == 0)
					? 90
					: 270,
				0
			];
			
			translate(_translation) {
				rotate(_rotation) {
					// Hexagon for fitting the threaded nut
					#cylinder($fn=6, h=THREADED_NUTS_HEIGHT, d=THREADED_NUTS_DMTR, center=false);
					
					// Hole for the bolt
					#cylinder($fn=20, h=2 * HOLDER_THICKNESS, d=BOLTS_DMTR, center=false);
				}
			}
		}
	}
}

/**
*	Side plate with XLR and DCV5 jacks
*/
translate([(-1.1) * (BOX_HEIGHT), 0, BASE_THICKNESS]) {
	rotate([0, 90, 0]) {
		difference() {
			cube([BASE_THICKNESS, BOX_WIDTH, BOX_HEIGHT]);

			rotate([90, 0, 90]) {
				// XLR > IN
				translate([(XLR_IN_HOLE_DMTR / 2) + 9, (BOX_HEIGHT / 2) - 2, 0]) {
					cylinder(h=BASE_THICKNESS, d=XLR_IN_HOLE_DMTR, center=false);
					
					// Small recess for the XRL jack
					if (DRAW_JACK_RECESSES) {
						_depth = 1.0;
						translate([0, 0, BASE_THICKNESS + (_depth / 2) - _depth])
						cube([XLR_IN_RECESS_SIZE, XLR_IN_RECESS_SIZE, _depth], center=true);
					}
					
					// XLR assembling holes
					for(i = [1, -1]) {
						translate([i * (19.8 / 2), i * (19.8 / 2), 0])
						cylinder(h=BASE_THICKNESS, d=1.5, center=false);
					}
				}

				// XLR > OUT
				translate([(BOX_WIDTH - (XLR_OUT_HOLE_DMTR / 2) - 9), (BOX_HEIGHT / 2) - 2, 0]) {
					cylinder(h=BASE_THICKNESS, d=XLR_OUT_HOLE_DMTR, center=false);
					
					// Small recess for the XRL jack
					if (DRAW_JACK_RECESSES) {
						_depth = 1.0;
						translate([0, 0, BASE_THICKNESS + (_depth / 2) - _depth])
						cube([XLR_OUT_RECESS_SIZE, XLR_OUT_RECESS_SIZE, _depth], center=true);
					}
					
					// XLR assembling holes
					for(i = [1, -1]) {
						translate([i * (19.8 / 2), i * (19.8 / 2), 0])
						cylinder(h=BASE_THICKNESS, d=1.5, center=false);
					}
				}
				
				// Hole for DC5V
				translate([BOX_WIDTH / 2 + 2, BOX_HEIGHT - 12, 0]) {
					cylinder(h=BASE_THICKNESS, d=DC5V_HOLE_DMTR, center=false);
					
					// Small recess for the DC5V
					if (DRAW_JACK_RECESSES) {
						_depth = 1.0;
						translate([0, 0, BASE_THICKNESS + (_depth / 2) - _depth])
						cube([DC5V_RECESS_SIZE, DC5V_RECESS_SIZE, _depth], center=true);
					}
				}
			}
		}
	}
}

```

rcolyer, 2021-09-18:
This design is failing the overlap requirement in a large number of places, which can result in rendering failures exactly like you are seeing.  Read the "Note:" section under "union", and then read the "Note:" section under "difference" right below it:  https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/The_OpenSCAD_Language#union

etkaar, 2021-09-18:
Thanks for your fast response. I did try letting it overlap already which can be seen in the preview. Unfortunately no sucess:

[image]

MichaelAtOz, 2021-09-18:
The problem is you have miss-positioned the bolts, in the loop you make one bar near y=0, but you try to difference bolts at near y=100.
[image]

closed 2021-09-18 by MichaelAtOz

etkaar, 2021-09-19:
@MichaelAtOz Good catch, thanks a lot! Perfectly working

[image]