openscad/openscad #3894: openscad version in ubuntu is ancient

ronsmits, 2021-09-21:
According to the downloads page openscad can be installed from the ubuntu repository. This version however is 2019.1. This is quite ancient. Maybe make it more clear on the download page that appImage should be used instead of the deb version?

rcolyer, 2021-09-21:
Ubuntu 21.04 has the latest OpenSCAD 2020.01 release in its package list, but Ubuntu 20.04LTS has 2019.05 available.  This matches expectations given the release dates of these two active support Ubuntu versions.

closed 2021-09-21 by rcolyer

t-paul, 2021-09-21:
Yes, also I tried keeping up with that kind of hints on the website and that's just not possible. The status of external stuff like that changes all the time, e.g. the PPA is maintained by someone else, so any of those suggestions might still be out of date and make things worse.