openscad/openscad #3895: Allow writing out only Echo text to file with user selected extension

WillAdams, 2021-09-22:
To allow folks to do B.O.M.s, and assembly instructions, and possibly even CAM, and to remove the need for post-processing the log file add a checkbox/preference which enables 

Write out echo() text (only) to file when Rendering object using the file extension: .csv/.txt/.log/.nc

t-paul, 2021-09-22:
This would be a different export format, maybe defaulting to `.txt` but as always the user has control over the file name.

WillAdams, 2021-09-22:
I'd like for there to at least be an option to generate this automatically when rendering, with the same filename as the OpenSCAD file, and a user-defined extension --- saves on the need to manually do the export.

jordanbrown0, 2021-09-22:
Or perhaps a different verb, e.g. `print(a,b,c)` instead of `echo(a,b,c)`.

I was assuming that this was primarily useful from the CLI.  If it was desirable from the GUI too, what I'd recommend would be that the program capture the echo stream for the most recent render, and have a "Save log" or "Export text" option that would save that output.  The GUI doesn't write any output files automatically, and shouldn't start.

jordanbrown0, 2021-09-22:
I did some work on printf() support that might be relevant.  That support was based on a C++ next-generation-printf library (prf?), but I eventually found myself in a maze of templates, unable to figure out how to get it to accept an OpenSCAD number as the argument to ones of the "size" parameters.  If I were to do it again, I think I'd do a dedicated OpenSCAD implementation that natively supports OpenSCAD types.

PRouzeau, 2021-09-30:
see also #2036   (closed)

WillAdams, 2021-10-10:
A different verb/command would work well.

WillAdams, 2021-12-13:
Apparently there was a patch for this submitted a while back:

https://github.com/openscad/openscad/pull/845

Is that still applicable?

t-paul, 2021-12-13:
Not sure what applicable means specifically. The referenced PR implements it as part of the language which raised too many concerns to be used in that exact form.