openscad/openscad #3898: Preview showing different result when looking from z<0 and z>0

phhusson, 2021-09-25:
With the following code:
```
        difference() {
            rotate([90, 0, 0]) cylinder(h=4, r = 2.2, center=true);
            rotate([90, 0, 0]) cylinder(h=6, r = 2, center=true);
            rotate([180, 0, 0]) translate([-50, -50, 0]) cube([100, 100, 100]);
        }
```

When previewing this from z<0, it looks like the last element isn't taken in account and we see a full circle cylinder, while when previewing from z>0, it is the half cylinder as expected, and as actually rendered.

This is with Debian's openscad 2021.01-1

rcolyer, 2021-09-25:
I cannot reproduce the reported issue in 2021.01 or the master branch.  2021.01:

From top:
[image]
From bottom:
[image]
Angled view:
[image]

phhusson, 2021-09-25:
Ok, a new information about this bug is that it depends on zoom level.

Video link of the issue, in case there is some ambiguity as to what I do:
https://phh.freeboxos.fr:38503/share/V1hDM7_SXyv6LIOv/simplescreenrecorder-2021-09-25_17.47.48.mkv

I'll compile master to see if i reproduce there.

t-paul, 2021-09-25:
`That happens if you zoom in far enough to end up inside the big cube differenced. That's tied to how preview works (in perspective mode) and I'm not sure there's a way to change that behavior.`

rcolyer, 2021-09-25:
Oh, yeah.  I can reproduce this with perspective mode on.  To fix, click "View", and "Orthogonal", and leave it there.  It's more reliable, and more convenient for design.  I'm not sure if this is the default or not right now, but it probably should be.

jordanbrown0, 2021-09-25:
I don't have a strong opinion on whether orthogonal or perspective should be the default, but I do think that the setting should persist across restarts.

(And my weak opinion is that perspective is the better default, because to the naïve viewer things will generally look more right.)

jordanbrown0, 2021-09-25:
But I hadn't realized that the "don't let the camera be inside the invisible cube" problem was only in perspective mode.

t-paul, 2021-09-27:
Closing the issue as there's not much that can be done at this point. If anyone has suggestions please point to possible solutions and reopen.

closed 2021-09-27 by t-paul