openscad/openscad #3902: GitHub 'Discussions'

MichaelAtOz, 2021-09-26:
@dinther let the cat out of the bag on the discussion Mailing-list:

https://docs.github.com/en/discussions

I haven't played with it yet, I want to see if it is a potential replacement for the Forum.
If so, whether we want to have the wider community that close to the developer community.  
If a separate repository may allow a degree of isolation if such is desired.
etc.

Separately, if it should be used for developer discussions instead of issues which have been used for such here.
(don't see why not, but I haven't played yet)

t-paul, 2021-09-26:
Sure, we can have a look. Shall I create a test repo to have a look or directly enable on the main repo? I'm ok either way.

MichaelAtOz, 2021-09-26:
I'd suggest a test, so it doesn't turn into a precedent (build it and they will come) which is hard to reverse.

MichaelAtOz, 2021-09-26:
Perhaps the test should be the place for the discussion of how/if/who it should be used, rather than this issue.

t-paul, 2021-09-26:
Ok, lets give it a try... https://github.com/openscad/discussion-test-repo/discussions/1

closed 2021-09-26 by t-paul