openscad/openscad #3903: linear_extrude with difference results in unwanted transparency

erichstuder, 2021-09-26:
Hello

I've looked up the other issues but haven't found any about this problem so far.

The following code shows a simplification of my problem.
If I do a linear_extrude first and then a difference then the extruded part becomes partly transparent in the preview (F5).
The rendering (F6) however looks fine.

I'm aware that there are most of the times workarounds, as for example in my code.
But in my case I have a library that generates a peace with linear extrude.
Later I want to cut a whole in I it with difference.
So I don't really have a workaround if I don't want to modify the library.

Is this a bug?

I'm using OpenSCAD version 2021.01 on Debian 11.

The example:
```
height = 20;

//the bad one
color("red")
	difference(){
		linear_extrude(height){
			square([70, 2]);
			rotate([0, 0, 5])
				square([70, 2]);
		}
		#cylinder(d=5, h=height+2);
	}

//the good one
color("green")
	translate([0, 10, 0])
		difference(){
			union(){
				cube([70, 2, height]);
				rotate([0, 0, 5])
					cube([70, 2, height]);
			}
			#cylinder(d=5, h=height+2);
		}
```
[image]

nophead, 2021-09-26:
You need to set the convexity parameter in linear_extrude.

erichstuder, 2021-09-26:
This solved the problem. Thank you!

But...
1. Because the problem only showed up after I applied 'difference()', I would never have thought of this solution.
2. According to the manual I expected that I had to set a convexity of 4 for my example, but 2 seems to work fine.

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_the_2D_Subsystem
[image]

Maybe this is something for future improvement...

Many Thanks.

nophead, 2021-09-26:
The manual example is wrong. The ray crosses two front faces so it has a convexity of 2, not 4. You can either count front or back face crossing but not both. Otherwise the minimum convexity for a closed shape would be 2 making the default convexity of 1 not make sense.

Yes if there isn't a CSG operation after the linear_extrude then convexity doesn't matter. It just produces a mesh that is drawn. If you have a difference or intersection then OpenCSG does some tricks to show the result without actually computing the new shape. To do that it has to know the convexity.

jordanbrown0, 2021-09-26:
We should really change the default convexity to 10 or so.  I don't think anybody has shown any significant problem with 10, and it would eliminate most of these troubles.

nophead, 2021-09-26:
I wonder why it isn't needed when you make a concave object in 3D. For example if you subtract a sphere from a cube you don't need to specify the convexity if you do another 3D operation on it but if you import an STL of the same you do. Does the 3D system keep track of convexity, in which case can the 2D substem do the same?

kintel, 2024-04-01:
Closing as not really an issue

closed 2024-04-01 by kintel