openscad/openscad #3907: Icon for macOS Big Sur

MohanMaker, 2021-09-29:
macOS Big Sur introduced a new icon style (square icons instead of round icons). It would be great if OpenSCAD had a new icon to agree with this design scheme. I know some individual users have already created their own versions of a square icon here: https://macosicons.com

TheJKM, 2021-10-05:
The only file that has to be updated is `icons/OpenSCAD.icns`, this container format is only used on macOS. This will change the icon for all versions of macOS, as Apple writes:

> IMPORTANT
> When you update your app for macOS 11, use your new app icon design to replace the icon you designed for earlier versions. You can’t include two different app icons for one app, and the macOS 11 app icon style looks fine on a Mac running Catalina or earlier.
(https://developer.apple.com/design/human-interface-guidelines/macos/icons-and-images/app-icon/)

I'm not a fan of the linked icon, I'd prefer the current icon on a white square to differ the icon from the background. In the linked version, they become blurred. An Example:

[image]

TheJKM, 2021-12-30:
Now that the icon is merged, I think we can close this.

t-paul, 2021-12-30:
Closed via #3970.

closed 2021-12-30 by t-paul