openscad/openscad #3908: echo() should not include any decoration - extra text, punctuation or formatting

dannypike, 2021-09-30:
This ticket is opened in response to [#3905](https://github.com/openscad/openscad/issues/3905) which started as a discussion about adding a setting to clear the console on each build and migrated to a discussion between @t-paul, @MichaelAtOz and @jordanbrown0, about having more control over the text that is output to the log by the ```echo()``` statement.

My current position is that the ```echo()``` command should not output any text other than that which is specified in the arguments to that call. No "ECHO:" header, no commas, no double quotes, just the arguments. It should continue to support a "suitable" subset of ANSI escape sequences, such as the colours and newlines.

So this set of statements:

```
echo("\r\n   cone_narrow_od:", cone_narrow_od
, "\r\n   cone_wide_od:", cone_wide_od
, "\r\n   cone_length:", cone_length
, "\r\n");
```

which currently produces this output:

```
ECHO: "
   cone_narrow_od:", 16.4, "
   cone_wide_od:", 32, "
   cone_length:", 88.472, "
"
```

would then produce this output:

```

   cone_narrow_od: 16.4
   cone_wide_od: 32
   cone_length: 88.472
```

(Note the blank line at the start of the block.)

rcolyer, 2021-09-30:
@dannypike Does it match your use-case thinking that this is only important for command-line execution?  The "-o output.echo" already captures only echo, plus warnings, it's just that it includes the "ECHO: ".  The rest is already obtainable with the following syntax, utilizing str:

```
cone_narrow_od = 16.4;
cone_wide_od = 32;
cone_length = 88.472;
echo(str("\r\n   cone_narrow_od: ", cone_narrow_od
, "\r\n   cone_wide_od: ", cone_wide_od
, "\r\n   cone_length: ", cone_length
, "\r\n"));
```
Output of:
```
ECHO: "
   cone_narrow_od: 16.4
   cone_wide_od: 32
   cone_length: 88.472
"
```

I could certainly see value in a clean echo command-line flag which additionally purges the "ECHO: " and the quote marks.

The operative question that should be discussed in concert, is what should happen with warnings and errors?  Ignoring them falls under "very bad idea", leaving a few options like:
- Bundle them even with clean echo, and it's the scad implementer's responsibility to clean up warnings and errors if output without them is desired.  (Which feels pretty reasonable to me.)
- Provide an alternate output for warnings and errors such as to stderr, and provide no capture to file support.
- Separate them, but complicate the command-line interface with additional flags to capture warnings and errors to a file.

jordanbrown0, 2021-09-30:
Normal output should go to standard out; errors should go to standard error.

(It is blurry what bucket normal progress reporting messages go into, but probably better standard error - or have a way to silence them.)

It is not necessary to provide additional options to redirect errors; all command processors (even including CMD!) provide ways to redirect standard error.

Neither errors nor normal progress reporting should go to a "-o xxx.echo" file; only explicit output from the OpenSCAD program should go there.

I would **not** recommend ANSI sequences, though I don't know if I'd explicitly prevent their use.  Note that CMD does not process them.  For the GUI, better would be to restore some support (perhaps via a different verb) for emitting HTML.  For CLI, is there really an important use case?

t-paul, 2021-09-30:
As someone already mentioned in the linked issue `echo()` will not change, there's no point in discussing a major change in behavior. The new feature will be a different function/module with a different parameter set providing the needed features (`print()`?).

GUI will not do HTML, it's a very bad fit for multi line formatting based on an internal unknown subset Qt widgets support.

So the question is basically how we can support the formatting part, as OpenSCAD is C++ and there's a clear winner now from C++ standardization, available for older environments via the `fmt` library. As this supports custom extensions, it should be possible to add this to OpenSCAD without inventing new formatting patterns. We can't prevent people outputting whatever control characters they like, but there probably needs to be one supported - maybe a bit limited - way for doing coloring that works for both GUI and console.

jordanbrown0, 2021-09-30:
I have a partial implementation of a `format()` function based on the fmt library.  I haven't looked at it in a while and it certainly needs to be updated to match recent changes.  My final stumbling block was in getting OpenSCAD numbers to work as formatting parameters.  (I don't remember what they look like in fmt-ese, but in printf-ese I'm talking about the `*` values in `%*.*d`.)  I got lost in a maze of templates and couldn't figure out how to get it to accept floating-point-esque numbers in that context.

If I was to do it again, I would probably implement fmt syntax but do an OpenSCAD-native implementation.

I don't know if a separate verb is really needed for formatted output.  If we had a `print` that just printed values (bare strings, maybe bare numbers, maybe other types) and we had a format() function that did fmt/printf processing, that might be enough.

Special effects (HTML, ANSI codes) probably belong in another issue.

dannypike, 2021-09-30:
@rcolyer: I was not aware of the ```str()``` function, though I do now see that it is mentioned in the help text for ```echo()```. The technique you described above would very effectively separate the ```echo()``` "noise" from the "signal" for me. Thank you for that.

@jordanbrown0 and @t-paul: I want to be able to record automatically all of the settings that were used to build a specific STL file, when the model can accept a range of different input values (sizes, options etc). So ```fmt()``` formatting instead of ANSI would be fine for me, if that makes more sense from your point of view.

Colours are less interesting at the moment, though I do see that it might be sensible to support them. I tend to look at and react to errors in real time using the GUI when I'm building a new STL file. They are transient, so I'm not interested in recording them in log files, and they don't really need to be formatted nicely.

I am, however, keen that it should support embedded tabs and newlines from a single statement.

jordanbrown0, 2021-09-30:
I don't have a really strong feeling on newlines embedded in strings.  I don't think I would use them, because I generally find that style to be no less ugly than just using multiple `print` statements.

Given the overall context, that this is *not* a general-purpose language, I would probably say that all print/echo operations should always append a newline, that there should be no way to display a partial line and then complete it later.  That's always a source of problems in other environments, and it doesn't seem necessary here.

Although I again probably would not explicitly prevent them, I would not encourage the use of tabs.  There is no consistency in how they are handled.  I would use spaces, perhaps assisted by library/language support for indentation and tables.

If this was a general-purpose language, where it was possible for random garbage to end up in a string or where the value of a string might be controlled by a villain, I would have the ordinary `print` mechanisms sanitize all control characters, and require the use of language/library features to achieve special effects, even including newline.  There are just too many issues associated with garbage data or malicious data being output by inadequately cautious programs; the language and library should arrange that special effects can be caused only through deliberate program action.  (Consider the behavior of `find` and `xargs` when presented with files with newlines in their names, or the chaos that often results from running `cat` on a binary file.)  HOWEVER, this is not a general-purpose environment, it's difficult to get random garbage into strings, and it's not clear how a villain would inject text, so maybe those precautions are not necessary.  HOWEVERHOWEVER, there are always pushes to add more general-purpose mechanisms (like reading files), and so having the precautions as early as possible might be best.  It's also an environment used heavily by beginning programmers, and control characters are tricky and an opportunity for error.

dannypike, 2021-09-30:
@jordanbrown0 Mine are possibly esoteric requirements, but I do have specific reasons to want them.

I agree with you that multiple print statements are the logical equivalent to embedded newlines, but I prefer to be able to "lock" related settings together by putting them all in one statement. It's a style that reminds me to keep them together, when I am cutting/copying/pasting them around the model.

Also, tabs allow me to import a combination of strings and values into a spreadsheet, when the raw strings are likely to contain commas. I never use tabs for anything other than delimiting fields in a table for the very valid reasons that you gave but, when I want them, nothing else works as well!

jordanbrown0, 2021-09-30:
You could always put blank lines or stylized comments around such blocks.

Export to spreadsheet... hmm.  My first thought is to arrange for it to be easy to produce properly quoted CSV files.  Perhaps, for instance, `print(array)` would print the array in proper CSV form.  (But we'd have to figure out whether there was an overall plan for how this hypothetical `print` verb handled various types - whether, for instance, it was defined to `str()` them to get their printable form.

But it would be pretty easy, even in OpenSCAD-language, to write a function `csv(string)` that would return a CSV-formatted version of the string, and then another module `printcsv(array)` that would dump the array as one line of CSV.  Maybe even if it contained nested arrays, though few tools would be able to do anything useful with those.

I've never tried to import tab-separated files into Excel.  CSV files it can just open; in fact, it's usually rigged to open `*.csv` files and `text/csv`.

jordanbrown0, 2021-10-01:
Here's a function csv(string) or csv(array) that returns a proper CSV form of the string or the array, suitable to import to Excel.  (It even handles nested arrays.)  All it needs is a decoration-less `print` verb.
```
function csv(v) = _csv_array(is_list(v) ? v : [v], 0);
function _csv1(v) =
    is_string(v) ? _csv_string(v)
    : is_num(v) ? str(v)
    : is_list(v) ? _csv_string(_csv_array(v, 0))
    : assert(false, "csv of unsupported type") 0;
function _csv_string(s) =
    _csv_needs_quote(s)
    ? str("\"", _csv_quote(s,0), "\"")
    : s;
function _csv_needs_quote(s) =
    len(search(",", s))>0 || len(search("\"", s)) > 0;

function _csv_quote(s, i) =
    i >= len(s)
    ? ""
    : str(s[i]=="\"" ? "\"\"" : s[i], _csv_quote(s, i+1));  
function _csv_array(a, i) =
    assert(len(a)>0)
    i == len(a)-1
    ? _csv1(a[i])
    : str(_csv1(a[i]), ",", _csv_array(a, i+1));

echo(csv(["hello", "world"]));
echo(csv(["a","b,c","d\"e",["f","g","h,i"]]));
```
There's probably a way to do a lot of it with list comprehensions instead of recursion, but I'm not as fluent there.

dannypike, 2021-10-01:
Wow, that's a lot of work you've done there. I hope you did it for fun, and not because you thought I needed an full-blown system to handle these cases. My vectors are mostly a few elements long (a label and three dimensions, for example) and I'm now quite happy to type them out using the ```str()``` function that **rcolyer** has just added to my repertoire.

Grouping the blocks with comments would help keep them together, but I just prefer having one statement. For me, it is a much more robust way to group things, as it'll produce a syntax error if I break the block and my style is to get the compiler to help me as much as possible.

I'm not a Luddite; I do adopt techniques that I like, but my personal style works for me, I do it automatically and almost subconsciously now. It takes zero effort for me to read, write and grok what I have created, even years later. It also has meant that I can recognise something as being my code when it goes wrong. That's saved my reputation more than once in the days before ```git blame```arrived on the scene and now it's a part of my coding soul. Habits are hard things to break after so many  decades of painful debugging!

So, with the aforementioned ```str()``` function, this output style is now working much better ...

```
for (ii = [["Fred, Joe & Sam plc", 101.909, 202.808, 303.707] /* , ... */])
  echo (str("\r\n\r\n",
    ii[0], "\t",
    ii[1], "\t",
    ii[2], "\t",
    ii[3], "\r\n\r\n"
    ));
```

... which produces this ...

```
ECHO: "
Client	x	y	z
"
ECHO: "
Fred, Joe & Sam plc	101.909	202.808	303.707
"
```

... which is much easier to edit using global search-replace on ```ECHO: "\r\n``` and ```"\r\n```.

If we can have a ```print()``` statement that doesn't output the ECHO: stuff, then that would be even better (for my specific case).

MichaelAtOz, 2021-10-01:
It would need to support Unicode.

Then complicated things arise, RTL,LTR...

 

  _____  

 

 

This email has been checked for viruses by AVG.
https://www.avg.com

dannypike, 2021-10-01:
Why would suppressing the ```ECHO: "``` and ```"``` delimiters add an extra requirement to support Unicode?

MichaelAtOz, 2021-10-01:
Well the title is now "Provide text output function without any decoration - extra text, punctuation or formatting" and talk was of `print()` etc. If it is just suppressing ECHO " etc, OK.
Note the console and `text()` support Unicode and we have users from many countries.

dannypike, 2021-10-01:
Ah, OK.

When I used the words "decoration" and "formatting" in the title, I meant characters that are auto-inserted by the ```echo()``` statement that I did not use in my model definition. Things like the ```ECHO:``` header, the ```"``` and the terminating newline (though I accept that some people want that to stay).

I wasn't talking about the non US-ASCII character tables (Unicode, Arabic, Cyrillic, Chinese etc). I have no experience of and therefore no opinion of how OpenSCAD should handle those.

jordanbrown0, 2021-10-01:
@dannypike I already had the basic logic from another project; I just had to translate it from JavaScript to OpenSCAD.  And yes, it was for fun, especially the part about handling nested arrays.

@MichaelAtOz I wouldn't say that this request interacts with Unicode support.  As Danny says, I consider the request to be only to have a way to emit text without the stuff that echo() adds.

With respect to embedded control characters in strings:  my draconian response based on trouble in other environments is to make it take special action for a program to emit any kind of "special effect" character, so that it won't do it by accident when processing untrusted data.  That draconian reaction may be overkill for the OpenSCAD environment.  A lesser reaction might be to have a small "allow list" of special-effect characters that are allowed - maybe just newline and tab.

With respect to newlines in particular:  My preference is to use separate calls rather than embedded \n, but if (as seems likely) the hypothetical `print()` module allows embedded newlines then one is free to use them.

With respect to tabs in particular:  Again, my preference is to avoid them.  For tabular data and for indenting (where one might think them valuable), there is just too much inconsistency and misbehavior in how they are handled.  (As a random example, consider that `diff` often causes tabs to misalign because it adds a few characters to each line.)  For structured data, as for import to a spreadsheet, CSV seems like a better form.  However, again, if (as seems likely) `print()` allows tabs, one is free to use them.

With respect to automatically including a newline at the end of any `print()` output... that would be my preference, since usually that's what a beginner will want, and an advanced user could construct the desired string and then print it all at once.  (Note that since OpenSCAD execution only very rarely takes any real time, the temporal issue around something saying "Working ... done"" mostly doesn't exist.)  However, that said, I'd be amenable to have a mechanism that would tell `print()` not to include the newline.  Perhaps, for instance, you could optionally include `newline=false` in the call.

dannypike, 2021-10-01:
I can see that omitting the final newline may confuse people, so I am happy for it to stay in. I can't think of any case where I will need to omit it in practice that can't be handled by the technique you describe - building a string in parts and outputting it in a single call at the end.

I am a Windows user, so please include ```\r``` in the list of allowed special characters.

MichaelAtOz, 2021-10-01:
Is this issue about modifying `echo()` or introducing an alternative?

jordanbrown0, 2021-10-02:
@dannypike With respect to \r, I would assume that any output would (and already does) follow the conventions of the platform.  I would assume that if your program says to output a \n, on Windows the output would be CR, LF, and on UNIX would be LF alone.

@MichaelAtOz For discussion purposes I've been assuming introducing an alternative.  We certainly could not change the default behavior of `echo()`.  Hypothetically, we could introduce a parameter to echo that would disable the decorations, or a $ variable that would disable them.  I don't think a preference would be appropriate, because the program is probably the one in the best position to know what's required of its output.  However, although adding such a mechanism to modify the behavior of `echo()` would maintain compatibility, it seems like it would be ugly, so my preference would be a new word.

rcolyer, 2021-10-02:
I would say the issue is now focused on the broad feature wish of better formatted output, and the discussion is open as to how to do it.  I reflected on the issue, and I agree with t-paul that echo is unmodifiable.  Even if we added an option on echo, it is not as simple as just removing ECHO: and some quotes, because for example:  ECHO: "foo", 2, [1, 2, 3, "ab"]   If quotes were removed from "foo", then why not "ab"?  This would require a deeper change, which means that the change desired is analogous to the difference between Python repr, represented by our echo which provides debugging outputs, and Python str/print, producing formatted output.

Therefore I would concur that we need a separate command such as print() to produce simplified output under different rules.  However, this is a functional language, so it would be a serious design mistake to put all the formatting logic into the output command.  I would concur with jordan's comment that the right solution is to also introduce a new command like format() which works analogously to std::format or Python's f-strings, and returns a formatted string.  In this manner formatted strings could be composed functionally, or if we design carefully, in a list comprehension manner, and then collected for final output.

Example code:
```
x = 5;
err = "is not the loneliest number!";
s = format("{color.red}Error:{color.normal} {x} {err}");
print(s);
```
Output (imagine that "Error:" ends up red):
```
Error: 5 is not the loneliest number!
```

The {} syntax could capture from context at the moment of the format call like f-strings as in the example, with most of the substitutions happening at the format call.  Since color is already a built-in, we have the luxury of reusing it here to make special color marker variables which do not get captured at format, but instead are processed at print time according to the output medium, possibly turning into ANSI color codes for console prints, or Qt color change calls for gui use.  It would also be conceivable to make a console output mode that spits out html instead, with appropriate &lt;p&gt;&lt;/p&gt; for lines and html color coding.

jordanbrown0, 2021-10-02:
Hmm.  Although the basic notion of having a markup mechanism for things like color changes seems appealing, I don't think I would mix it into `format()` as I have envisioned it.

First, this seems to imply that the return value from `format()` is not a string, or if it is it's kind of magical.  Maybe it embeds ANSI X3.64 codes, but yuck.  You want the output from `format()` to be usable for, e.g., `text()`.

Second, it seems likely that we would want to follow [fmt](https://fmt.dev)'s lead, and it uses braces as its "insert variable here" markup, including allowing named arguments.

I don't have any really good ideas for supporting special effects like color (and font choice?  and size?), but if I was to have any they'd probably involve an HTML subset rather than an ANSI X3.64 subset.  Perhaps `printhtml(string)`, combined with some HTML-creation functions.

But mostly, aside perhaps from a distinction between errors, warnings, normal output, and progress noise, I wouldn't feel a need for special effects.  They are pretty far from the core goal of 3D modeling.

jordanbrown0, 2021-10-02:
And indeed it is an interesting question what `print(array)` should do, but perhaps `print()` should be defined to emit whatever `str()` would have emitted, so that `print(x)` yields the same output as `print(str(x))`.

Another question is what `print(a,b)` should put between a and b.  My inclination, against based on the beginner / simplest cases, would be that it should put a single space, to make `print("label",value)` produce something appropriate, not jam the two together.  (Note that this is what Python does.)  Somebody who actually wants to jam the two together can always `print(str(a,b))`.

rcolyer, 2021-10-02:
My above example did not adequately clarify how format() differs from str() in utility.  For some examples, for floating point values it would become possible to include float precision formatting analogously to the f-string syntax.  Also, there is a significant and very important opportunity to treat vectors differently.  For example:
```
data = [[1, 2, 3], [4, 5, 6]];
s = format([for (row=data) format(row, $sep=", ")], $sep="\n");
print(s);
```
Output:
```
1, 2, 3
4, 5, 6
```
> this seems to imply that the return value from format() is not a string ... You want the output from format() to be usable for, e.g., text().

Completely agree on the last part.  The result from format() should in fact be a true and proper string, entries like "{color.red}" simply stay unprocessed as text by format, and only print processes the color codes.  The solution to using this for text is to simply not place color codes for things passed to text(), as text does not process color codes, and you'd simply get "{color.red}" showing up in your rendered text.  The behavior would end up obvious and not be an issue.

> But mostly, aside perhaps from a distinction between errors, warnings, normal output, and progress noise, I wouldn't feel a need for special effects. They are pretty far from the core goal of 3D modeling.

Actually one of the core goals of formatted output is 3D modeling with part design, where it is important to have formatted numerical reports that correspond to the specifications of the design that result from input parameters.  Since computing the specifications is OpenSCAD's job, outputting them in a useful manner for human or machine digestion should also be within OpenSCAD's capability.

jordanbrown0, 2021-10-02:
Note that the `print(x)` equivalence to `print(str(x))` appears to be comparable to what Python does; printing strings does not include the quotes, but printing arrays and dicts yields syntax suitable as Python input.

jordanbrown0, 2021-10-02:
Quasi-concrete proposal, at least for a first round:
* `print(string)` emits the string undecorated and unmodified, followed by a newline.
* `print(number)` emits the number as for `print(str(number))`, followed by a newline.
* `print(boolean)` emits the boolean as for `print(str(boolean))`, followed by a newline.
* `print(a,b)` prints a, a space, b, and then a newline.
* `print(undef)` ... don't know.  Either equivalent to `print(str(undef))` and so yields `undef`, or perhaps an error.
* `print(list)`, `print(range)`, `print(function)`, and `print(object)` yield errors.  (Did I miss any?)  This leaves these available for future consideration.
* print(name=value) yields an error.  This too leaves it available for future consideration.

jordanbrown0, 2021-10-02:
Let's take `format()` to a different venue - either a different issue, or something.

dannypike, 2021-10-02:
> @dannypike With respect to \r, I would assume that any output would (and already does) follow the conventions of the platform. I would assume that if your program says to output a \n, on Windows the output would be CR, LF, and on UNIX would be LF alone.

@jordanbrown0 Your point about the model being platform-neutral is well made. I am happy for the conversion to be done internally by the ```print()``` code that writes the file, when OpenSCAD is running on Windows.

As for the discussion about implementing colour codes and other extras, I have no need for it. My need is this one from @rcolyer, who expresses it much better than I did:

> Actually one of the core goals of formatted output is 3D modeling with part design, where it is important to have formatted numerical reports that correspond to the specifications of the design that [...]

My instinct is to avoid the urge to add bloat to OpenSCAD's language and focus on its core behaviour. Having pretty colours in the logs is something that I would consider to be bloat and numbers can be formatted by using the ```round()``` function, if that's important. I would be much more interested in having that effort spent on adding 3D modelling features and optimising the rendering performance.

jordanbrown0, 2021-10-04:
With respect to formatting, there is *some* need to have programs produce output, report-like or data-like.  For report-like output, it's nice to get things like columns that line up, with aligned decimal points.  That's hard to do with mathematical tools like `round()`, because for mathematical purposes `1` and `1.000` are the same, but they are the same for report purposes.

With respect to prioritization, I'd totally agree with you if the available labor was totally fungible and could be directed to any task, in priority order.  However, for instance, although I'm a very experienced programmer I have next to no experience in 3D geometry and 3D rendering.  I can contribute productively on features like output formatting, but would not be nearly so productive working on geometry and rendering.

dannypike, 2021-10-05:
Good points. Would you be any good at making the render use all of the available threads? That would be the next biggest improvement from my point of view. Or, perhaps, help me to get it to compile on Windows and I could take a look at it, myself.

I did have a quick look but decided that the build docs were too old to risk jumping into the rabbit hole. BTDTGTTS :)

jordanbrown0, 2021-10-05:
This is moving away from the topic of this issue.  I'll start a separate discussion thread.

jordanbrown0, 2021-10-05:
MT in preview is covered by #1195.  MT in full F6 render is covered by #391.  I could conceivably be useful on those.  Let's move any further discussion there.

With respect to building on Windows, drop me a note at openscad(at)jordan.maileater.net and I'll walk you through it.  It's not too bad.  (But even the newer doc is a bit stale.)