openscad/openscad #3916: MacOS build broken - Qt 5.9.9 no longer available

ChrisCoxArt, 2021-10-03:
macosx-build-dependencies.sh specifies QT 5.9.9.
Qt 5.9.x is no longer available at http://download.qt.io/official_releases/qt/
Qt 5.12 is the oldest version available on Qt's website.
A misinformed developer claimed in #3896 that "The download script tries the main location first and falls back trying the archive folder." --- but that is not what happens. The download goes to a 404 page, the HTML of the 404 is downloaded into the archive filename, and of course that fails to decompress.  There is no backup step, there is no attempt to access the archive folder, and reading macosx-build-dependencies.sh / build_qt5() I highly doubt that such a backup step ever happens.
But the problem remains - if someone does a clean download and build on MacOS, the OpenSCAD build is broken because dependencies cannot be downloaded.
I'll add a separate bug for the fact that the CI builds claim to be working even though dependencies are not available.

t-paul, 2021-10-03:
You may need to read again what I wrote. The fallback I've mentioned pointed to a new PR #3915 which was still running through CI (even though it's not going to change anything, the normal way is to still wait for everything to go green).

It's now merged, so the changes from #3915 including the fallback to download from https://download.qt.io/archive/qt/5.9/5.9.9/ is now merged.

https://github.com/openscad/openscad/blob/eb14aad24baaa7253182082fb62e2aa505ee020c/scripts/macosx-build-dependencies.sh#L172-L175

ChrisCoxArt, 2021-10-03:
Your reply failed to mention a new version of the download/dependency script and spoke as if referring to the current script, which did not do the things you claimed.
Your condescension, insults, and dismissive attitude really need to stop.  Do I need to remind you of https://docs.github.com/en/github/site-policy/github-community-guidelines ?

closed 2021-10-03 by rcolyer