openscad/openscad #3917: MacOS CI builds are showing success, despite dependencies being unavailable (Qt 5.9.9 missing)

ChrisCoxArt, 2021-10-03:
For some reason, the CI and master builds of OpenSCAD for MacOS are succeeding, despite the fact that Qt 5.9.9 (a required dependency) has not been available for download for some time. See #3916 
A clean build done by downloading all source fails on my MacOS systems, because Qt 5.9.9 is no longer available.

This suggests two possible problems: a) the build is not clean, relies on cached files, and is failing to check dependencies; or b) the build is somehow claiming success when it could not possible succeed.
I strongly suspect that (a) is the case - the builds are not clean and are failing to check the dependencies.

t-paul, 2021-10-03:
We can't do a full dependency build as we have limited amount of minutes to spend on CI. A full build of just Qt takes > 2h which would essentially eat a month worth of minutes in one or 2 builds. CI only does a clean build of OpenSCAD itself, which is the main point.

ChrisCoxArt, 2021-10-03:
Then some steps need to be added to check that the dependencies still exist.
Nobody else can do a build unless the ENTIRE build works, and that includes the dependencies.
The MacOS build has been broken for some time, claimed that it was succeeding, but could only succeed on your servers because you cached the dependencies.
If the build is broken, it should say that it is broken, because you need to know that it is broken, even if the cause is external.

ChrisCoxArt, 2021-10-03:
BTW - this is still a bug, not a discussion.
You claimed the build was working, when it could not possibly work unless old data was cached.

rcolyer, 2021-10-03:
@ChrisCoxArt As mentioned, the existing build process cannot be changed to compiling all dependencies for the CI resourcing reasons mentioned, so that is not a resolution option.  Also, in terms of process design, the primary goal of the build process is to successfully produce a compiled version of OpenSCAD to validate the OpenSCAD code itself, and to produce a nightly release, both of which are succeeding (hence the automated build system's automatic marking of this).  We're not the ones responsible for providing the Qt 5.9.9 source (this is a legal obigation on the Qt Company), but it does appear available here still: https://download.qt.io/archive/qt/5.9/5.9.9/single/

If that doesn't resolve it, the rest of the comments have not provided an alternate resolution path for anything yet.  If 5.9.9 is failing for some reason in systems other than our build process, then some sort of analysis of the failure on your system might lead to a fix to the repository.  You could also consider compiling with a different version of Qt.  I use Qt 5.12.8 for the builds on my own system, just because that's what's there, and this does not cause problems.

ChrisCoxArt, 2021-10-03:
You may not compile all dependencies. But you can check to see if those dependencies are downloadable.

Also, I previously attempted to use 5.12.11 and 5.12.0 instead of 5.9.9 -- and both failed with errors when compiling OpenSCAD source.

As someone attempting to contribute to OpenSCAD:  finding that the builds are broken, developers don't care, and multiple developers don't even understand the problem with their builds... is more than a little disappointing.

rcolyer, 2021-10-03:
All dependencies appear to be downloadable as we all download them pretty regularly.  It sounds like one older version link simply moved.  5.12 is known to work, 5.14 was used for some builds on the last release, and 5.15 is used on some of the nightly builds.  This should and commonly does work across a pretty broad range.  If there is a specific build error, that could be an actionable issue to open.

Please note that the developer base here is a caring, collaborative, self-motivated group of volunteers who are here because they care about the project and its goals, and that insulting them or other inflammatory comments would be outside of expected community standards.  I already had to delete one post, so I'm asking you to please conform to polite professional discourse for continued discussion and productive results.

ChrisCoxArt, 2021-10-03:
And yet the builds were not working because the dependencies were not downloadable for quite some time. And you apparently had no idea that the build was broken or why.
Plus, I've already established that Qt 5.12 and later definitely do not build with OpenSCAD on MacOS (part looks like bad macros from OpenSCAD, part looks like the compiler flags may be mis-set).  It looks like someone has a branch open to work on related issues, but those changes are not in the mainline at this time.

Fortunately one of your developers did address the dependency download issue.  But you still have a fragile build that only works for you because of caching, and fails to notify you when dependencies are unavailable.

Now please, take some time to try and understand how this looks from my viewpoint.
I want to contribute, but the dependencies script was obviously broken and I couldn't even complete a baseline build (and still can't - there are yet more problems for me to investigate).  Reports that the build is broken are met with dismissal and condescension (coming just short of insults).  Further details are met with more dismissal, and then yet another developer shows up to further dismiss real issues (of both code and communication) instead of solving them.
Have you ever walked away from an open source project because the existing developers were dismissive and insulting when you reported problems or tried to help?

rcolyer, 2021-10-04:
I read the exchanges before I arrived, and I think you are unfairly mischaracterizing them.  You opened by calling the current head developer misinformed, when in fact he works hard and competently to keep the mac CI builds running despite no personal benefit to him from this.  In response, you were treated uniformly with politeness by everyone who responded, as we would like you to follow the same example.

Please note that other than a source file having moved, which was addressed, we have not seen a single detailed error report or bug report here yet, so there is not actually something we can address.  Currently as far as I know there are zero active core developers actually using macs for this project, leaving the only core build testing for this being the CI system, which is kept running so that the mac releases continue.

A call has been out for some time for more users on macs to take a core role in testing, and for developer-users who develop on macs to contribute to build testing or looking into platform-specific bugs.  If you want to contribute to that, please do so.  But what it would need to look like would be detailed bug reports of specific build issues, technically focused low-emotion discussions, and a collaborative willingness to help with trying things out to discover the correct resolutions.

I will close this issue, as its initial topic was resolved by a different url, and there is no viability of adding CI-automated Qt builds in our pipeline.  If there are other build version bugs, please open them in a separate issue detailing those problems.  But I encourage you to please open with a positive collaborative tone, as that is required for open source projects to thrive.

closed 2021-10-04 by rcolyer