openscad/openscad #3919: MacOS build broken - hidapi 0.9.0 cannot compile due to duplicated line in configure.ac file

ChrisCoxArt, 2021-10-04:
hidapi 0.11.0 fixes this error

Building hidapi 0.9.0...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   124  100   124    0     0    407      0 --:--:-- --:--:-- --:--:--   407
100  131k    0  131k    0     0   144k      0 --:--:-- --:--:-- --:--:--  144k
Archive:  hidapi-0.9.0.zip
7da5cc91fc0d2dbe4df4f08cd31f6ca1a262418f
   creating: hidapi-hidapi-0.9.0/
  inflating: hidapi-hidapi-0.9.0/.gitattributes  
  inflating: hidapi-hidapi-0.9.0/.gitignore  
  inflating: hidapi-hidapi-0.9.0/AUTHORS.txt  
  inflating: hidapi-hidapi-0.9.0/HACKING.txt  
  inflating: hidapi-hidapi-0.9.0/LICENSE-bsd.txt  
  inflating: hidapi-hidapi-0.9.0/LICENSE-gpl3.txt  
  inflating: hidapi-hidapi-0.9.0/LICENSE-orig.txt  
  inflating: hidapi-hidapi-0.9.0/LICENSE.txt  
  inflating: hidapi-hidapi-0.9.0/Makefile.am  
  inflating: hidapi-hidapi-0.9.0/README.md  
   creating: hidapi-hidapi-0.9.0/android/
   creating: hidapi-hidapi-0.9.0/android/jni/
  inflating: hidapi-hidapi-0.9.0/android/jni/Android.mk  
  inflating: hidapi-hidapi-0.9.0/bootstrap  
  inflating: hidapi-hidapi-0.9.0/configure.ac  
   creating: hidapi-hidapi-0.9.0/doxygen/
  inflating: hidapi-hidapi-0.9.0/doxygen/Doxyfile  
   creating: hidapi-hidapi-0.9.0/hidapi/
  inflating: hidapi-hidapi-0.9.0/hidapi/hidapi.h  
   creating: hidapi-hidapi-0.9.0/hidtest/
  inflating: hidapi-hidapi-0.9.0/hidtest/.gitignore  
  inflating: hidapi-hidapi-0.9.0/hidtest/Makefile.am  
  inflating: hidapi-hidapi-0.9.0/hidtest/hidtest.cpp  
   creating: hidapi-hidapi-0.9.0/libusb/
  inflating: hidapi-hidapi-0.9.0/libusb/.gitignore  
  inflating: hidapi-hidapi-0.9.0/libusb/Makefile-manual  
  inflating: hidapi-hidapi-0.9.0/libusb/Makefile.am  
  inflating: hidapi-hidapi-0.9.0/libusb/Makefile.freebsd  
  inflating: hidapi-hidapi-0.9.0/libusb/Makefile.linux  
  inflating: hidapi-hidapi-0.9.0/libusb/hid.c  
   creating: hidapi-hidapi-0.9.0/linux/
  inflating: hidapi-hidapi-0.9.0/linux/.gitignore  
  inflating: hidapi-hidapi-0.9.0/linux/Makefile-manual  
  inflating: hidapi-hidapi-0.9.0/linux/Makefile.am  
  inflating: hidapi-hidapi-0.9.0/linux/README.txt  
  inflating: hidapi-hidapi-0.9.0/linux/hid.c  
   creating: hidapi-hidapi-0.9.0/m4/
  inflating: hidapi-hidapi-0.9.0/m4/.gitignore  
  inflating: hidapi-hidapi-0.9.0/m4/ax_pthread.m4  
  inflating: hidapi-hidapi-0.9.0/m4/pkg.m4  
   creating: hidapi-hidapi-0.9.0/mac/
  inflating: hidapi-hidapi-0.9.0/mac/.gitignore  
  inflating: hidapi-hidapi-0.9.0/mac/Makefile-manual  
  inflating: hidapi-hidapi-0.9.0/mac/Makefile.am  
  inflating: hidapi-hidapi-0.9.0/mac/hid.c  
   creating: hidapi-hidapi-0.9.0/pc/
 extracting: hidapi-hidapi-0.9.0/pc/.gitignore  
  inflating: hidapi-hidapi-0.9.0/pc/hidapi-hidraw.pc.in  
  inflating: hidapi-hidapi-0.9.0/pc/hidapi-libusb.pc.in  
  inflating: hidapi-hidapi-0.9.0/pc/hidapi.pc.in  
   creating: hidapi-hidapi-0.9.0/testgui/
  inflating: hidapi-hidapi-0.9.0/testgui/.gitignore  
  inflating: hidapi-hidapi-0.9.0/testgui/Makefile-manual  
  inflating: hidapi-hidapi-0.9.0/testgui/Makefile.am  
  inflating: hidapi-hidapi-0.9.0/testgui/Makefile.freebsd  
  inflating: hidapi-hidapi-0.9.0/testgui/Makefile.linux  
  inflating: hidapi-hidapi-0.9.0/testgui/Makefile.mac  
  inflating: hidapi-hidapi-0.9.0/testgui/Makefile.mingw  
   creating: hidapi-hidapi-0.9.0/testgui/TestGUI.app.in/
   creating: hidapi-hidapi-0.9.0/testgui/TestGUI.app.in/Contents/
  inflating: hidapi-hidapi-0.9.0/testgui/TestGUI.app.in/Contents/Info.plist  
 extracting: hidapi-hidapi-0.9.0/testgui/TestGUI.app.in/Contents/PkgInfo  
   creating: hidapi-hidapi-0.9.0/testgui/TestGUI.app.in/Contents/Resources/
   creating: hidapi-hidapi-0.9.0/testgui/TestGUI.app.in/Contents/Resources/English.lproj/
  inflating: hidapi-hidapi-0.9.0/testgui/TestGUI.app.in/Contents/Resources/English.lproj/InfoPlist.strings  
  inflating: hidapi-hidapi-0.9.0/testgui/TestGUI.app.in/Contents/Resources/Signal11.icns  
  inflating: hidapi-hidapi-0.9.0/testgui/copy_to_bundle.sh  
  inflating: hidapi-hidapi-0.9.0/testgui/mac_support.cpp  
  inflating: hidapi-hidapi-0.9.0/testgui/mac_support.h  
  inflating: hidapi-hidapi-0.9.0/testgui/mac_support_cocoa.m  
  inflating: hidapi-hidapi-0.9.0/testgui/start.sh  
  inflating: hidapi-hidapi-0.9.0/testgui/test.cpp  
  inflating: hidapi-hidapi-0.9.0/testgui/testgui.sln  
  inflating: hidapi-hidapi-0.9.0/testgui/testgui.vcproj  
   creating: hidapi-hidapi-0.9.0/udev/
  inflating: hidapi-hidapi-0.9.0/udev/99-hid.rules  
   creating: hidapi-hidapi-0.9.0/windows/
  inflating: hidapi-hidapi-0.9.0/windows/.gitignore  
  inflating: hidapi-hidapi-0.9.0/windows/Makefile-manual  
  inflating: hidapi-hidapi-0.9.0/windows/Makefile.am  
  inflating: hidapi-hidapi-0.9.0/windows/Makefile.mingw  
   creating: hidapi-hidapi-0.9.0/windows/ddk_build/
 extracting: hidapi-hidapi-0.9.0/windows/ddk_build/.gitignore  
  inflating: hidapi-hidapi-0.9.0/windows/ddk_build/hidapi.def  
  inflating: hidapi-hidapi-0.9.0/windows/ddk_build/makefile  
  inflating: hidapi-hidapi-0.9.0/windows/ddk_build/sources  
  inflating: hidapi-hidapi-0.9.0/windows/hid.c  
  inflating: hidapi-hidapi-0.9.0/windows/hidapi.sln  
  inflating: hidapi-hidapi-0.9.0/windows/hidapi.vcproj  
  inflating: hidapi-hidapi-0.9.0/windows/hidtest.vcproj  
+ autoreconf --install --verbose --force
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
configure.ac:23: error: AC_CONFIG_MACRO_DIR can only be used once
./lib/autoconf/general.m4:1970: AC_CONFIG_MACRO_DIR is expanded from...
configure.ac:23: the top level
autom4te: error: /usr/local/opt/m4/bin/m4 failed with exit status: 1
aclocal: error: /usr/local/Cellar/autoconf/2.71/bin/autom4te failed with exit status: 1
autoreconf: error: aclocal failed with exit status: 1

rcolyer, 2021-10-04:
This is not an OpenSCAD bug.  Also, CMakeLists.txt in the master branch looks for hidapi 0.10 without specifying beyond the minor version number.

ChrisCoxArt, 2021-10-04:
The macosx dependency script specifies 0.9.0.  But that version does not build.
Thus the script needs to be updated to version 0.11.0 to pick up the fix.
Yes, the fact that the OpenSCAD dependency script specifies a version that cannot build - is a bug, that needs to be fixed in the OpenSCAD dependency script.

ChrisCoxArt, 2021-10-04:
Please reopen this bug so one of your engineers can make the fix to the dependency script.

rcolyer, 2021-10-04:
You didn't provide instructions for getting this error, but by chance are you getting this while running the macosx-build-dependencies.sh script?  I don't think this script is in very common use, and is not the way most people compile OpenSCAD.  But I do see 0.9.0 in there.  I think this might be out of date, as the CMakeLists.txt is not looking for that.  I will clarify that an inability to compile hidapi 0.9.0 on one specific system is out of scope for our issue tracker.  Resolving upstream build issues is either the responsibility of the upstream project, or if it requires a system reconfiguration, the responsibility of the developer attempting compilation.

But if 0.9.0 mismatches what CMakeLists is looking for with "find_package(HidAPI 0.10 QUIET)", then there may be an issue to resolve in that file.

ChrisCoxArt, 2021-10-04:
The dependencies script is a must have for building on MacOS, according to the instructions in the README file, see https://github.com/openscad/openscad#building-for-mac-os-x
The instructions are: follow the MacOS  build instructions and run the scripts specified.
Since this is the only way to compile OpenSCAD on MacOS, I suspect it is in use for anyone working on MacOS.

And you could have just asked for more info.

rcolyer, 2021-10-04:
It's actually not a "must have", as those instructions also provide the more common option of simply installing the dependencies.

Compilation on specific systems can be challenging and require expertise, and is not a normal user workflow.  This is why no project supports the details of direct compilation of all dependencies on all systems.  Even upstream projects themselves typically don't support compilation of their old versions.  I will edit the title to reflect the surviving issue, which is a version mismatch.

ChrisCoxArt, 2021-10-04:
Of course, the MacOS instructions say "3 options" and only gives 2.
But both approaches have problems.

PR #3924 (2021-10-04): Bump hidapi to version 0.11.0.
Closes #3919.
closed 2021-10-04 by t-paul