openscad/openscad #3920: MacOS build broken because "poppler 21.01.0" does not compile

ChrisCoxArt, 2021-10-04:
"poppler 21.01.0" cannot compile due to missing JPEG library and header references
This appears to be a compile options error, it is not fixed with later library versions.
It appears to be fixed if the options "-DENABLE_LIBOPENJPEG=none -DENABLE_DCTDECODER=none" are removed from macosx-build-dependencies.sh
But since the full MacOS build still does not work, I haven not been able to actually test the compiled poppler library in OpenSCAD.

bash-3.2$  ./scripts/macosx-build-dependencies.sh
Detected BigSur (11.x) or later
Building for 10.9 or later
Setting number of CPUs to 24
Using basedir: /Volumes/Development/openscad/../libraries
double_conversion eigen gmp mpfr glew gettext libffi freetype ragel harfbuzz libzip libxml2 fontconfig hidapi libuuid lib3mf glib2 boost poppler pixman cairo cgal qt5 opencsg qscintilla
Building packages: double_conversion eigen gmp mpfr glew gettext libffi freetype ragel harfbuzz libzip libxml2 fontconfig hidapi libuuid lib3mf glib2 boost poppler pixman cairo cgal qt5 opencsg qscintilla 

double_conversion 3.1.5 already installed - not building
eigen 3.3.7 already installed - not building
gmp 6.1.2 already installed - not building
mpfr 4.0.2 already installed - not building
glew 2.1.0 already installed - not building
gettext 0.21 already installed - not building
libffi 3.2.1 already installed - not building
freetype 2.9.1 already installed - not building
ragel 6.10 already installed - not building
harfbuzz 2.3.1 already installed - not building
libzip 1.5.1 already installed - not building
libxml2 2.9.9 already installed - not building
fontconfig 2.13.1 already installed - not building
hidapi 0.11.0 already installed - not building
libuuid 1.6.2 already installed - not building
lib3mf 1.8.1 already installed - not building
glib2 2.56.3 already installed - not building
boost 1.74.0 already installed - not building
Building poppler 21.01.0 ...
-- The C compiler identification is AppleClang 13.0.0.13000029
-- The CXX compiler identification is AppleClang 13.0.0.13000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Checking _FILE_OFFSET_BITS for large files
-- Checking _FILE_OFFSET_BITS for large files - not needed
CMake Warning at CMakeLists.txt:82 (message):
  

   No test data found in $testdatadir.
   You will not be able to run 'make test' successfully.

  

   The test data is not included in the source packages
   and is also not part of the main git repository. Instead,
   you can checkout the test data from its own git
   repository with:

  

     git clone git://git.freedesktop.org/git/poppler/test

  
[... 347 lines cut ...]
/Volumes/Development/libraries/src/poppler-21.01.0/poppler/DCTStream.cc:97:5: error: use of undeclared identifier 'src'
    src.pub.resync_to_restart = jpeg_resync_to_restart;
    ^
/Volumes/Development/libraries/src/poppler-21.01.0/poppler/DCTStream.cc:98:5: error: use of undeclared identifier 'src'
    src.pub.term_source = str_term_source;
    ^
/Volumes/Development/libraries/src/poppler-21.01.0/poppler/DCTStream.cc:99:5: error: use of undeclared identifier 'src'
    src.pub.bytes_in_buffer = 0;
    ^
/Volumes/Development/libraries/src/poppler-21.01.0/poppler/DCTStream.cc:100:5: error: use of undeclared identifier 'src'
    src.pub.next_input_byte = nullptr;
    ^
/Volumes/Development/libraries/src/poppler-21.01.0/poppler/DCTStream.cc:101:5: error: use of undeclared identifier 'src'
    src.str = str;
    ^
/Volumes/Development/libraries/src/poppler-21.01.0/poppler/DCTStream.cc:102:5: error: use of undeclared identifier 'src'
    src.index = 0;
    ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [CMakeFiles/poppler.dir/poppler/DCTStream.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/poppler.dir/all] Error 2
make: *** [all] Error 2
bash-3.2$

TheJKM, 2021-10-05:
I've came across the same issue when updating Qt (see branch `qt5-upgrade-macos`). It works on the CI though, I always used the homebrew version for testing builds. I think it can also be solved by adding a missing dependency, it would be good to have those scripts working completely on recent macOS versions.

t-paul, 2021-10-05:
I've updated `master` to macports 2.7.1 and added python3 to the installed packages as latest poppler failed with "can't find python". But I'm still having problems building that poppler version.

t-paul, 2021-10-05:
Oh, and `qt5-upgrade-macos` is rebased to this `master` branch, so it includes that updated macports too. If the build goes through, we'll see in a couple of hours (assuming macOS builds are still open).

ChrisCoxArt, 2021-10-05:
Yes, the homebrew option is more reliable IFF you don't have any other Qt versions installed (otherwise they conflict all over the place).  However, it will always give you the latest (or close to latest) version of each package - not known good versions, or versions that support the OS versions you want to support.
Someone was already complaining on IRC that upgrading Qt versions will mean they cannot run OpenSCAD on their current MacOS systems (which are limited to MacOS 10.13 because Apple deprecated a lot of hardware, including NVidia GPUs).

kintel, 2021-10-30:
@t-paul Btw., why do we need `poppler`? I looked through code and tests but cannot find any usage of it?

t-paul, 2021-10-30:
It's used by cairo as backend for the PDF export.

t-paul, 2021-10-30:
Hmm, that said, I'm not seeing it in the dependency list on Debian at first glance. I'll have a closer look if that's actually what's going on :smile:.

t-paul, 2021-10-30:
The cairo build does want to use poppler, but if I'm reading https://github.com/freedesktop/cairo/blob/master/configure.ac#L633-L651 correctly, it might use it for testing purposes, not the actual PDF generation. That sounds like we may be able to drop the dependency on our side.

kintel, 2021-10-30:
Yeah, sound correct, building cairo after removing poppler works with this notice:
```
  pdf testing:   no (requires poppler-glib >= 0.17.4)
```

t-paul, 2021-10-30:
Cool, nice catch. I like that fix (assuming PDF export still works) :grinning:.

t-paul, 2021-10-30:
I think we also considered using `poppler-utils` for our tests, but that ended up with `ghostscript` as that seemed easier to get stable on the various platform.

kintel, 2021-10-31:
Fixed by removing poppler: https://github.com/openscad/openscad/pull/3965

closed 2021-10-31 by kintel