openscad/openscad #3921: script macosx-build-dependencies.sh does not detect recent MacOS versions correctly

ChrisCoxArt, 2021-10-04:
simple fix - replace the version check with this code. I don't think the version is used in the script and is only cosmetic for documenting the build environment.

OSX_MAJOR_VERSION=`sw_vers -productVersion | cut -d. -f1`
OSX_VERSION=`sw_vers -productVersion | cut -d. -f2`
if (( $OSX_MAJOR_VERSION >= 11 )); then
  echo "Detected BigSur (11.x) or later"
elif (( $OSX_VERSION >= 15 )); then
  echo "Detected Catalina (10.15) or later"
elif (( $OSX_VERSION >= 14 )); then
  echo "Detected Mojave (10.14) or later"
elif (( $OSX_VERSION >= 13 )); then
  echo "Detected High Sierra (10.13) or later"
elif (( $OSX_VERSION >= 12 )); then
  echo "Detected Sierra (10.12) or later"
elif (( $OSX_VERSION >= 11 )); then
  echo "Detected El Capitan (10.11) or later"
elif (( $OSX_VERSION >= 10 )); then
  echo "Detected Yosemite (10.10) or later"
elif (( $OSX_VERSION >= 9 )); then
  echo "Detected Mavericks (10.9)"
elif (( $OSX_VERSION >= 8 )); then
  echo "Detected Mountain Lion (10.8)"
else
  echo "Detected Lion (10.7) or earlier"
fi

rcolyer, 2021-10-04:
Since you seem to have ideas on what should be altered, and you have a mac to test them on, might I suggest consolidating these changes into a PR that resolves all of the issues you opened by editing the macosx-build-dependencies.sh script?

ChrisCoxArt, 2021-10-04:
I would like to - but still can barely get past the cmake step and still have yet to compile most of OpenSCAD.  So I cannot test most of the changes I'm making to work around the build errors.  I can't start altering the build until I get a build that works and I can test.

Fortunately this particular bug is simple and straightforward, and won't affect the build as far as I can tell.

PR #3923 (2021-10-04): Add more macOS versions.
Closes #3921.
closed 2021-10-04 by t-paul