openscad/openscad #3922: MacOS cmake build step fails because 2 programs it runs crash with Qt dependency errors

ChrisCoxArt, 2021-10-04:
MacOS cmake build fails because 2 programs crash, because are seeing the wrong version of Qt
This is likely because the dependencies script install 5.9.9 , while the homebrew script installs qt5 == 5.12.2
crash reports are in comments

bash-3.2$ cmake ..
CMake Deprecation Warning at CMakeLists.txt:18 (cmake_policy):
  The OLD behavior for policy CMP0071 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Using header-only CGAL
-- Targetting Unix Makefiles
-- Using /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ compiler.
-- DARWIN_VERSION=20
-- Mac Leopard detected
-- Found Boost: /usr/local/lib/cmake/Boost-1.76.0/BoostConfig.cmake (found suitable version "1.76.0", minimum required is "1.48")  
-- Boost include dirs: /usr/local/include
-- Boost libraries:    
-- CGAL: 5.3
-- CGAL: Using target CGAL::CGAL CGAL::CGAL_Core
-- Eigen3: 3.4.0
-- Found Boost: /usr/local/lib/cmake/Boost-1.76.0/BoostConfig.cmake (found suitable version "1.76.0", minimum required is "1.36") found components: filesystem system regex program_options 
-- Boost: 1.76.0
-- Harfbuzz: 3.0.0
-- Fontconfig: 
-- running openscad/cmake/Modules/FindGLIB2.cmake ...
-- saving pkg-config env
-- found glib.h under OPENSCAD_LIBRARIES.
-- redirecting pkg-config to look under OPENSCAD_LIBRARIES
-- GLIB2_VERSION 2.70.0
-- GLIB2_LIBRARY_DIRS /usr/local/Cellar/glib/2.70.0/lib;/usr/local/opt/gettext/lib
-- GLIB2_INCLUDE_DIRS:
--   /usr/local/Cellar/glib/2.70.0/include/glib-2.0
--   /usr/local/Cellar/glib/2.70.0/lib/glib-2.0/include
--   /usr/local/opt/gettext/include
--   /usr/local/Cellar/pcre/8.45/include
-- GLIB2_LIBDIR /usr/local/Cellar/glib/2.70.0/lib
-- GLIB2_LIBRARIES:
--   /usr/local/Cellar/glib/2.70.0/lib/libglib-2.0.dylib
--   /usr/local/opt/gettext/lib/libintl.dylib
-- restoring pkg-config env
-- Glib: 2.70.0
-- OpenCSG: 
-- GLEW_INCLUDE_DIR: /usr/local/include
-- GLEW_LIBRARY: /usr/local/lib/libGLEW.dylib
-- OpenGL: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/System/Library/Frameworks/OpenGL.framework;/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/System/Library/Frameworks/OpenGL.framework
-- libzip: 1.8.0
-- Freetype: 2.11.0
-- LibXml2: 2.9.4
-- HIDAPI: disabled
-- Cairo: 1.16.0
-- Flex: 2.6.4
-- Bison: 2.3
-- lib3mf: 1.8.1
-- Qt5: 5.15.2
-- QScintilla: 2.13.0
-- DBus input driver enabled
-- Qt5Gamepad input driver enabled
-- Offscreen OpenGL Context - using Apple CGL
--  
-- ====================================
-- OpenSCAD Build Configuration Summary
-- ====================================
--  
-- CMAKE_BUILD_TYPE:  
-- CMAKE_PREFIX_PATH: /Volumes/Development/openscad/../libraries/install;/usr/local/opt/qt;/usr/local/opt/qt@5
-- SUFFIX:            
--  
-- Experimental Features: OFF
-- Snapshot build:        OFF
-- Headless build:        OFF
-- NULLGL build:          OFF
-- Profiling/coverage:    OFF
-- CSG node id-prefix:    OFF
--  
-- lib3mf:                API 1.x
--  
-- Input Drivers
-- - HID API Driver (disabled)
-- - Space Navigator Library Driver (disabled)
-- - Joystick Driver (disabled)
-- - DBus Remote Driver (enabled)
--  
-- DEFINES = _REENTRANT;UNICODE;_UNICODE;STACKSIZE=8388608;ENABLE_CGAL;EIGEN_DONT_ALIGN;ENABLE_OPENCSG;ENABLE_LIBZIP;ENABLE_CAIRO;USE_QOPENGLWIDGET;USE_SCINTILLA_EDITOR;ENABLE_MDI;ENABLE_DBUS;ENABLE_QGAMEPAD;OPENSCAD_VERSION=2021.10.03;OPENSCAD_SHORTVERSION=2021.10.03;OPENSCAD_YEAR=2021;OPENSCAD_MONTH=10;OPENSCAD_DAY=3
--  
-- ImageMagick convert executable found: /usr/local/bin/convert
-- creating CTestCustom.cmake
-- Configuring done
CMake Error: AUTOUIC for target OpenSCAD: Test run of "uic" executable "/usr/local/opt/qt@5/bin/uic" failed.
/usr/local/opt/qt@5/bin/uic -h

dyld: Library not loaded: /usr/local/Cellar/qt@5/5.15.2/lib/QtCore.framework/Versions/5/QtCore
  Referenced from: /usr/local/opt/qt@5/bin/uic
  Reason: Incompatible library version: uic requires version 5.15.0 or later, but QtCore provides version 5.9.0
Subprocess aborted
CMake Generate step failed.  Build files cannot be regenerated correctly.

ChrisCoxArt, 2021-10-04:
UIC from Qt
Process:               uic [17722]
Path:                  /usr/local/Cellar/qt@5/5.15.2/bin/uic
Identifier:            uic
Version:               0
Code Type:             X86-64 (Native)
Parent Process:        cmake [17687]
Responsible:           Terminal [449]
User ID:               501

Date/Time:             2021-10-03 22:08:37.129 -0700
OS Version:            macOS 11.5.2 (20G95)
Report Version:        12
Bridge OS Version:     5.5 (18P4759a)
Anonymous UUID:        096D5B68-B94E-63F0-17DD-11136D1F6B88

Sleep/Wake UUID:       D9DBE1AD-68DC-4E43-A54F-C50671F43489

Time Awake Since Boot: 360000 seconds
Time Since Wake:       960 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x3] Wrong version

Application Specific Information:
dyld: launch, loading dependent libraries
DYLD_FRAMEWORK_PATH=/Volumes/Development/openscad/../libraries/install/lib
DYLD_LIBRARY_PATH=/Volumes/Development/openscad/../libraries/install/lib

Dyld Error Message:
  Library not loaded: /usr/local/Cellar/qt@5/5.15.2/lib/QtCore.framework/Versions/5/QtCore
  Referenced from: /usr/local/opt/qt@5/bin/uic
  Reason: Incompatible library version: uic requires version 5.15.0 or later, but QtCore provides version 5.9.0

Binary Images:
       0x10c190000 -        0x10c1ebfff +uic (0) <F07193E9-4C29-3BB5-86A0-43FFF83C7A71> /usr/local/opt/qt@5/bin/uic
       0x1163a9000 -        0x116444fff  dyld (852.2) <57DB2053-BFD5-3683-97C6-F1DB2A1F1D09> /usr/lib/dyld

Model: MacPro7,1, BootROM 1554.140.20.0.0 (iBridge: 18.16.14759.0.1,0), 12 processors, 12-Core Intel Xeon W, 3.3 GHz, 32 GB, SMC 
Graphics: AMD Radeon Pro W5700X, AMD Radeon Pro W5700X, spdisplays_pcie_device, 16 GB
Memory Module: Slot 3 (Channel E / DIMM 1), 8 GB, DDR4 R-DIMM, 2933 MHz, Micron, 9ASF1G72PZ-2G9E6
Memory Module: Slot 5 (Channel D / DIMM 1), 8 GB, DDR4 R-DIMM, 2933 MHz, Micron, 9ASF1G72PZ-2G9E6
Memory Module: Slot 8 (Channel A / DIMM 1), 8 GB, DDR4 R-DIMM, 2933 MHz, Micron, 9ASF1G72PZ-2G9E6
Memory Module: Slot 10 (Channel B / DIMM 1), 8 GB, DDR4 R-DIMM, 2933 MHz, Micron, 9ASF1G72PZ-2G9E6
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Jul  7 2021 02:54:21 version 9.30.444.17.32.5.70 FWID 01-6551fb93
Bluetooth: Version 8.0.5d7, 3 services, 18 devices, 1 incoming serial ports
Network Service: multiether 2, Ethernet, en1
PCI Card: AMD Radeon Pro W5700X, gpu-controller, Slot-1@9,0,0
PCI Card: pci1002,ab38, sppci_audiodevice, Slot-1@9,0,1
PCI Card: XHC4, sppci_usbxhci, Thunderbolt@200,0,0
PCI Card: XHC2, sppci_usbxhci, Thunderbolt@83,0,0
PCI Card: pci8086,15eb, sppci_thunderbolt, Thunderbolt@201,0,0
PCI Card: pci8086,15eb, sppci_thunderbolt, Thunderbolt@84,0,0
PCI Card: XHC3, sppci_usbxhci, Thunderbolt@24,0,0
PCI Card: pci8086,15eb, sppci_thunderbolt, Thunderbolt@25,0,0
USB Device: USB 3.0 Bus
USB Device: Hub
USB Device: USB Laser Mouse
USB Device: macally Kbd Hub
USB Device: macally Kbd Hub
USB Device: USB-Serial (0001)
USB Device: LCS USB Audio
USB Device: ASM107x
USB Device: TerraMaster
USB Device: TerraMaster
USB Device: TerraMaster
USB Device: TerraMaster
USB Device: USB3.1 Hub
USB Device: USB3.0 Card Reader
USB Device: USB3.1 Hub
USB Device: ASM107x
USB Device: USB2.1 Hub
USB Device: USB2.1 Hub
USB Device: Apple T2 Bus
USB Device: Headset
USB Device: Apple T2 Controller
Thunderbolt Bus: Mac Pro, Apple Inc., 63.3
Thunderbolt Bus: Mac Pro, Apple Inc., 63.3
Thunderbolt Bus: Mac Pro, Apple Inc., 63.3
Thunderbolt Bus: Mac Pro, Apple Inc., 63.3

ChrisCoxArt, 2021-10-04:
This one isn't as clear in cause.

Process:               magick [17705]
Path:                  /usr/local/Cellar/imagemagick/7.1.0-9/bin/magick
Identifier:            magick
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        cmake [17687]
Responsible:           Terminal [449]
User ID:               501

Date/Time:             2021-10-03 22:08:35.269 -0700
OS Version:            macOS 11.5.2 (20G95)
Report Version:        12
Bridge OS Version:     5.5 (18P4759a)
Anonymous UUID:        096D5B68-B94E-63F0-17DD-11136D1F6B88

Sleep/Wake UUID:       D9DBE1AD-68DC-4E43-A54F-C50671F43489

Time Awake Since Boot: 360000 seconds
Time Since Wake:       960 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x3] Wrong version

Application Specific Information:
dyld: launch, loading dependent libraries
DYLD_FRAMEWORK_PATH=/Volumes/Development/openscad/../libraries/install/lib
DYLD_LIBRARY_PATH=/Volumes/Development/openscad/../libraries/install/lib

Dyld Error Message:
  Library not loaded: /usr/local/opt/glib/lib/libglib-2.0.0.dylib
  Referenced from: /usr/local/bin/mogrify
  Reason: Incompatible library version: mogrify requires version 7001.0.0 or later, but libglib-2.0.0.dylib provides version 5601.0.0

Binary Images:
       0x10f222000 -        0x10f229fff +mogrify (0) <D1C1EC36-954F-3398-A9A8-359BFC265BD5> /usr/local/bin/mogrify
       0x10f23a000 -        0x10f3d9fff +libMagickCore-7.Q16HDRI.10.dylib (0) <5C2702B4-4251-3599-8A23-781E37A06E51> /usr/local/Cellar/imagemagick/7.1.0-9/lib/libMagickCore-7.Q16HDRI.10.dylib
       0x10f431000 -        0x10f524fff +libMagickWand-7.Q16HDRI.10.dylib (0) <F26FD970-BA6A-3755-B446-05FC40192C85> /usr/local/Cellar/imagemagick/7.1.0-9/lib/libMagickWand-7.Q16HDRI.10.dylib
       0x10f558000 -        0x10f5c3fff +libomp.dylib (0) <EF5296BF-22E3-354B-AFAE-37C90CFC6887> /usr/local/opt/libomp/lib/libomp.dylib
       0x10f602000 -        0x10f63dfff +liblcms2.2.dylib (0) <69269C86-3B95-333C-92F5-DF17C2B887A2> /usr/local/opt/little-cms2/lib/liblcms2.2.dylib
       0x10f65a000 -        0x10f663fff +liblqr-1.0.dylib (0) <2B1F8F34-C32A-3C6B-B2B5-A959A7FD4647> /usr/local/opt/liblqr/lib/liblqr-1.0.dylib
       0x1157c9000 -        0x115864fff  dyld (852.2) <57DB2053-BFD5-3683-97C6-F1DB2A1F1D09> /usr/lib/dyld

Model: MacPro7,1, BootROM 1554.140.20.0.0 (iBridge: 18.16.14759.0.1,0), 12 processors, 12-Core Intel Xeon W, 3.3 GHz, 32 GB, SMC 
Graphics: AMD Radeon Pro W5700X, AMD Radeon Pro W5700X, spdisplays_pcie_device, 16 GB
Memory Module: Slot 3 (Channel E / DIMM 1), 8 GB, DDR4 R-DIMM, 2933 MHz, Micron, 9ASF1G72PZ-2G9E6
Memory Module: Slot 5 (Channel D / DIMM 1), 8 GB, DDR4 R-DIMM, 2933 MHz, Micron, 9ASF1G72PZ-2G9E6
Memory Module: Slot 8 (Channel A / DIMM 1), 8 GB, DDR4 R-DIMM, 2933 MHz, Micron, 9ASF1G72PZ-2G9E6
Memory Module: Slot 10 (Channel B / DIMM 1), 8 GB, DDR4 R-DIMM, 2933 MHz, Micron, 9ASF1G72PZ-2G9E6
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Jul  7 2021 02:54:21 version 9.30.444.17.32.5.70 FWID 01-6551fb93
Bluetooth: Version 8.0.5d7, 3 services, 18 devices, 1 incoming serial ports
Network Service: multiether 2, Ethernet, en1
PCI Card: AMD Radeon Pro W5700X, gpu-controller, Slot-1@9,0,0
PCI Card: pci1002,ab38, sppci_audiodevice, Slot-1@9,0,1
PCI Card: XHC4, sppci_usbxhci, Thunderbolt@200,0,0
PCI Card: XHC2, sppci_usbxhci, Thunderbolt@83,0,0
PCI Card: pci8086,15eb, sppci_thunderbolt, Thunderbolt@201,0,0
PCI Card: pci8086,15eb, sppci_thunderbolt, Thunderbolt@84,0,0
PCI Card: XHC3, sppci_usbxhci, Thunderbolt@24,0,0
PCI Card: pci8086,15eb, sppci_thunderbolt, Thunderbolt@25,0,0
USB Device: USB 3.0 Bus
USB Device: Hub
USB Device: USB Laser Mouse
USB Device: macally Kbd Hub
USB Device: macally Kbd Hub
USB Device: USB-Serial (0001)
USB Device: LCS USB Audio
USB Device: ASM107x
USB Device: TerraMaster
USB Device: TerraMaster
USB Device: TerraMaster
USB Device: TerraMaster
USB Device: USB3.1 Hub
USB Device: USB3.0 Card Reader
USB Device: USB3.1 Hub
USB Device: ASM107x
USB Device: USB2.1 Hub
USB Device: USB2.1 Hub
USB Device: Apple T2 Bus
USB Device: Headset
USB Device: Apple T2 Controller
Thunderbolt Bus: Mac Pro, Apple Inc., 63.3
Thunderbolt Bus: Mac Pro, Apple Inc., 63.3
Thunderbolt Bus: Mac Pro, Apple Inc., 63.3
Thunderbolt Bus: Mac Pro, Apple Inc., 63.3

ChrisCoxArt, 2021-10-04:
I'm not entirely sure if the magick crash is due to multiple versions of glib, or something wrong with the version fetched by the dependencies script.

ChrisCoxArt, 2021-10-04:
Yes, I have tried removing then reinstalling and rebuilding the related components (imagemagic, glib, Qt 5.9.9 and Qt 5.12.2) using the OpenSCAD macosx build scripts.
There was no change in behavior.

thehans, 2021-10-04:
@ChrisCoxArt when including console logs or code snippets in github comments, please wrap the log or code in a markdown code block, so that the log content does not get misinterpreted as other markdown formatting.  I have made such an edit to your original issue comment here as an example.

Multi-line code blocks can be made with three backquotes on a line by themselves immediately before and after the code.

inline code can also be done via a single backquote before and after the code: `like this`.

ChrisCoxArt, 2021-10-05:
It is possible that these were due to a hidden copies of Qt that homebrew left behind.  I'll try and reproduce this on a clean system (but it'll take a while).

kintel, 2021-12-16:
Is it safe to assume that this is resolved by recent updates to the dependency build script?

ChrisCoxArt, 2021-12-16:
Yes, this can be closed now.

closed 2021-12-16 by ChrisCoxArt