openscad/openscad #3927: script translation-update.sh creates "command not found" error on MacOS (and probably Windows)

ChrisCoxArt, 2021-10-05:
The latest version of translation-update.sh calls "realpath" to set CURDIR.
Unfortunately the realpath command is a linux feature, also found on some BSD distributions, but not on MacOS or Windows.
The script continues despite the command not found error.
Unfortunately I don't know enough about the translation functions and scripts to guess at the impact of the error.

```
./scripts/translation-update.sh: line 130: realpath: command not found
msgfmt -c -v -o ./locale/cs/LC_MESSAGES/openscad.mo ./locale/cs.po
244 translated messages, 71 fuzzy translations, 156 untranslated messages.
msgfmt -c -v -o ./locale/de/LC_MESSAGES/openscad.mo ./locale/de.po
467 translated messages, 1 fuzzy translation, 3 untranslated messages.
msgfmt -c -v -o ./locale/es/LC_MESSAGES/openscad.mo ./locale/es.po
218 translated messages, 110 fuzzy translations, 143 untranslated messages.
msgfmt -c -v -o ./locale/fr/LC_MESSAGES/openscad.mo ./locale/fr.po
382 translated messages, 54 fuzzy translations, 35 untranslated messages.
msgfmt -c -v -o ./locale/hy/LC_MESSAGES/openscad.mo ./locale/hy.po
384 translated messages, 52 fuzzy translations, 35 untranslated messages.
msgfmt -c -v -o ./locale/pl/LC_MESSAGES/openscad.mo ./locale/pl.po
410 translated messages, 61 fuzzy translations.
msgfmt -c -v -o ./locale/ru/LC_MESSAGES/openscad.mo ./locale/ru.po
386 translated messages, 51 fuzzy translations, 34 untranslated messages.
msgfmt -c -v -o ./locale/uk/LC_MESSAGES/openscad.mo ./locale/uk.po
283 translated messages, 65 fuzzy translations, 123 untranslated messages.
msgfmt -c -v -o ./locale/zh_CN/LC_MESSAGES/openscad.mo ./locale/zh_CN.po
448 translated messages, 5 fuzzy translations, 18 untranslated messages.
msgfmt -c -v -o ./locale/zh_TW/LC_MESSAGES/openscad.mo ./locale/zh_TW.po
408 translated messages, 50 fuzzy translations, 13 untranslated messages.
```

rcolyer, 2021-10-05:
It introduces a different dependency, but one resolution option is:
```
CURDIR="`python -c "import os; print(os.path.relpath(os.path.realpath('$BASEDIR'), '$TOPDIR'))"`"
```
The behavior should be equivalent.  I don't see any other Python in there, but in general our scripts have a lot of python dependencies already, so it seems to me this should be expected of all people working hands-on with the scripts in the repository.

ChrisCoxArt, 2021-10-05:
Seems reasonable at first glance.

closed 2021-10-06 by t-paul