openscad/openscad #3928: MacOS build fails on M1 based machines because of GMP compilation errors

ChrisCoxArt, 2021-10-05:
The version downloaded is GMP 6.1.2, which works fine on Intel processor Macintoshes.
But GMP 6.2.1 is the first version to support ARM64 on MacOS.

Also, the dependency script forces the host to --host=core2-apple-darwin13.0.0
And that will only work on Intel x86 processors, not the ARM64 based M1.

But I suspect many changes will be needed to make an M1 native build of OpenSCAD.
I'll try some fixes to at least get it building.

ChrisCoxArt, 2021-10-05:
Yep, lots of arch definitions hardcoded to x86_64 that need to be made variable.
I'm testing a fix now.

ChrisCoxArt, 2021-10-05:
Sigh, fontconfig has a crashing bug when built on M1 processors.
Logged as https://gitlab.freedesktop.org/fontconfig/fontconfig/-/issues/298

ChrisCoxArt, 2021-10-05:
Qt, OpenCSG, and Qscintilla use qmake, which seems to be hard coded to arch x86_64. 
I'll have to find how to override that.

HoneypotFX, 2021-11-29:
I've sucesfully built it on M1.
```
git clone git://github.com/openscad/openscad.git
cd openscad
./scripts/macosx-build-homebrew.sh
git submodule update --init
cmake .
make
```

ChrisCoxArt, 2021-11-29:
homebrew is not how the official builds are made.  And I made note that it could build with homebrew above.
Also, several libraries have updated to fix bugs since I filed the report.
And someone else is working on the M1 build, and updating the scripts.

thorrak, 2021-11-29:
> And it is looking like we'll need different versions of Qt for x86_64 and ARM64 (older Qt needed to keep older OS support on x86_64, newer Qt needed for M1 support).

Is this the case? I thought they said Qt 6 was built for universal2 which means it can be targeted for either universal2 binaries or ”shrunk” (can’t remember the official term) to either arm64 or x86_64 if you wanted a smaller binary with targeted platform support.

ChrisCoxArt, 2021-11-29:
Trying to build to two different Qt versions would be a mess to support.  Either openscad sticks with the older Qt for x86_64 to support older OSes (but cannot support M1), or moves to a newer Qt to support the M1, and drops support for older OSes.  From other changes, it looks like openscad will be moving to the newer Qt and dropping older OS support.

thorrak, 2021-11-29:
If Qt supports universal2, then that solves all problems at the cost of a larger binary, I think

ChrisCoxArt, 2021-11-29:
That's not the issue I was talking about. Yes, the latest Qt supports a universal binary.  But moving to the latest Qt means dropping support of older OS versions. THAT was the issue and the decision that needed to be made for how to support the M1.

t-paul, 2021-11-29:
Snapshot builds should be on Qt 5.15.2 now. Qt 6 is not yet an option due to forced C++17 dependency. Also waiting for it to stabilize seems a good idea considering how they decided to release with a partial feature support first.

tcurdt, 2021-12-03:
@ChrisCoxArt  Do I read this correctly: Qt is no longer a blocker for M1 support? It just means older OS versions are no longer supported?

@t-paul So snapshot build could (in theory) include M1 support?

t-paul, 2021-12-03:
In *theory*, *probably* yes. I don't have a full overview about any remaining fixes needed in external components. This also assumes it's possible to build the combined binary. With the fresh restrictions imposed on the build servers we currently can't even build the Intel snapshot.

tcurdt, 2021-12-03:
> In _theory_, _probably_ yes. I don't have a full overview about any remaining fixes needed in external components. This also assumes it's possible to build the combined binary.

With powerpc+intel this was pretty straight forward. You built both and then combined the binaries.
If the homebrew build does indeed already work this certainly should be possible.

> With the fresh restrictions imposed on the build servers we currently can't even build the Intel snapshot.

The build server story seems like a continuous nightmare for openscad.
Maybe we should find a sponsor to host some runners.

t-paul, 2021-12-03:
So far there's always a place to run stuff, problem is that this changes constantly so the time spent on moving things around is a big problem. I gave up on Travis, I'm not sure they still want open source projects even though they claim that. I want to check with CircleCI, I'm not sure what the status is, if it's just the "normal" free plan, or if there's options for open source projects.
Their estimation of a paid plan is not much use as it suggests $150/month which does not make sense, for that money we can get a huge machine. But maybe there's a more sensible option for a reasonable price.

t-paul, 2021-12-03:
...well, that said, a macOS hoster would be interesting. Right now the option would be fully moving to github, but putting all eggs into a single basket did prove problematic in the past already. Also that would break the automatic release to the download page of the website.

kintel, 2021-12-03:
FYI: I'm (slowly) working my way towards a universal build, by doing exactly what @tcurdt indicated; build every library for both archs and merge them into a fat binary, then in the end doing the same for the OpenSCAD binary. Some build systems can do this in a single step, while others need some love. We used do to this for PPC, so all the build system magic is in git history. I did succeed in making and M1 build using the non-homebrew build script, so I quite confident this will eventually work.

kintel, 2021-12-03:
@t-paul do you know what happened with CircleCI? It looks like they imposed a 1h build timeout, and for some reason their build cache seems to have broken, triggering a full dependency rebuild? 
Perhaps my trust in CircleCI's built-in caching was a bit naive; I can look into putting back our own caching system to make this a bit more robust.

thorrak, 2021-12-03:
>  Also that would break the automatic release to the download page of the website.

For what it’s worth, this can be easily accomplished with GitHub actions runners, depending on how the interface to the website is set up.

t-paul, 2021-12-03:
Yes, they decided to limit to 1h for all builds... And put out a blog post after the fact :angry:. It's fair to limit their free stuff, but the communication was quite bad. And we lost Travis as they require a credit card now for the free tier + it seems they expect open source projects to beg for credits as there does not seem to be auto-renewal (last part is from someone experiencing the issue, not something I had, but it matches their announcement from a year go).

On the bright side, there's a rumor, we just need to contact them, and they'll remove the 1h limitation for OpenSource projects, so maybe it's still all good, just needs some mails. :smile:

t-paul, 2021-12-03:
@thorrak No, with CircleCI I can pull the builds from via the public API, so no need to store credentials on the file server. Github does not seem to provide that access without having an AUTH token with huge access rights which I don't really want to have on the file server.

So we would need to setup some sort of build artifact deployment to somewhere (maybe github releases) in the actions and then pull from there. So not impossible but not how I would like to see it (= having public access to the build artifacts per build).

t-paul, 2021-12-03:
Hmm, that "remove limitations" post I found was not related to the current changes, but I've sent out an email anyway. Lets see if they have a proposal.

ChrisCoxArt, 2021-12-04:
> @ChrisCoxArt Do I read this correctly: Qt is no longer a blocker for M1 support? It just means older OS versions are no longer supported?
Yes, we can get an M1 build that works - but it does require using newer versions of several libraries, including Qt, that will remove compatibility with older OS versions.
Kintel seems to be working on it actively (judging by his commit comments).

kintel, 2021-12-12:
Check out #4005 - this updates macosx-build-dependencies.sh to build for whatever is your local architecture. Tested on Intel native, Rosetta 2 and M1 native. If anyone would like to give it a shot before it's merged to master, that would be awesome!

Follow-up after merge is likely: 1) support cross-compiling and 2) support building universal binaries. Both should be relatively low risk, just tedious scripting work.

ChrisCoxArt, 2021-12-16:
Kintel's branch is working well, we can close this report once it is merged into main.

closed 2021-12-16 by kintel

kintel, 2022-01-03:
For people interested in following progress on and/or helping test universal binaries, please follow #4036

PR #4005 (2021-12-16): Support M1 build of dependencies
Fixes #3928

This makes `macosx-build-dependencies.sh` support building for the local architecture on either x86_64 or arm64.
Cross compilation not yet supported.