openscad/openscad #3933: Issue in latest macOS nightly builds (from 2010-10-04)

kapfab, 2021-10-10:
Latest macOS nightly builds (starting from 2010-10-04) are missing `libicu` libs and embed a `libboost_regex.dylib` with incorrect links to local development paths:
```
	@rpath/libboost_regex.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/local/lib/libicudata.58.dylib (compatibility version 58.0.0, current version 58.2.0)
	/opt/local/lib/libicui18n.58.dylib (compatibility version 58.0.0, current version 58.2.0)
	/opt/local/lib/libicuuc.58.dylib (compatibility version 58.0.0, current version 58.2.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 902.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.100.1)
```
Previous versions (until 2010-10-03) were fine:
```
	@rpath/libboost_regex.dylib (compatibility version 0.0.0, current version 0.0.0)
	@executable_path/../Frameworks/libicudata.58.dylib (compatibility version 58.0.0, current version 58.2.0)
	@executable_path/../Frameworks/libicui18n.58.dylib (compatibility version 58.0.0, current version 58.2.0)
	@executable_path/../Frameworks/libicuuc.58.dylib (compatibility version 58.0.0, current version 58.2.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 902.1.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1281.100.1)
```

These builds might run on Intel Macs used for development, but not on Apple Silicon.

t-paul, 2021-10-16:
Can you please test with this build: https://8963-1049088-gh.circle-artifacts.com/0/build/OpenSCAD-2021.10.16.dmg

t-paul, 2021-10-17:
Thanks for testing! I don't know what's going on with `libicu` but it seems it was picking up libraries from somewhere else, so https://github.com/openscad/openscad/commit/4319fbd3f159be3bd29a8e658a2e30218ea2f17b#diff-78a8a19706dbd2a4425dd72bdab0502ed7a2cef16365ab7030a5a0588927bf47R151-R154 probably fixed that.

Change is merged now, so the automatic snapshot builds from the download page should be fixed too. Please reopen if you observe any issue with that.

closed 2021-10-17 by t-paul