openscad/openscad #3967: STL export: Command line produces different output than GUI

thohh, 2021-11-04:
I'm building masks to slice a bigger circular model into printable pieces. After switching from GUI to command line mode I got artifacts in the output. I'm totally open to a different way to build an arc in openscad, but I think this inconsistency should be fixed anyways.

Opening the test.scad file in GUI mode, rendering with F6 and exporting with F7 produces the expected result. 
Running in command line mode with -o produces different result.

Happens with 2019.5 and 2021.01.
Please find test model, STL files for GUI and console mode, and console outputs attached.
I was not able to get any console output for 2019.5 in command line mode (-o out.echo produced an empty file)
[bugreport_20211104_stl_cmd_vs_gui.zip](https://github.com/openscad/openscad/files/7473523/bugreport_20211104_stl_cmd_vs_gui.zip)

t-paul, 2021-11-04:
GUI now defaults go binary STL, command line remains on ASCII for compatibility reasons. to get same output use `--export-format binstl`.

closed 2021-11-04 by t-paul

thohh, 2021-11-04:
Thanks for the hint with export options. Still same problem.
[test_cmd_2021.01_bin.stl.zip](https://github.com/openscad/openscad/files/7473623/test_cmd_2021.01_bin.stl.zip)

t-paul, 2021-11-04:
Difference with coincident faces (which also cause the flickering artifacts in GUI) always has a risk for unstable output. Different geometry between GUI and command line is indeed a bit strange though.

In this specific case the better option is to just translate the expected part before the `rotate_extrude()`:
```
rotate_extrude(angle=a) translate([r1, 0]) square([r1-r,h]);
```

thohh, 2021-11-04:
Thanks for the different solution. Works for me :-)

nophead, 2021-11-04:
The different result is probably the preview caching different geometry issue. Flushing the cache before F6 in the GUI might get the same result.

thohh, 2021-11-04:
Confirmed, flushing the cache in GUI produces the same output.