openscad/openscad #3971: ctrl + q leaves changes unsaved

UBaer21, 2021-11-07:
quitting with ctrl+q didn't pop up dialog  for unsaved changes it just closes reverting to last save.

thehans, 2021-11-07:
I'm not able to reproduce with the latest nightly.  Please include OpenSCAD version, OS, and specific steps to reproduce (i.e. are you starting from a fresh file, or existing file, did you have multiple files open, etc.)

t-paul, 2021-11-07:
Strange, happens for me too with latest nightly (2021.10.31.nightly (git caadce2)) - Just open with new empty editor, type something, CTRL+Q.

UBaer21, 2021-11-07:
I just installe 21.10.19 windows 64bit.  When opening a new file  (or existing) and change anything "ctrl+q" closes the program without warning. While ALT+F4 or the X  will pop up "unsaved... "

whosawhatsis, 2021-11-15:
Command-Q is quitting without a prompt for unsaved changes on the Mac too, in recent nightlies (but not the most recent release)

ChrisCoxArt, 2021-12-27:
Problem confirmed -- the text windows are not being queried about unsaved changes, they've not even getting the events.(can't find MainWindow::closeEvent being called on Quit).

ChrisCoxArt, 2022-01-19:
Can't catch QTEvent::Quit, or filter for it. But we can intercept in MainWindow::quit().
We try to send a close event to the application in  MainWindow::quit(), but it never reaches the windows (not even MainWindow).
But we can call this->closeEvent(&ev);  and then filter on ev.isAccepted() afterward.

ChrisCoxArt, 2022-01-19:
I'm not sure this is the best approach, but it mostly works.
```
// NOTE - this gets called twice on exit: once by Qt, and once by ~MainWindow
void MainWindow::quit()
{
    QCloseEvent ev;
    QApplication::sendEvent(QApplication::instance(), &ev);     // BUG! #3971 MainWindow::closeEvent(QCloseEvent *event) does not get called from this!
    this->closeEvent(&ev);      // solve #3971 by explicitly calling close and asking about unsaved documents, but sometimes pops question twice if you choose "don't save"

    if (ev.isAccepted()) {      // if cancelled by window/tab save prompt, don't quit
        QApplication::instance()->quit();
// FIXME: Cancel any CGAL calculations
#ifdef Q_OS_MAC
        CocoaUtils::endApplication();
#endif
    }
}
```

closed 2025-10-25 by kintel