openscad/openscad #3972: rotate_extrude misaligned for angle=360

UBaer21, 2021-11-10:
the corner is on -x  while a circle is on +x 
when the angle is reduced it is on +x
```
  rotate_extrude($fn=5)square();//180° wrong
  
  rotate_extrude($fn=6,angle=359.9)square();
  circle(1,$fn=5);
```
[image]

resulting in issues when you rotate_extrude something around a cylinder.

t-paul, 2021-11-10:
I think that's intentional for backward compatibility, see discussion in https://github.com/openscad/openscad/pull/1478

UBaer21, 2021-11-10:
how can this be compatible to anything if All Objects center their point on +x  (sphere, circle, cylinder)?
And as soon an angle <360 is used it is jumping 180° into the right position.
As $fa is limited to 72 (not or 120) we end up with 5-gon elements which are not matching.

t-paul, 2021-11-10:
It probably should depend on the `angle` parameter having a value or not. The compatibility would be for scripts written for older OpenSCAD versions that don't have the `angle` parameter at all.

UBaer21, 2021-11-10:
But it doesn't make a difference if angle has a value or if the value is 360 it is always wrong.
Only when angle gets <360 and the loop opens it has it start point correctly on [+X , 0].
So older scripts may have corrected this by an additional rotation of 180 - but this would mean we can never correct the problem and new scripts will also need to correct for this. 
So at least the polyhedron need to be turned by 180 when angle == 360 (or 0).

thehans, 2021-11-15:
Duplicate of #1886

closed 2021-11-15 by thehans

thehans, 2021-11-15:
Also see issue #3142 for backwards compatibility discussion.