openscad/openscad #3975: Segfault with autocompletion window upon keyboard input on KDE Wayland

phrxmd, 2021-11-12:
I'm getting segfaults in OpenSCAD 21.01 when entering alphanumeric characters in the editor. This is on KDE Plasma 5.23.2 on Wayland with Qt 5.15.2.

I can avoid it by either launching OpenSCAD in XWayland with `QT_QPA_PLATFORM=xcb`, or by disabling autocompletion.

I presume it happens when the autocompletion window opens, because it does not happen with autocompletion disabled, and it usually happens when entering the first alphanumeric character of something: as in I would start a new file, type `a`, or even `/* a`, and when I reach the `a` it segfaults.

Sometimes it will let me enter characters for a few minutes before it happens, but it inevitably happens.

Console output:
```
> openscad
GLEW Error: Unknown error
xkbcommon: ERROR: /home/reichmuth/.XCompose:11:13: unrecognized keysym "?" on left-hand side
xkbcommon: ERROR: /home/reichmuth/.XCompose:161:17: unrecognized keysym "_" on left-hand side
xkbcommon: ERROR: /home/reichmuth/.XCompose:162:17: unrecognized keysym "^" on left-hand side
xkbcommon: ERROR: /home/reichmuth/.XCompose:163:17: unrecognized keysym "_" on left-hand side
xkbcommon: ERROR: /home/reichmuth/.XCompose:164:17: unrecognized keysym "^" on left-hand side
xkbcommon: ERROR: /home/reichmuth/.XCompose:165:17: unrecognized keysym "_" on left-hand side
xkbcommon: ERROR: /home/reichmuth/.XCompose:166:17: unrecognized keysym "^" on left-hand side
xkbcommon: ERROR: /home/reichmuth/.XCompose:262:17: unrecognized keysym "number" on left-hand side
xkbcommon: ERROR: /home/reichmuth/.XCompose:267:28: unrecognized token
xkbcommon: ERROR: /home/reichmuth/.XCompose:308:26: unrecognized keysym "four" on left-hand side
xkbcommon: ERROR: /home/reichmuth/.XCompose:309:26: unrecognized keysym "eight" on left-hand side
xkbcommon: ERROR: /home/reichmuth/.XCompose:309:26: too many errors
xkbcommon: ERROR: /home/reichmuth/.XCompose:309:26: failed to parse file
qt.qpa.input.methods: failed to create compose table
[1]    7684 segmentation fault (core dumped)  openscad
```

Versions:
OpenSCAD: 21.01 (OpenSUSE RPM 021.01-3.3)
Operating System: openSUSE Tumbleweed 20211110
KDE Plasma Version: 5.23.2
KDE Frameworks Version: 5.87.0
Qt Version: 5.15.2
Kernel Version: 5.14.14-2-default (64-bit)
Graphics Platform: Wayland

PaulWay, 2021-12-28:
I can confirm this is happening to me as well.

Compiled on commit 04aa887cd3e842fa7e117817c727c53ff3876d72.
Versions:
Operating system: Fedora 35 x86_64
Qt Version: 5.15.2
Kernel: 5.15.10-200
Graphics Platform: Wayland

PaulWay, 2021-12-28:
OK, well, weirdly, my version doesn't seem to do this any more.  I've tried the compiled version, and the RPM version from `openscad-2021.01-4.fc35.x86_64.rpm`.  I've got a debug-enabled version to try if I can work out the actual steps to reliably replicate this problem.