openscad/openscad #3980: OpenSCAD can't render, upon import(), the STL it created

krhackbarth, 2021-11-19:
The files referenced are available at: https://www.thingiverse.com/thing:5139762

I can import() the file “afraid.stl” into OpenSCAD and it will display when Preview’d.

I took the same file and broke it into “parts” using PrusaSlicer, deleted one of the parts, and saved the result as “afraid1.stl”

I then import() that file and it doesn’t display when Preview’d.

I can render the model and save it as “Untitled.stl”. This file displays properly in PrusaSlicer but if I attempt to “import()” it into OpenSCAD, nothing
displays when Preview’d.

krhackbarth, 2021-11-19:
Note that "afraid.stl" was created from an SVG file using OpenSCAD.

ChrisCoxArt, 2022-01-16:
In OpenSCAD, go to view -> View All.
The location changed on the modified STL files, and you have to zoom out and translate over a bit to see the objects.
Or you can add translate( [-130,-100,0] )before the import call.

closed 2022-03-04 by rcolyer

hamster4430, 2023-06-25:
Just a small tip for everyone who struggles with a same issue:
Use hotkeys Ctrl+Shift+V to zoom out and see all parts of a project in OpenSCAD