openscad/openscad #3982: Suggestion: F7 does "preferred export".

rewolff, 2021-11-21:
I often work with 3D models and F7-export-STL is very convenient. 

I now that I'm working with 2D stuff, I find myself hitting F7 and getting "top level object is not a 3D object". 

I think it would be useful if: 
* when the top level opbject is 2D, then F7 would do: "preferred 2D export". 
* that "preferred 2D export" and "preferred 3D export" would be configurable. 

I can imagine the frustration that those working with say AMF have that they don't have a hotkey like STL-users have. 
And similarly, I respect others who's 2D workflow doesn't use DXF.

t-paul, 2021-11-21:
Right, in general we need to get #3338 brushed up. But it seems useful to have this kind of "Smart-Export" action.

jordanbrown0, 2024-11-23:
The toolbar export button is controlled by a preference.  It seems like it would be straightforward and desirable to have F7 be controlled by that same preference.

jordanbrown0, 2025-03-09:
We now have two export buttons, currently one 2D and one 3D.  There's a request to make there be two 3D export buttons (perhaps allowing the two buttons to each be either 2D or 3D), see #5714.  Such a change would mean that the button's exporter selection would not be enough to set F7's export.  We'd need a separate selection.

It's not exactly the target market, but we might also consider a default 2D export.  Perhaps F7 would export to a specified 3D format if you have a 3D object, and to a specified 2D format if you have a 2D object.