openscad/openscad #3983: Error with Command Line .png export in a Docker container

ilya-pro, 2021-11-21:
I've been tried different configurations to render .scad to .png in a Docker container, but get the same error:
```
Unable to open a connection to the X server.
DISPLAY=
Can't create OpenGL OffscreenView. Code: -1.
```
Dockerfile:
```
FROM ubuntu:20.04

RUN apt-get update --fix-missing -qq
RUN apt-get -y -qq install ca-certificates xvfb imagemagick unzip
RUN apt-get -y install openscad
```
Build:  `docker build -t openscad:ubuntu20.04xvfbExt .`
Run command on Windows: 
`
docker run -v C:/imgUbuntuX11/src:/openscadVol openscad:ubuntu20.04xvfbExt openscad -o /openscadVol/snowflake.png /openscadVol/snowflake.scad`

Export to .svg and .stl works fine.

Scopeuk, 2021-11-21:
The png render (and other image files) rely on rendering into the frame buffer and then exporting to file.
see https://github.com/openscad/openscad/blob/master/tests/virtualfb.sh for details on how the test environment works around this (it appears you have installed the perquisites)

ilya-pro, 2021-11-21:
@Scopeuk  Thank for your explain. So, should I run such script (virtualfb.sh) in the container after it start and before run openscad command?

t-paul, 2022-02-19:
Docker images are at https://hub.docker.com/r/openscad/openscad now.

closed 2022-02-19 by t-paul