openscad/openscad #3985: Rendering only 1 cpu core

djfix13, 2021-11-24:
with the latest realese the long rendering time intrigued me; I saw in the taskmanager that only one core is used... why??
[image]

t-paul, 2021-11-24:
See #391.

closed 2021-11-24 by t-paul

djfix13, 2021-11-24:
sorry for question but I read the project goes on 2013, 8 year late...notingh happen?

jordanbrown0, 2021-11-24:
Many things have happened since 2013.  Just not that, because it's really hard.  Feel free to take a stab at it.

jasonrhardman, 2022-09-20:
How far have people come? Has it even been pursued beyond a cursory assessment that it's too hard? I want to take a stab at it, but need to know where the giant's shoulders are so I can stand on them.

JayRugMan, 2022-09-20:
How far have people come? Has it even been pursued beyond a cursory assessment that it's too hard? I want to take a stab at it, but need to know where the giant's shoulders are so I can stand on them.

t-paul, 2022-09-20:
Active work in #4293.