openscad/openscad #3986: openscad fails to open on Windows when installed on drive mounted as folder

Dromantor, 2021-11-25:
This is a error I already encountered a long time ago but never bothered to report it / really use openscad.

If you install openscad into a directory that lies on a drive mounted as directory on Windows ...

Something like this in my case:
```
Drive A     => K:\ (HDD)
Drive B     => K:\Programme\ (SSD, the drive is mounted as the directory "Programme" under K:\)
Install dir => K:\Programme\openscad\
```

... openscad fails to launch.
The following error is reported when launched from the command line (using the openscad.com file):
```
K:\Programme\openscad>openscad.com
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::read_symlink: Die Anforderung wird nicht unterst³tzt: "K:/Programme"
```

Some speculation: It seems your application or the boost library mistakes the mounted drive as a symlink (which it is not).

As a workaround I tried to symlink the installation directory to the drive root ...
`K:\openscad\ => K:\Programme\openscad\`
... but nope:
```
K:\openscad>openscad.com
terminate called after throwing an instance of 'boost::filesystem::filesystem_error'
  what():  boost::filesystem::read_symlink: Die Anforderung wird nicht unterst³tzt: "K:"
```

As soon as there is a directory-mounted drive involved somewhere it fails.

t-paul, 2021-11-25:
Known issue #1309 but there's no solution yet. The boost bug tracker, linked in that ticket, shows the possible root cause as still open.

t-paul, 2021-11-25:
Thanks for the pointer to the newer boost ticket, I'll close this one as duplicate. With some luck that fix works once the windows build is fixed.

closed 2021-11-25 by t-paul