openscad/openscad #3989: Problems importing DXF files in NanoCAD 5

PRouzeau, 2021-11-30:
Recent OpenSCAD snapshots DXF exports polylines instead of line sequences (tested on October 21 Snapshot: see pull request https://github.com/openscad/openscad/pull/3869)
These new DXF files cannot be imported in NanoCAD 5 (the free version of NanoCAD, issued in 2013).
This NanoCAD version can export and import polylines it created itself.

An Openscad DXF export file of a 100x100 square, which does not import in NanoCAD:
```
999
DXF from OpenSCAD
  0
SECTION
  2
HEADER
  9
$ACADVER
  1
AC1006
  9
$INSBASE
 10
0.0
 20
0.0
 30
0.0
  9
$EXTMIN
 10
-50
 20
-50
  9
$EXTMAX
 10
50
 20
50
  9
$LINMIN
 10
-50
 20
-50
  9
$LINMAX
 10
50
 20
50
  0
ENDSEC
  0
SECTION
  2
TABLES
  0
TABLE
  2
LTYPE
 70
1
  0
LTYPE
  2
CONTINUOUS
 70
64
  3
Solid line
 72
65
 73
0
 40
0.000000
  0
ENDTAB
  0
TABLE
  2
LAYER
 70
6
  0
LAYER
  2
0
 70
64
 62
7
  6
CONTINUOUS
  0
ENDTAB
  0
TABLE
  2
STYLE
 70
0
  0
ENDTAB
  0
ENDSEC
  0
SECTION
  2
BLOCKS
  0
ENDSEC
  0
SECTION
  2
ENTITIES
  0
LWPOLYLINE
100
AcDbEntity
  8
0
100
AcDbPolyline
 90
4
 70
1
 10
50
 20
50
 10
-50
 20
50
 10
-50
 20
-50
 10
50
 20
-50
  0
ENDSEC
  0
EOF
``` 
The Ouptut of same square by Nanosoft is quite large  but I have cancelled many lines of it and reduced to a file which can still import in NanoCAD:

```
  0
SECTION
  2
HEADER
  9
$ACADVER
  1
AC1027
  9
$ACADMAINTVER
 70
     8
  9
$REQUIREDVERSIONS
160
                 0
  9
$INSBASE
 10
0.0
 20
0.0
 30
0.0
  9
$SKPOLY
 70
     0
  0
ENDSEC
  0
SECTION
  2
TABLES
  0
TABLE
  2
VPORT
  5
26
330
0
100
AcDbSymbolTable
 70
     1
  0
VPORT
  5
2B
330
26
  0
ENDTAB
  0
TABLE
  2
LTYPE
  5
B
330
0
100
AcDbSymbolTable
 70
     1
  0
LTYPE
  5
C
330
B
  0
ENDTAB
  0
TABLE
  2
LAYER
  5
2
330
0
100
AcDbSymbolTable
 70
     2
  0
LAYER
  5
1
330
2
100
AcDbLayerTableRecord
  2
0
 70
     0
 62
     7
  6
Continuous
370
    -3
390
3
347
6
348
0
  0
ENDTAB
  0
TABLE
  2
STYLE
  5
F
330
0
100
AcDbSymbolTable
 70
     3
  0
STYLE
  5
E
330
F
  0
ENDTAB
  0
TABLE
  2
VIEW
  5
24
330
0
100
AcDbSymbolTable
 70
     0
  0
ENDTAB
  0
TABLE
  2
UCS
  5
25
330
0
100
AcDbSymbolTable
 70
     0
  0
ENDTAB
  0
TABLE
  2
APPID
  5
9
330
0
100
AcDbSymbolTable
 70
    15
  0
APPID
  5
8
330
9
100
AcDbSymbolTableRecord
100
AcDbRegAppTableRecord
  2
MC_VERSION_DATA
 70
     0
  0
ENDTAB
  0
TABLE
  2
DIMSTYLE
  5
12
330
0
100
AcDbSymbolTable
 70
     1
100
AcDbDimStyleTable
  0
DIMSTYLE
105
11
330
12
100
AcDbDimStyleTableRecord
  2
Standard
 70
     0
 40
100.0
 41
2.5
 42
0.5
 43
6.0
 44
0.0
 50
45.000001
 73
     0
140
2.5
141
2.5
143
25.4
147
1.0
178
     0
340
E
345
D
346
D
347
D
  0
ENDTAB
  0
TABLE
  2
BLOCK_RECORD
  5
15
330
0
100
AcDbSymbolTable
 70
     9
  0
ENDTAB
  0
ENDSEC
  0
SECTION
  2
ENTITIES
  0
LWPOLYLINE
  5
53
330
16
100
AcDbEntity
  8
0
100
AcDbPolyline
 90
4
 70
1
 10
50
 20
50
 10
-50
 20
50
 10
-50
 20
-50
 10
50
 20
-50
  0
ENDSEC
  0
EOF

```

kintel, 2024-04-01:
@PRouzeau  Is this still a problem? Was it reported to NanoCAD?

OratHelm, 2024-06-08:
@kintel This seems to be related to #3869 and #4268, which have not been resolved. The problem doesn't seem to come from NanoCAD but rather from the polyline handling in OpenSCAD's DXF export. However, I don't know enough about DXF syntax to understand the problem better...

Regarding the two examples of DXF files given by @PRouzeau, neither seems to open in AutoCAD 2024. So I'm not sure that the examples are very good. Here's a 100x100 square made in AutoCAD 2024 and exported in DXF, if that helps: 
[square_autocad2024_100x100.dxf.txt](https://github.com/user-attachments/files/15748093/square_autocad2024_100x100.dxf.txt)

kintel, 2024-12-22:
Closing as fixed due to no further reports, feel free to open a new ticket to further investigate this.

closed 2024-12-22 by kintel