openscad/openscad #3990: Rendering individual Polyhedon OK but multiples gives "mesh not closed" error: solved with hull()

bakerjvb, 2021-12-02:
Rendering an individual tooth created with a polyhedron works fine.
but multiples fail to render with error "mesh not closed"
This can be resolved by "hull"ing the Polyhedron.
 Eg code below fails to render:
tw=6;//width of tooth z
tl=2;//length of tooth x
th=4;//height of tooth -y

ratchet();
ratchettooth();

module ratchet(){
for (i =[40:tl:98])
translate ([i,-2,-13])
ratchettooth();}

module ratchettooth(){
polyhedron (
points = [[0,0,0],[0,-th*0.75,0],[tl,-th*0.25,0],[tl,0,0],[0,0,tw],[0,-th*0.75,tw],[tl,-th*0.25,tw],[tl,0,tw]], 
faces =[[0,1,2,3],[4,5,6,7],[1,2,6,5],[0,1,5,4],[0,3,7,4],[3,2,6,7]]
);}

Code below renders correctly (the only difference is the hull() in front of the polyhedron()):
tw=6;//width of tooth z
tl=2;//length of tooth x
th=4;//height of tooth -y

ratchet();
ratchettooth();

module ratchet(){

for (i =[40:tl:98])
translate ([i,-2,-13])
ratchettooth();}
module ratchettooth(){
**hull()**polyhedron (
points = [[0,0,0],[0,-th*0.75,0],[tl,-th*0.25,0],[tl,0,0],[0,0,tw],[0,-th*0.75,tw],[tl,-th*0.25,tw],[tl,0,tw]], 
faces =[[0,1,2,3],[4,5,6,7],[1,2,6,5],[0,1,5,4],[0,3,7,4],[3,2,6,7]]
);}
I hope this helps others but it may point to a structural failure to render.

UBaer21, 2021-12-02:
your code doesn't work - i assume you have to add a + between  `-th` and the number.
When doing so i get geometry with wrong aligned faces (use thrown together view F12) - This probably causes your issues.
Reorient the faces by changing the order

```
module ratchettooth(){
polyhedron (
points = [[0,0,0], [0, -th +0.75, 0], [tl, -th +0.25, 0],[tl, 0, 0],[0, 0, tw], [0,-th +0.75,tw], [tl, -th +0.25, tw],[tl, 0, tw]],
faces =[
  [0,1,2,3],
  [7,6,5,4],
  [5,6,2,1],
  [4,5,1,0],
  [0,3,7,4],
  [3,2,6,7]
  ]
);}
```

t-paul, 2021-12-02:
Most of the faces have wrong winding order (and will show up in pink-ish color in ThrownTogether view mode).

https://en.m.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#Debugging_polyhedra

closed 2021-12-02 by t-paul

bakerjvb, 2021-12-03:
Which begs the question: why does this render one perfectly well but fails
with multiples?

bakerjvb, 2021-12-03:
Thank you for this, but the question remains: why does this render a single
polydon perfectly well but fail at multiples?

MichaelAtOz, 2021-12-03:
With one object there is no CSG operation so CGAL does not get called, it just makes the mesh (badly as instructed). 
With multiples or just an extra `cube()` that forces a `union()` CGAL gets calls, and finds the bad polygs.

bakerjvb, 2021-12-03:
Ok, thanks for the clarification.

jordanbrown0, 2021-12-03:
It would be tempting to somehow force CGAL to get involved, so that we don't have the "works if it's by itself" case.

t-paul, 2021-12-03:
Just forcing that would mean there's no way for debugging the issue. Some sort of better visualization and debugging support would be another option.

jordanbrown0, 2021-12-03:
F5 preview would still work, but F6 would fail - as happens if you have another object in the model today.  You could use F5 for visualization and debugging.

jordanbrown0, 2021-12-03:
My first thought was to always include a microscopic cube in the model, but that would be a problem for STL export.

t-paul, 2021-12-03:
Hmm, ok, just for F6 could be an option. It would impose the extra time spend on all designs though.

jordanbrown0, 2021-12-03:
It would only be an extra cost for single-polyhedron models.  Anything with more than that already involves CGAL.

The idea, in case it wasn't clear, was to turn "it works by itself but fails when I add anything else" into "it fails"... simplifying the diagnostic process for those who don't understand that the single-polyhedron case is special.

UBaer21, 2021-12-03:
is it possible that the issue with wireframe view (F11) is related to the
CGAL involvement? As this is also only working if two objects are there.

nophead, 2021-12-03:
Yes the wireframe only works when CGAL produces the result.

The polyhedron problem is nearly always inverted faces. It should
definitely validate its input.

UBaer21, 2021-12-03:
would be nice if we could have a better error msg,
"

The given mesh is not closed!
"
is likely misunderstood as it is closed but with inverted normals - so it
would be great if we could differntiate between a missing face (not closed)
and inverted faces,  if possible indicate which face is wrong (by number)
or automatically force F12 view in those cases.

jordanbrown0, 2021-12-04:
I think it might be possible to report missing faces and faces with inconsistent twist.

Each edge needs to be used twice.  If it's less than twice, the polyhedron is not closed.  If it's more than twice, the polyhedron is not manifold.

The two uses need to be in opposite directions.  That is, if one is [A,B] the other must be [B,A].  If they are in the same direction, there's a winding inconsistency.

These two rules won't detect all errors, but they'll detect a fair number, and they should be able to give decent diagnostics.

bakerjvb, 2021-12-04:
Thank you guys, helpful error messages are just what we need.
Jordan: your rules are quite simple to understand, can they be
incorporated into the cheatsheet on polygons?
Best regards, Julian

jordanbrown0, 2021-12-04:
They're easy for a computer to understand, but I don't think they'd be all that helpful to humans.  Humans don't want to be inspecting every edge.

For humans, the rules are
* Polyhedron has to be closed.  It has to be completely covered in faces.
* Winding order has to be consistent, and preferably clockwise when looking into the face.
* Polyhedron has to be non-self-intersecting.  The faces must not cross each other.
* Polyhedron has to be manifold.  Only two faces can meet at each edge.

The first two are the usual problems, and are discussed in the documentation that the cheat sheet points at:  https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#polyhedron

There's a section on debugging polyhedra at
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Primitive_Solids#Debugging_polyhedra

Mostly, if you're hand-constructing a polyhedron, switch on View/Thrown Together and add one face at a time until you're done, ensuring that each face is yellow (not purple) when viewed from outside the polyhedron.  It's simplest if all of the faces are triangles, but if they aren't then make sure that they aren't twisted and that the points are all on the same plane.  When you are done, if you can see any purple it's wrong.

bakerjvb, 2021-12-05:
Dear Jordan,
Yes, but we poor users don't understand some of the words, like winding
order and the implications and yet we have to generate the faces in a
polygon using the correct sequence and order of edges.
So I do think incorporating your elegantly phrased tests would make our job
easier and the method of achieving  manifoldness clearer.
Best wishes.
Julian

jordanbrown0, 2021-12-06:
Sorry for using jargon.

Winding order means the order that you specify the vertices of a particular face.

Please look over the documentation that I pointed at and see if it can be made clearer.  It does intend to cover these issues in a human-friendly way.

bakerjvb, 2021-12-08:
Oh I'm ok with this (now!) My ratchet rack with multiple teeth is made and
the assembly (a big sliding clamp) is doneI

I just thought your two tests were very elegant and understandable.
Have a great Christmas.
Julian