openscad/openscad #4001: Torus doesnt Render with F6

Sploopiii, 2021-12-09:
rotate([90,0,0]) translate([160,240,-220]) cylinder(h = 100, r = 80, $fn = 50);
	rotate([90,0,0]) translate([160,240,-120]) cylinder(h = 40, r1 = 80, r2 = 50, $fn = 50);
	union() {
	rotate([90,0,0]) translate([160,240,-220]) rotate_extrude(convexity = 20, $fn = 100) translate([40, 0, 0]) circle(r = 40, $fn = 100) ; 
	}
	rotate([90,0,0]) translate([160,240,-260]) cylinder(h = 80, r= 40,$fn = 50);
	
	F5:
[image]

        F6:
        
[image]

t-paul, 2021-12-09:
The torus is created with a circle touching the middle at exactly a single point. This produces invalid geometry.

As simple way around that is to fill the shape in 2D already, instead of using the cylinder to fill the space:
```
rotate([90,0,0]) translate([160,240,-220]) rotate_extrude(convexity = 20, $fn = 50) {
    translate([40, 0]) circle(r = 40, $fn = 50);
    translate([0, -40]) square([40, 80]);
}
```

Sploopiii, 2021-12-09:
Yup, this works, thank you!

closed 2021-12-09 by t-paul