openscad/openscad #4003: WIndows 64-bit 2021.12.07 fails to start due to missing DLL entry point

jcwren, 2021-12-09:
OpenSCAD 2021.12.07 fails to start on Windows 7 64-bit with an error message of "The procedure entry point CreateEventA could not be located in the dynamic link library api-ms-win-core-synch-l1-2-0.dll".  
[image]

t-paul, 2021-12-09:
Hmpf, not sure if https://github.com/mxe/mxe/pull/2660 means the default is Win10 now. We have to find out if there's a way around that.

jcwren, 2021-12-09:
Hopefully. Not supporting Windows 7 would be double-plus ungood. It still represents about 18% of all machines running any version of Windows.

nophead, 2021-12-16:
Which is the last snapshot that works on Win7 please?

t-paul, 2021-12-16:
Probably https://files.openscad.org/snapshots/Win7/

thehans, 2021-12-16:
> Hopefully. Not supporting Windows 7 would be double-plus ungood. It still represents about 18% of all machines running any version of Windows.

Win7 was released over 11 years ago and is coming up on 2yrs past End of Life.  How many years **past** EOL is reasonable for an open source project with very limited development resources to support?  I would argue the answer is 0.

t-paul, 2021-12-16:
While I do agree in general, it's a bit annoying that this randomly broke with the MXE upgrade and maybe without strict need for that from OpenSCAD side.
As the linked change was introduced due to a build failure of `qtbase` there might be no way to get things back even with the hinted build option. At this point it's not clear to me how and where to set that though.

nophead, 2021-12-16:
I agree with @thehans and 0. The problem is Windows peaked at XP and has gone downhill since then. I reluctantly moved to Win7 and had to replace a perfectly working scanner with a new model. It has less capable optics, so can now only scan flat sheets, whereas I could scan PCB's with the old one. The software interface that came with is better though.

I looked at upgrading to Win10, which is already replaced, but it said I needed a new computer and presumably I would need new peripherals as well. It also updates itself without permission, so is no use to me. I don't need a new PC as they don't get much faster for single threaded programs. And I have 3, so I would need to replace all of them. Bonkers as they still work flawlessly with Win7.

I don't need a new OS because all it needs to do is launch applications and no reason that needs to change.

I might move to Linux eventually but I have programs that only run on Windows, so perhaps I will just stick with old versions programs and stay on Win7 until I die. I still slice all my models with Skeinforge which hasn't been supported for about 10 years. Still works as well as it ever did.

MichaelAtOz, 2021-12-17:
I have a few systems and was Win7 exclusively for a while, but needed a new one as the box was failing. This was my everyday PC. It started failing fast, so I migrated it to a VMware Player VM (free for personal use, there are alternatives), running on my other 8 thread 16GB box. The VM is 8GB 6 CPU works well*.

I also wanted to try Linux, but wasn't ready to let go of my Outlook/Office and a few others, and am pretty experienced with Windows Admin/internals, and next to useless with Linux technicalities buy can do with Dr Google.

At that time the current Intel CPUs (7th Gen) only supported Win10, I bought a 6th Gen NUC Skull Canyon 8 thread 32GB, which could have run Win7, but decided to try Linux first, and run RDP (xfreerdp) to access the VM which is still on the other box. I was going to move the VM to the Debian NUC but haven't felt the need yet. Also haven't found the time or need to change email client, I do use LibreOffice sometimes, but still prety much Win7 for day to day non-browsing.

So I still have my Everyday Win7 box (VM) but also do browsing, media, and progressively more on Debian. Typing this via xdrfree on Debian to browser on W7-VM.

I suspect Linux will support older peripherals, but you'd need to see if you are happy with the Linux software.

*works well, until #3719. 

For testing I also have W10 in a Boxes VM on Debian, it runs 2021.01 well, but same #3719 problem with later versions.
Later versions work well on Debian, but apart from nightlies, playing with specific versions with Snap is a PITA. That could just be my lack of Linux skills.

t-paul, 2021-12-18:
Declaring Win7 minimum version on just the OpenSCAD build did not help unfortunately (#4012).

kintel, 2022-01-15:
The MXE setting (--with-default-win32-winnt=0x0600) should mean Windows Vista. I haven't used Windows since Win 2000, so my experience is a bit outdated ;)

t-paul, 2022-02-20:
Going back to older gcc version for the whole MXE environment did not help either. However I found that SoapUI includes a JDK version that ships a whole bunch of those DLLs, incuding the one flagged here. So maybe the option is not changing the build, but shipping the DLLs if that's permitted somehow.

MichaelAtOz, 2022-07-15:
Is OpenSCAD on Windows 7 officially dead?

jordanbrown0, 2025-03-09:
I think it's dead, but I think that signing its death certificate is above my pay grade.  @kintel, @t-paul ?

t-paul, 2025-03-09:
Yes, I tried multiple times to convince the build to not require the new API stuff but that all failed. As nobody else seems to desire spending time on this, I'm calling it dead.

closed 2025-03-09 by t-paul

kintel, 2025-03-09:
Qt requires Windows >= 10. While technically possible to backport, it's too much work to maintain, but if someone wants to put in the effort, #ifdef their way to success, and maintain an older Windows build, feel free to step up :)