openscad/openscad #4004: macOS M1 build falls back to SW rendering

kintel, 2021-12-10:
When running a native M1 build of OpenSCAD, it looks like OpenGL falls back to SW rendering in some cases.
Cmd-line output:

```
FALLBACK (log once): Fallback to SW vertex for line stipple
FALLBACK (log once): Fallback to SW vertex processing, m_disable_code: 2000
FALLBACK (log once): Fallback to SW vertex processing in drawCore, m_disable_code: 2000
```

No performance measurements has been made to evaluate if this has a performance impact.

kintel, 2021-12-10:
Screenshot indicating that the (buggy) SW renderer is indeed being used:
[image]

jspiro, 2022-01-12:
Did you build yourself, or is the M1 build published somewhere?

kintel, 2022-01-12:
This was a build I made as part of #4011, we plan to publish the binaries once we have universal binary support.
The current master should build native M1 binaries if you want to do it yourself.