openscad/openscad #4015: Shebang causes an unexpected syntax error

Kreyren, 2021-12-19:
[image]

### Expected:
Ignore shebang

t-paul, 2021-12-19:
Duplicate of #1172.

t-paul, 2021-12-19:
Oh, sorry, misunderstood the report.

But that's not going to happen, `#` is not a comment character.

closed 2021-12-19 by t-paul

jordanbrown0, 2021-12-19:
We could specially treat `#!` at the top of a file as a comment, with only very limited impact.  We could specially treat `#! <whitespace> /` as a comment and I can't see how there would be any impact.

But I'm also not at all clear how making an OpenSCAD program directly executable would be useful.

t-paul, 2021-12-19:
`#!cube();`

Is valid OpenSCAD code. Declaring it different because it's on the first line is a very bad idea.

Kreyren, 2021-12-19:
> We could specially treat #! at the top of a file as a comment, with only very limited impact.  -- @jordanbrown0  (https://github.com/openscad/openscad/issues/4015#issuecomment-997306787)

Note that Linux's shebang is a **BLOCK** comment so `^(#!)[a-zA-Z/ \n](!#)` should be ignored (could be multi-line)

> But I'm also not at all clear how making an OpenSCAD program directly executable would be useful. -- @jordanbrown0 (https://github.com/openscad/openscad/issues/4015#issuecomment-997306787)

It's an important part of Linux and GNU/Linux compatibility so there are multiple usecases, but the major one for me is to make OpenSCAD-scripts (would be cool if some additional functionality be added to interact with POSIX e.g. execting file to generate the manufacturing instructions and start the manufacturing) and declare the file as openscad without using xattr and bash metadata which are less storage efficient.

t-paul, 2021-12-19:
On Linux you can simply define a custom `binfmt` and declare whatever valid comment you like to have.

See for example the Debian configuration: https://manpages.debian.org/testing/binfmt-support/update-binfmts.8.en.html

Kreyren, 2021-12-19:
> On Linux you can simply define a custom binfmt and declare whatever valid comment you like to have. @t-paul (https://github.com/openscad/openscad/issues/4015#issuecomment-997309757)

That is dependent on system configuration and somewhat pain to set up so not a valid solution for my usecase.

t-paul, 2021-12-19:
Yes, it requires root as it's a kernel level configuration. But it's a kernel interface, you can drive that with whatever user land you have. The Debian script was just an example.

I don't see any reasonable option on OpenSCAD side. I'm open to suggestions, but with `#` not being a comment character, I'm not sure there's much possibilities.

Kreyren, 2021-12-19:
Shebang is kernel level interface too and does not require root..?

> I don't see any reasonable option on OpenSCAD side. I'm open to suggestions, but with # not being a comment character, I'm not sure there's much possibilities. -- @t-paul (https://github.com/openscad/openscad/issues/4015#issuecomment-997313793)

It's not expected to prevent `#` comments just ignore `#! .. !#` block comment on first line

t-paul, 2021-12-19:
Again, '#' is NOT a comment and it's already in use for the debug modifier feature. Only `//' and '/* */'.

See https://blog.gonzih.me/blog/2019/04/26/scripting-with-go-on-nixos/ for options in NixOS.

Kreyren, 2021-12-19:
I am not proposing to implement a special behavior for `#` to be used as part of OpenSCAD -> i am proposing the implementation compatibility for *shebang* which is *nix-specific functionality (Linux, Redox, *BSD, Darwin, Cygwin) for `#! .. !#` **only** on first line (not to be confused with `#`).

> See https://blog.gonzih.me/blog/2019/04/26/scripting-with-go-on-nixos/ for options in NixOS. -- @t-paul 

That is functional programing defining a definition for system rebuild which is in Nix"s spec by edolstra the most guarded functionality on the system that non-root user will never get access to as that would mean that the user would have the access to the interface to change the whole system (and in some cases the whole network/cluster).

Even if the user convinced the sysadmin to make changes they would be impossible to do in a way that doesn't cause conflicts with other users.

jordanbrown0, 2021-12-19:
Torsten is absolutely right that
#!cube();
is a valid first line for an OpenSCAD program.  Kind of unlikely, but valid.  (That's the "limited impact" I was mentioning.)

Hmm.  I had thought that `#!<optional whitespace>/` was safe, but in theory the first line of an OpenSCAD file can be
```
#!/*comment*/cube();
```

I'm still really unclear on the use case.  Could you give a small-but-concrete example?

MichaelAtOz, 2021-12-19:
`.scad` are platform independent OpenSCAD source files. They are not *nix scripts. 

You also do not want `.scad` files downloaded from somewhere firing off programs.

They should not be corrupted in the fashion proposed.

I don't support this proposal, nor including any Windows or anything else's syntax.

Kreyren, 2021-12-19:
> I'm still really unclear on the use case. Could you give a small-but-concrete example? -- @jordanbrown0 (https://github.com/openscad/openscad/issues/4015#issuecomment-997319770)

The simplest i can think of would be:

`#!/usr/bin/env -S openscad  -D kreyren=something`

to define file specific variables which can be used alike

`#!/usr/bin env -S openscad -D x="${1:-"2"}" -o "${2:-"-"}"`

to define a purpose-specific files that can be used in further processing with e.g.

`./cube.scad 5 st\l | prusa-slicer`

> .scad are platform independent OpenSCAD source files. They are not *nix scripts. -- @MichaelAtOz (https://github.com/openscad/openscad/issues/4015#issuecomment-997342412)

Everything is a *nix script if you try hard enough :p 

> You also do not want .scad files downloaded from somewhere firing off programs. -- @MichaelAtOz (https://github.com/openscad/openscad/issues/4015#issuecomment-997342412)

None sane on linux doesn't execute random files from the web without reviewing them or requiring source code.

MichaelAtOz, 2021-12-19:
> None sane on linux doesn't 

You will be a small percent of the global population who do not at some time have a pebcak.
Built it and they will come... 
And abuse it in the worst possible way.

jordanbrown0, 2021-12-19:
Seems like an **extremely** limited use case - seems to apply only to those models that are parameterized and reusable enough to **want** to execute them ultra-conveniently, **and** have custom requirements that are simple enough that wrapping them in a one-line OpenSCAD invocation is enough.  **And** having enough such cases that just wrapping the file in a small shell script is too onerous.

But hmm.  You can *almost* make this work now.

First, the `#!` is not **illegal** in OpenSCAD.  It just has a particular meaning:  highlight the part, and build only that part.

Second, most or all UNIX variants allow multiple slashes anywhere one slash is allowed.

Third, `//` starts an OpenSCAD comment.

Thus:
```
#! //bin/openscad
main();
module main() {
   ....
}
```

is valid OpenSCAD and also a valid #! file.  (Tested on Ubuntu.)  The problem is that the entire program gets subjected to the # modifier.  Still, for generating STL that's OK.

Kreyren, 2021-12-19:
> You will be a small percent of the global population who do not at some time have a pebcak. -- @MichaelAtOz (https://github.com/openscad/openscad/issues/4015#issuecomment-997382653)

pebcak is part of threat modeling where common handling is making a backups of  ungeneratable files

for example i am using reproducible distro synced on E2EE nextcloud so even if i nuke the system i can have it back in around 7 min.